micropython/micropython #16025: Unanswered ARP requests - Destination host unreachable

ploughpuff, 2024-10-16:
### Port, board and/or hardware

Raspberry Pi Pico W with RP2040

### MicroPython version

MicroPython v1.23.0 on 2024-06-02

### Reproduction

I running in Windows environment.  My Pico is setup for DHCP from my router with static address of 192.168.1.210:

1. Power-up board, and copy main.py code from Gist and flash to board: https://gist.github.com/ploughpuff/caef2270c73014ad83ecb73337a487eb
2.  Power-down board: 
3. Open serial terminal (I use CoolTerm) on COM port
4. Open WireShark
5. Open CMD prompt in admin mode

Power-up the Pico, and in Wireshark notice ARP broadcast message from Pico:
`1216	9.285395	RaspberryPiT_66:c8:77	Broadcast	ARP	60	ARP Announcement for 192.168.1.210`

With trace=15 enabled, note the ARP messages being sent in the serial terminal:
```
[    3807] ETHTX len=350 dst=ff:ff:ff:ff:ff:ff src=d8:3a:dd:66:c8:77 type=IPv4 srcip=0.0.0.0 dstip=255.255.255.255 UDP srcport=68 dstport=67 DHCPC 01 01 06 00 c4 66 9f 3f 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 d8 3a dd 66 c8 77 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 opts: DISCOVER data= 63 82 53 63 35 01 01 39 02 05 dc 37 04 01 03 1c 06 ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 40 1f 00 00 00 00 00 00
[    3936] ETHRX len=342 dst=d8:3a:dd:66:c8:77 src=e8:ad:a6:ee:0d:58 type=IPv4 srcip=192.168.1.254 dstip=192.168.1.210 UDP srcport=67 dstport=68 DHCPS 02 01 06 00 c4 66 9f 3f 00 00 00 00 00 00 00 00 c0 a8 01 d2 c0 a8 01 fe 00 00 00 00 d8 3a dd 66 c8 77 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 opts: OFFER data= 63 82 53 63 35 01 02 36 04 c0 a8 01 fe 33 04 00 1b af 80 01 04 ff ff ff 00 03 04 c0 a8 01 fe 1c 04 c0 a8 01 ff 06 04 c0 a8 01 fe ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    3943] ETHTX len=350 dst=ff:ff:ff:ff:ff:ff src=d8:3a:dd:66:c8:77 type=IPv4 srcip=0.0.0.0 dstip=255.255.255.255 UDP srcport=68 dstport=67 DHCPC 01 01 06 00 c4 66 9f 3f 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 d8 3a dd 66 c8 77 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 opts: REQUEST data= 63 82 53 63 35 01 03 39 02 05 dc 32 04 c0 a8 01 d2 36 04 c0 a8 01 fe 37 04 01 03 1c 06 0c 05 50 69 63 6f 57 ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 40 1f 00 00 00 00 00 00
[    4925] ETHRX len=342 dst=d8:3a:dd:66:c8:77 src=e8:ad:a6:ee:0d:58 type=IPv4 srcip=192.168.1.254 dstip=192.168.1.210 UDP srcport=67 dstport=68 DHCPS 02 01 06 00 c4 66 9f 3f 00 00 00 00 00 00 00 00 c0 a8 01 d2 c0 a8 01 fe 00 00 00 00 d8 3a dd 66 c8 77 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 opts: ACK data= 63 82 53 63 35 01 05 36 04 c0 a8 01 fe 33 04 00 1b af 80 01 04 ff ff ff 00 03 04 c0 a8 01 fe 1c 04 c0 a8 01 ff 06 04 c0 a8 01 fe ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    4932] ETHTX len=42 dst=ff:ff:ff:ff:ff:ff src=d8:3a:dd:66:c8:77 type=ARP data= 00 01 08 00 06 04 00 01 d8 3a dd 66 c8 77 c0 a8 01 d2 00 00 00 00 00 00 c0 a8 01 d2
WiFi: GOT_IP 192.168.1.210  Signal Strength -69 dB
MicroPython v1.23.0 on 2024-06-02; Raspberry Pi Pico W with RP2040
Type "help()" for more information.
>>> 
```

So far, so good.  We can PING the Pico board ok, and see expected messages in serial terminal, and wireshark:
```
4733	47.714153	RaspberryPiT_66:c8:77	Broadcast	ARP	60	Who has 192.168.1.108? Tell 192.168.1.210
4734	47.714163	GigaByteTech_38:54:eb	RaspberryPiT_66:c8:77	ARP	42	192.168.1.108 is at 18:c0:4d:38:54:eb
5345	52.399046	GigaByteTech_38:54:eb	RaspberryPiT_66:c8:77	ARP	42	Who has 192.168.1.210? Tell 192.168.1.108
5346	52.404013	RaspberryPiT_66:c8:77	GigaByteTech_38:54:eb	ARP	60	192.168.1.210 is at d8:3a:dd:66:c8:77
```

```
C:\WINDOWS\system32>ping 192.168.1.210

Pinging 192.168.1.210 with 32 bytes of data:
Reply from 192.168.1.210: bytes=32 time=19ms TTL=255
Reply from 192.168.1.210: bytes=32 time=6ms TTL=255
Reply from 192.168.1.210: bytes=32 time=6ms TTL=255
Reply from 192.168.1.210: bytes=32 time=5ms TTL=255

Ping statistics for 192.168.1.210:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 5ms, Maximum = 19ms, Average = 9ms

C:\WINDOWS\system32>arp -a 192.168.1.210

Interface: 192.168.1.108 --- 0x9
  Internet Address      Physical Address      Type
  192.168.1.210         d8-3a-dd-66-c8-77     dynamic
```

Finally, delete the ARP entry from the PC and try to PING the Pico again.  The PING fails with 'Destination host unreachable'.  The ARP requests towards the Pico are seen in Wireshark but they all go unanswered:
```
C:\WINDOWS\system32>arp -d 192.168.1.210

C:\WINDOWS\system32>ping 192.168.1.210

Pinging 192.168.1.210 with 32 bytes of data:
Reply from 192.168.1.108: Destination host unreachable.
Reply from 192.168.1.108: Destination host unreachable.
Reply from 192.168.1.108: Destination host unreachable.
Reply from 192.168.1.108: Destination host unreachable.

Ping statistics for 192.168.1.210:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
```

### Expected behaviour

Expectation is the Pico responds to all ARP requests.  In my case:
`RaspberryPiT_66:c8:77	GigaByteTech_38:54:eb	ARP	60	192.168.1.210 is at d8:3a:dd:66:c8:77`

### Observed behaviour

See reproduction log.

### Additional Information

[image]

### Code of Conduct

Yes, I agree

dpgeorge, 2024-10-17:
Thanks for the clear bug report.

I had a quick go at reproducing this (host PC an Arch Linux machine), but could not.  The Pico W responds just fine to an ARP the second time around.  Here's the log from the Pico W:

Initial ping from the PC, note the ARP at the start which is replied to by the Pico W:
```
[  122378] ETHRX len=42 dst=ff:ff:ff:ff:ff:ff src=fc:77:74:1e:a3:2a type=ARP data= 00 01 08 00 06 04 00 01 fc 77 74 1e a3 2a c0 a8 00 0c 00 00 00 00 00 00 c0 a8 00 2f
[  122380] ETHTX len=42 dst=fc:77:74:1e:a3:2a src=28:cd:c1:00:0c:0f type=ARP data= 00 01 08 00 06 04 00 02 28 cd c1 00 0c 0f c0 a8 00 2f fc 77 74 1e a3 2a c0 a8 00 0c
[  122406] ETHRX len=98 dst=28:cd:c1:00:0c:0f src=fc:77:74:1e:a3:2a type=IPv4 srcip=192.168.0.12 dstip=192.168.0.47 prot=1 data= 08 00 03 59 00 23 00 01 b5 51 10 67 00 00 00 00 67 f7 08 00 00 00 00 00 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33 34 35 36 37
[  122409] ETHTX len=98 dst=fc:77:74:1e:a3:2a src=28:cd:c1:00:0c:0f type=IPv4 srcip=192.168.0.47 dstip=192.168.0.12 prot=1 data= 00 00 0b 59 00 23 00 01 b5 51 10 67 00 00 00 00 67 f7 08 00 00 00 00 00 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33 34 35 36 37
[  123402] ETHRX len=98 dst=28:cd:c1:00:0c:0f src=fc:77:74:1e:a3:2a type=IPv4 srcip=192.168.0.12 dstip=192.168.0.47 prot=1 data= 08 00 24 51 00 23 00 02 b6 51 10 67 00 00 00 00 45 fe 08 00 00 00 00 00 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33 34 35 36 37
[  123405] ETHTX len=98 dst=fc:77:74:1e:a3:2a src=28:cd:c1:00:0c:0f type=IPv4 srcip=192.168.0.47 dstip=192.168.0.12 prot=1 data= 00 00 2c 51 00 23 00 02 b6 51 10 67 00 00 00 00 45 fe 08 00 00 00 00 00 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33 34 35 36 37
[  124426] ETHRX len=98 dst=28:cd:c1:00:0c:0f src=fc:77:74:1e:a3:2a type=IPv4 srcip=192.168.0.12 dstip=192.168.0.47 prot=1 data= 08 00 b0 49 00 23 00 03 b7 51 10 67 00 00 00 00 b8 04 09 00 00 00 00 00 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33 34 35 36 37
[  124430] ETHTX len=98 dst=fc:77:74:1e:a3:2a src=28:cd:c1:00:0c:0f type=IPv4 srcip=192.168.0.47 dstip=192.168.0.12 prot=1 data= 00 00 b8 49 00 23 00 03 b7 51 10 67 00 00 00 00 b8 04 09 00 00 00 00 00 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33 34 35 36 37
[  125450] ETHRX len=98 dst=28:cd:c1:00:0c:0f src=fc:77:74:1e:a3:2a type=IPv4 srcip=192.168.0.12 dstip=192.168.0.47 prot=1 data= 08 00 6d 44 00 23 00 04 b8 51 10 67 00 00 00 00 fa 08 09 00 00 00 00 00 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33 34 35 36 37
[  125454] ETHTX len=98 dst=fc:77:74:1e:a3:2a src=28:cd:c1:00:0c:0f type=IPv4 srcip=192.168.0.47 dstip=192.168.0.12 prot=1 data= 00 00 75 44 00 23 00 04 b8 51 10 67 00 00 00 00 fa 08 09 00 00 00 00 00 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33 34 35 36 37
```

Then after removing the ARP on the host PC (via `sudo arp -d 192.168.0.47`) a ping again starts with an ARP that is responded to:
```
[  161700] ETHRX len=42 dst=ff:ff:ff:ff:ff:ff src=fc:77:74:1e:a3:2a type=ARP data= 00 01 08 00 06 04 00 01 fc 77 74 1e a3 2a c0 a8 00 0c 00 00 00 00 00 00 c0 a8 00 2f
[  161702] ETHTX len=42 dst=fc:77:74:1e:a3:2a src=28:cd:c1:00:0c:0f type=ARP data= 00 01 08 00 06 04 00 02 28 cd c1 00 0c 0f c0 a8 00 2f fc 77 74 1e a3 2a c0 a8 00 0c
[  161708] ETHRX len=98 dst=28:cd:c1:00:0c:0f src=fc:77:74:1e:a3:2a type=IPv4 srcip=192.168.0.12 dstip=192.168.0.47 prot=1 data= 08 00 67 94 00 24 00 01 dc 51 10 67 00 00 00 00 d7 ba 0d 00 00 00 00 00 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33 34 35 36 37
[  161711] ETHTX len=98 dst=fc:77:74:1e:a3:2a src=28:cd:c1:00:0c:0f type=IPv4 srcip=192.168.0.47 dstip=192.168.0.12 prot=1 data= 00 00 6f 94 00 24 00 01 dc 51 10 67 00 00 00 00 d7 ba 0d 00 00 00 00 00 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33 34 35 36 37
[  162734] ETHRX len=98 dst=28:cd:c1:00:0c:0f src=fc:77:74:1e:a3:2a type=IPv4 srcip=192.168.0.12 dstip=192.168.0.47 prot=1 data= 08 00 a9 8b 00 24 00 02 dd 51 10 67 00 00 00 00 94 c2 0d 00 00 00 00 00 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33 34 35 36 37
[  162737] ETHTX len=98 dst=fc:77:74:1e:a3:2a src=28:cd:c1:00:0c:0f type=IPv4 srcip=192.168.0.47 dstip=192.168.0.12 prot=1 data= 00 00 b1 8b 00 24 00 02 dd 51 10 67 00 00 00 00 94 c2 0d 00 00 00 00 00 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33 34 35 36 37
[  163748] ETHRX len=98 dst=28:cd:c1:00:0c:0f src=fc:77:74:1e:a3:2a type=IPv4 srcip=192.168.0.12 dstip=192.168.0.47 prot=1 data= 08 00 7c 85 00 24 00 03 de 51 10 67 00 00 00 00 c0 c7 0d 00 00 00 00 00 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33 34 35 36 37
[  163751] ETHTX len=98 dst=fc:77:74:1e:a3:2a src=28:cd:c1:00:0c:0f type=IPv4 srcip=192.168.0.47 dstip=192.168.0.12 prot=1 data= 00 00 84 85 00 24 00 03 de 51 10 67 00 00 00 00 c0 c7 0d 00 00 00 00 00 10 11 12 1 [... cut ...]
[  164669] ETHRX len=98 dst=28:cd:c1:00:0c:0f src=fc:77:74:1e:a3:2a type=IPv4 srcip=192.168.0.12 dstip=192.168.0.47 prot=1 data= 08 00 55 80 00 24 00 04 df 51 10 67 00 00 00 00 e6 cb 0d 00 00 00 00 00 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33 34 35 36 37
[  164673] ETHTX len=98 dst=fc:77:74:1e:a3:2a src=28:cd:c1:00:0c:0f type=IPv4 srcip=192.168.0.47 dstip=192.168.0.12 prot=1 data= 00 00 5d 80 00 24 00 04 df 51 10 67 00 00 00 00 e6 cb 0d 00 00 00 00 00 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33 34 35 36 37
```

dpgeorge, 2024-10-17:
I tried with both 1.23.0 and latest master.  @ploughpuff can you please retest with latest master preview release, because lwIP was updated since 1.23.0.

ploughpuff, 2024-10-18:
The positive results from your testing made me question my home network.  I flashed the latest build and got the same results with Pico, but it also caused me to clear my filter in WireShark, and doing so I noticed similar ARP requests going unanswered by other (non Pico) devices.

The router is an ISP supplied jobbie, 5+ years old, and switching to something newer and I was unable to reproduce my issue.  All ARP requests were answered as expected.

I must conclude my ISP router does not like ARP broadcast frames and refused to FWD them to the network.  Thank you for the help and apologies for wasting any of your time.  Please close the ticket.

```
C:\WINDOWS\system32>ping 192.168.1.210

Pinging 192.168.1.210 with 32 bytes of data:
Reply from 192.168.1.210: bytes=32 time=21ms TTL=255
Reply from 192.168.1.210: bytes=32 time=6ms TTL=255
Reply from 192.168.1.210: bytes=32 time=6ms TTL=255
Reply from 192.168.1.210: bytes=32 time=8ms TTL=255

Ping statistics for 192.168.1.210:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 6ms, Maximum = 21ms, Average = 10ms

C:\WINDOWS\system32>arp -d 192.168.1.210

C:\WINDOWS\system32>ping 192.168.1.210

Pinging 192.168.1.210 with 32 bytes of data:
Reply from 192.168.1.210: bytes=32 time=99ms TTL=255
Reply from 192.168.1.210: bytes=32 time=15ms TTL=255
Reply from 192.168.1.210: bytes=32 time=8ms TTL=255
Reply from 192.168.1.210: bytes=32 time=5ms TTL=255

Ping statistics for 192.168.1.210:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 5ms, Maximum = 99ms, Average = 31ms
```

dpgeorge, 2024-10-21:
> I must conclude my ISP router does not like ARP broadcast frames and refused to FWD them to the WiFi network.

OK, very interesting.  Thanks for following up and working out the issue.

> Thank you for the help and apologies for wasting any of your time.

Not a problem, glad this was quickly resolved.

closed 2024-10-21 by dpgeorge