micropython/micropython #16053: `FrameBuffer.ellipse` goes into infinite loop with 0 radius

corranwebster, 2024-10-21:
### Port, board and/or hardware

unix, RP2

### MicroPython version

MicroPython v1.23.0 on 2024-06-02; Raspberry Pi Pico with RP2040
MicroPython v1.23.0 on 2024-05-31; darwin [GCC 4.2.1] version

### Reproduction

In an interactive session (or otherwise), run the following code:
```
import framebuf
buf = bytearray(10*10*2)
f = framebuf.FrameBuffer(buf, 10, 10, framebuf.RGB565)
f.ellipse(5, 5, 0, 0, 0xffff, True)
```

### Expected behaviour

Nothing should be drawn.  Code should complete normally.

*Maybe* an exception should be raised, but I think it's OK to just draw nothing.

### Observed behaviour

Micropython hangs; 100% CPU usage observed in unix port.  Strong suspicion this is an infinite loop.

### Additional Information

I'm very suspicious of the following code: https://github.com/micropython/micropython/blob/1b89c503db690967d50699abe0bfa942f6f6b15e/extmod/modframebuf.c#L542-L554

In this case `two_asquare` and `two_bsquare` are both 0, so the `stoppingx` and `stoppingy` look like they are never changed, but I haven't 100% proven it to myself.

### Code of Conduct

Yes, I agree

corranwebster, 2024-11-14:
Some more thought about this, and I think the expected behaviour should be to draw a single pixel at the centre.

mattytrentini, 2024-11-14:
> Some more thought about this, and I think the expected behaviour should be to draw a single pixel at the centre.

Either a single pixel or no pixel would be fine IMO. But infinite loop=bad. 😜

dpgeorge, 2024-11-28:
Fixed by e70048cf5934bf3db65ec7a1b6490635716282f5

closed 2024-11-28 by dpgeorge