micropython/micropython #16079: UART with the same pin for both RX and TX doesn't work on ESP32 variants

arduino12, 2024-10-25:
### Port, board and/or hardware

esp32 port

### MicroPython version

MicroPython v1.23.0 on 2024-06-02; LOLIN_C3_MINI with ESP32-C3FH4
MicroPython v1.23.0 on 2024-06-02; LOLIN_S2_MINI with ESP32-S2FN4R2
MicroPython v1.23.0 on 2024-06-02; Generic ESP32S3 module with ESP32S3

### Reproduction

```python
from micropython import const
from machine import Pin, UART, mem32

PIN_NUM = const(2)
PIN_FUNC_REG = const(0x60004554 + 4 * PIN_NUM) # GPIO_FUNCx_OUT_SEL_CFG_REG 
UART_NUM = const(1)
UART_BAUDRATE = const(1000000)

pin = Pin(PIN_NUM)

# pass the same `pin` to `tx` and `rx` doesn't work!
uart = UART(UART_NUM, UART_BAUDRATE, tx=pin, rx=pin)
uart.write(b'test') # UART signal doesn't routed to `PIN_NUM `.
print(uart.read()) # nothing gets printed..

# pass only `tx` (to get the `UART_SIG` number)
uart = UART(UART_NUM, UART_BAUDRATE, tx=pin)
UART_SIG = mem32[PIN_FUNC_REG] # see "Peripheral Signals via GPIO Matrix" in datasheet

# now try again, this time overwrite `PIN_FUNC_REG` with previous value
uart = UART(UART_NUM, UART_BAUDRATE, tx=pin, rx=pin)
mem32[PIN_FUNC_REG] = UART_SIG # ESP32-C3 is 6, ESP32-S3 is 12...
uart.write(b'test')
print(uart.read()) # b'test' is printed!

```

### Expected behaviour

I expected:
```python
uart = UART(UART_NUM, UART_BAUDRATE, tx=pin, rx=pin)
```
to work- but is seems to overwrite the TX output configuration and only the RX pin is configured correctly.

My workaround is to re-write the TX output configuration after the UART initialization:
```python
uart = UART(UART_NUM, UART_BAUDRATE, tx=pin, rx=pin)
mem32[PIN_FUNC_REG] = UART_SIG
```
I think the `UART` class needs to support this case of a single pin for both TX and RX.
I use it to control some single line smart servo motors with open-drain UART bus...

The ESP32 support this common UART configuration (like RS485),
it also support **[echo-cancellation](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/uart.html#overview-of-rs485-specific-communication-0ptions)** that may also be a nice option for the `UART` class..

### Observed behaviour

Tested the provided sample code on ESP32-S2, ESP32-S3, ESP32-C3 -
All have the same bug and the same suggested workaround fix.

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

arduino12, 2024-10-25:
**Hi,**

Thanks for your quick reply!

I started with bridging RX and TX lines together-
but then I thought its much better to "bridge" them in the ESP32 GPIO MATRIX.

**I still think its a bug of the `UART` class** - because as I wrote- 
providing same `Pin` object to both `rx` and `tx` arguments resulting in a non working TX output...
If this is not supported- maybe raise an `Exception` so it will be clear?
(Also I think that supporting it in the ESP32 platform is too simple - as my workaround proofs)

I will try implementing support for same RX and TX pin by myself and send a PR after testing it :)

robert-hh, 2024-10-25:
See PR #16080. Mentioned there is the addition of a mode=... keyword argument for the proper mode. That would in the beginning consistent with other ports, which e.g. do not require or support specifying rx and tx pins.

arduino12, 2024-10-25:
I think that PR only changes the behavior of the `RTS` pin to indicate TX is active-
to control the rs485 transceiver `TX_EN` pin?

Anyway I played with the ESP-IDF and found out that the `uart_set_pin(..)` is causing this bug:
https://github.com/espressif/esp-idf/issues/14787

Also added my workaround there - hope they can accept it because if they fix it in the ESP-IDF-
the fix should also work here on the next version :)