micropython/micropython #16083: docs: DMA chain example shows incorrect use of DMA.registers

sk8board, 2024-10-26:
### Documentation URL

https://docs.micropython.org/en/latest/library/rp2.DMA.html#chaining-and-trigger-register-access

### Description

### Port, board and/or hardware

rp2, Pico RP2040

### MicroPython version

MicroPython v1.23.0 on 2024-06-02; Raspberry Pi Pico with RP2040

### Reproduction

``` py
import rp2

test_dma = rp2.DMA()

test_dma.active(1)

print("R",hex(test_dma.registers[0]))
print("W",hex(test_dma.registers[1]))
print("T",hex(test_dma.registers[2]))
print("C",hex(test_dma.registers[3]))
print("C",hex(test_dma.registers[4]))
print("R",hex(test_dma.registers[5]))
print("W",hex(test_dma.registers[6]))
print("T",hex(test_dma.registers[7]))
print("C",hex(test_dma.registers[8]))
print("T",hex(test_dma.registers[9]))
print("R",hex(test_dma.registers[10]))
print("W",hex(test_dma.registers[11]))
print("C",hex(test_dma.registers[12]))
print("W",hex(test_dma.registers[13]))
print("T",hex(test_dma.registers[14]))
print("R",hex(test_dma.registers[15]))
```

### Expected behaviour

Base on the [DMA chain example](https://docs.micropython.org/en/latest/library/rp2.DMA.html#chaining-and-trigger-register-access), I expected a result with the address of a DMA register.

[image]

### Observed behaviour

I received this result, which shows the value within the DMA register.

[image]

### Additional Information

With regard to `DMA.config()`, this [documentation](https://docs.micropython.org/en/latest/library/rp2.DMA.html#rp2.DMA.config) states:

> write: The address to which the DMA controller will start writing or an object into which data will be written.

This [example](https://docs.micropython.org/en/latest/library/rp2.DMA.html#chaining-and-trigger-register-access), shows:

``` py
gather_dma.config(
        read=gather_list, write=buffer_dma.registers[14:16],
        count=2, ctrl=gather_ctrl
    )
```

where `DMA.registers` is used to obtain the address of the register.

Because of how `DMA.registers` is used in the example, I expected `DMA.registers` to provide DMA register addresses, but the result shows `DMA.registers` provides values within the register.

Therefore, I believe the example is showing incorrect use of `DMA.registers`, which has caused confusion as shown [here](https://github.com/orgs/micropython/discussions/16055). 

### Recommendation:

1) Update [`DMA.registers`](https://docs.micropython.org/en/latest/library/rp2.DMA.html#rp2.DMA.registers) with additional language to explain how to use `DMA.registers` and the expected result.

2) Update the chain example to include two common chain use cases: 
    a) "ping-pong" when changes in DMA config are not needed
    b) when one DMA is used to change the config of another DMA
         i) provide a way to write to a DMA register

### Code of Conduct

Yes, I agree

fmafma, 2024-10-27:
@sk8board I found that `DMA.register[0]` returns the content of the register 0, but `DMA.register[0:0]` returns a `<memoryview>` object pointing to register 0, which can be used in the `write=` param.

That's why the example works, as it uses a slice...

If this is intended, it just needs to be documented.

dpgeorge, 2024-10-28:
@nickovs FYI

nickovs, 2024-10-28:
The behaviour you see is indeed intended. As per the existing documentation:

> **DMA.registers**
>   This attribute is an **_array-like_** object that allows direct access to the DMA channel’s registers.

and

> **DMA.write**
> This attribute reflects the address to which the next bus transfer will write. It may be written with either an integer or an object that supports the **_buffer protocol_** and doing so has immediate effect.

Since arrays and array-like objects support the buffer protocol, arrays and slices of arrays can be provided for the read or write addresses, and that includes the registers array.

If you think it's necessary I can add an extra line or two to the documentation to explain this.

fmafma, 2024-10-28:
Yes, just to reiterate the difference between accessing a single array element and a slice.... I think something like this might be useful:

>Since accessing a single element of an array returns that element, not an array [supporting the buffer protocol], one should always use a slice.

(feel free to rephrase, English is not my first language).

Thanks!

sk8board, 2024-10-28:
I believe additional language is needed to explain how to use `DMA.registers`.

I recently went through this learning curve of how to use DMA with MicroPython, which seemed unnecessarily difficult. I noticed there are other people who struggled to understand how to implement DMA with MicroPython. I believe this struggle is due to documentation that does not provide examples of how to use MicroPython with DMA.

From my experience, there seems to two main use cases applied to DMA, which I have documented above. I recommend to add two examples that clearly explain how to apply MicroPython to these two main use cases of DMA.

nickovs, 2024-10-28:
Can you provide some more specific examples of what your use cases are and what you struggled to do?

The bottom line is that DMA is fundamentally complicated, since it requires the user to understand that hardware with which they are interacting as well as the nuances of the DMA system. I would rather not fill up the MicroPython class documentation with a tutorial about hardware DMA. There is probably a case for having some separate, more complete examples but I think that these will need to be distinct from the class docs.

sk8board, 2024-10-28:
I believe Raspberry Pi realized these were the two most common use cases for DMA with the RP2040.

[image]

sk8board, 2024-10-28:
[Here](https://github.com/HughMaingauche/PICO-VGA-Micropython/blob/1729a09f0e3295e43b2b5b7c7a3560ec4c19d239/VGA.py#L129-L165) is an example using MicroPython Viper for the first application mentioned in the RP2040 Datasheet.

I also recommend checking DMA2 and DMA3 in [my MicroPython example.](https://github.com/sk8board/DCC_MicroPython_Decoder_for_Pico_RP2040/blob/274e2d8240ad18a0e69efcca1032b1b100718903/DMA%20Examples/DMA%20Chain%20Example%20with%20PIO.py#L89-L124) which shows how to support the second application mentioned in the RP2040 Datasheet.

sk8board, 2024-10-28:
Please notice how both examples have many easy to understand comments which help reduce the complication of understanding how to configure DMA.

nickovs, 2024-10-28:
@sk8board  Thanks for the input. I'm sorry that you're struggling with this.

> I believe Raspberry Pi realized these were the two most common use cases for DMA with the RP2040.
> ...

Both of those examples that you point to are use cases for chaining specifically rather than use cases for DMA per se. Are there any other areas that you feel need further details? I can prepare some examples for inclusion in the `micropython/examples/rp2` directory.

nickovs, 2024-10-30:
> BTW, with a standard `array` **a**, `addressof(a[n:m])` return the address of the element n only if m > n, 0 otherwise.
> 
> On the other hand, on `DMA.registers`, `addressof()` returns the address of the element n, whatever the value of m is.

The `registers` attribute is a `memoryview`, (which is why it says "array-like", not `array`). If you take a zero-length slice of a standard `memoryview` it still has an address. This is not a feature of the DMA code; it's standard MicroPython.

Note that slices of `array`s take a copy of the contents and make a new `array`; zero-length arrays don't get any memory allocated since they don't need it, hence them not having an address. Slices of `memoryview`s on the other hand are always references to an offset in the original `memoryview`, so the address of the slice view is just the address of the original view plus an offset.

fmafma, 2024-10-30:
Thanks for the explanations!

nickovs, 2024-10-30:
> I noticed the documentation does not provide default values for:
> 
> ring_size: 

Yes, this one deserves an explicit statement for the default.

> ring_sel: 

If you've not explicitly set this one then there is almost certainly a bug in your code. As it happens it defaults to wrapping the read address, but this should always be set whenever you are also setting ring_size; otherwise the intent of your code will be unclear.

> write_err: 
> read_err:

As documented, these are error-clearing bits, not useful values. The documentation currently says _"Setting this to True will clear a previously reported write error."_ I really do not think it's necessary to add "If you don't set this flag then it won't clear the error."

> After using documentation and implementing DMA as well as helping Fred with his DMA application, I still do not understand the following:
> 
> * With regard to `inc_read:` and `inc_write:`, is the size of the increment determined by `size:` or `ring_size:`? How would someone know which is used?
> * What is `ring_size:`? Should a simple example be provided to help with understanding?

These topics are covered extensively in the DMA section of the RP2040 datasheet, which is linked directly near the top of the page. As I mentioned above, the DMA system is a complex system with a great many features. The class documentation for the MicroPython driver is not the right place to have a tutorial about the whole DMA architecture.

> `DMA.registers` states that is is "array-like", which is why Fred used `DMA.registers[15]`, but it did not work. 

With respect, actually, it did work. If you index any Python or MicroPython array with a single value you will get the *value* at that index, which is precisely what he got, even if it is not what he wanted.

> It might be helpful to show what works and what array-like does not work? Maybe show how `DMA.registers` can be used as an input to a register and an output of register values?

All array operations work exactly as with any other array. In general most users shouldn't need to be accessing the register bank directly. The point of having a high-level interface is to avoid users needing to know anything about the registers. If you just need to transfer data from a peripheral to memory or back again then your code will be more understandable and easier to write if you don't touch the register bank. The register access is there for people who need to do much more complex things for which it is necessary to have read the whole of the datasheet first.

> `DMA.registers[14:16]` is provided in the example. Why was 16 used if the highest available value is 15?

This is standard Python! 
```
>>> a = "0123456789abcdef"
>>> a[14:15]
'e'
>>> a[14:16]
'ef'
```

The example you are referring to is copying new values into the last *two* registers.

fmafma, 2024-10-30:
> The example you are referring to is copying new values into the last two registers.

But in fact, we only need the address of the **first** register of all registers we want to write to, so anything returning this address is fine. In this case, what matters is the `ring_size` value, to know how many registers we want to write to, right?

I agree, we don't really need to access to the registers themselves, but their **addresses**. So, maybe a mecanism to do that whould be more clear. What about a `register_addr[n]`, returning the address of that register, so we can give it to the `write` param?

About your comment about `ring_sel`, you don't need to set it when `ring_size` is 0, which disables the wrap mecanism. In many cases, we don't need to wrap, so I guess having a default value to 0 for `ring_size` is good (which is the default at RP204 startup).

sk8board, 2024-10-30:
> These topics are covered extensively in the DMA section of the RP2040 datasheet, which is linked directly near the top of the page. As I mentioned above, the DMA system is a complex system with a [...]

I have spent significant time in the RP2040 datasheet and have not been able to find a description of `ring_size` that makes sense. Therefore, I do not understand how to use `ring_size` with `size` and `inc_read` or `inc_write`. 

If MicroPython developers understand how to use these parameters in MicroPython, then it is quite acceptable to provide better understanding than a weak area of the RP2040 Datasheet. Please share insightful information with users of MicroPython if you have it to share.

sk8board, 2024-10-30:
> > `DMA.registers[14:16]` is provided in the example. Why was 16 used if the highest available value is 15?
> 
> This is standard Python!
> 
> ```
> >>> a = "0123456789abcdef"
> >>> a[14:15]
> 'e'
> >>> a[14:16]
> 'ef'
> ```
> 
> The example you are referring to is copying new values into the last _two_ registers.

Thank you for your reply. I believe the confusion is around this language.

[DMA.registers](https://docs.micropython.org/en/latest/library/rp2.DMA.html#rp2.DMA.registers)
`This attribute is an array-like object that allows direct access to the DMA channel’s registers. The index is by word, rather than by byte, so the register indices are the register address offsets divided by 4. See the RP2040 data sheet for register details.`

According to the RP2040 Datasheet, the highest register offset is 60. If divided by 4, then 15 would be used with `DMA.registers` to write to the last register. This is why the use of 16 in the example can cause confusion.

Which of these should be used when writing to only the last offset, `write=DMA.registers[15]` or `write=DMA.registers[15:15]` or `write=DMA.registers[15:16]`?

FYI: I created [a single example](https://github.com/sk8board/DCC_MicroPython_Decoder_for_Pico_RP2040/blob/8dd3c756d14783fdd269c2ac39e10398bc0f5dc9/DCC.py#L110-L193) which uses both of the applications that I mentioned above. dma0 and dma1 are chained in parallel "ping-pong" while dma2 and dma3 are chained in series which requires dma3 to write to the last offset of dma2 register.

fmafma, 2024-11-02:
> So, maybe a mecanism to do that whould be more clear. What about a `register_addr[n]`, returning the address of that register, so we can give it to the `write` param?

I think `register_addr(n)` would be better, as it hasn't to be an array-like.

nickovs, 2024-11-02:
> Which of these should be used when writing to only the last offset, write=DMA.registers[15] or write=DMA.registers[15:15] or write=DMA.registers[15:16]?

Either of the later two will give you the correct address, the first will not because it will read the value from the register instead.

This simply comes down to standard Python semantics; the first one **reads the value** from the entry 15 places on from the start of the memory view, while both of the later two **return new views on memory** that start 15 places from the start of the original memory view. The first of these later two happens to be a zero-length view, so you can't read any values out of it, but it still has an address. The behaviour is exactly the same as for any other memory view.

At the moment there is no special-casing whatsoever for the register bank. It is just another `memoryview` object. The `DMA.read` and `DMA.write` fields accept any sort of object that supports the MicroPython equivalent of the standard Python [buffer protocol](https://docs.python.org/3/c-api/buffer.html); this means you can pass `bytes` or `bytearray`s or `array`s or `memoryview`s or anything else that supports the buffer protocol. Any sort of view onto memory inherently **has an address**. When you index a block of memory you are reading from (or writing to) that memory and the value itself **does not have an address**. 

It appears that there needs to be some more clarification in the documentation, but I don't see any reason for any change in behaviour or adding new, special cases that make the registers behave differently to any other view onto a block of memory. The register bank is a block of memory and it should behave like one.

sk8board, 2024-11-05:
> It appears that there needs to be some more clarification in the documentation, but I don't see any reason for any change in behaviour or adding new, special cases that make the registers behave [...]

If you would like, I could make an attempt at this documentation change by submitting a PR. 

I created a thorough DMA example that is likely overkill, but should support most use cases.

``` py
###
# This is an example of two common DMA use cases for RP2040
#
# 1) The first use case shows how to configure dma0 and dma1
# in parallel as described in the RP2040 Datasheet as "ping-pong".
# dma 0 and dma1 will perform the same work, but alternate their
# time of work by triggering the other dma after completing
# a data transfer.
#
# 2) The second use case shows how to configure dma2 and dma3
# in series, where dma2 performs the data transfer and
# dma3 writes to dma2 register as a restart trigger to dma2.
#
# 3) dma2 uses an interrupt handler to timely print data
# in the serial monitor.
#
# This example will print three rows of 32-bit random numbers
#
# This was made using MicroPython version v1.23
###

import rp2, time, random, array, uctypes

data = array.array('L', [0,0])

# generate random bit then put the bit into State Machine 0
def random_bits():
    for i in range (103):
        x = random.getrandbits(1)
        sm0.put(x)
        time.sleep_us(100)
######

# State Machine 0 passes the bit to DMA 0 or DMA 1
@rp2.asm_pio(set_init=rp2.PIO.IN_HIGH, in_shiftdir=0, out_shiftdir=1, autopull=True, pull_thresh=1, autopush=True, push_thresh=1)    
def pass_bit():
    wrap_target()
    out(x, 1)
    in_(x, 1)
    wrap()

sm0 = rp2.StateMachine(0, pass_bit)
sm0.active(1)
######

# DMA 0 and DMA 1 are chained together in parallel (ping-pong per RP2040 datasheet) 
# DMA 0 and DMA 1 alternate transfer of a bit from State Machine 0 to State Machine 1
# After DMA 0 completes the transfer, DMA 0 will trigger DMA 1 using the "chain_to" parameter
# Then DMA 1 does the same to DMA 0. These two DMA's alternate their data transfer and trigger
# responsibilities.
dma0 = rp2.DMA()
dma1 = rp2.DMA()

dma0_ctrl = dma0.pack_ctrl(
    enable = True,          # enable DMA channel
    high_pri = True,        # set DMA bus traffic priority as high
    size = 0,               # Transfer size: 0=byte, 1=half word, 2=word (default: 2)
    inc_read = False,       # do not increment to read address
    inc_write = False,      # do not increment the write address
    ring_size = 0,          # increment size is zero
    ring_sel = False,       # not used since ring_sel is zero
    treq_sel = 4,           # select transfer rate of PIO0 RX FIFO, DREQ_PIO0_RX0
    irq_quiet = True,       # do not generate an interrupt after transfer is complete
[... 123 lines cut ...]
                          ctrl=dma3_ctrl,
                          trigger=False)

def dma2_irq_handler(dma2):
    print(f"{data[0]:032b}",f"{data[1]:032b}")

dma2.irq(handler=dma2_irq_handler, hard=False)  # dma2 will trigger the handler after dma2 completes two transfers (count=2)
######

# start random bits sent to State Machine 0
print("            data[0]         ","               data[1]")
random_bits()

# close DMA's so RP2040 can run this again without needing a hard reset
sm0.active(0)
sm1.active(0)
dma0.active(0)
dma1.active(0)
dma2.active(0)
dma3.active(0)
dma0.close()
dma1.close()
dma2.close()
dma3.close()
```

fmafma, 2024-11-19:
I'm trying to reload the source address of a DMA, using the `registers` attribute, from the interrupt routine of another DMA (triggered when this DMA receives a null trigger, as explained in 2.5.4. Interrupts, when `irq_quiet` is set to `True`). But using either:

```
self._dmaFrame.registers[15] = self._frame
```

or:

```
self._dmaFrame.registers[15] = uctypes.addressof(self._frame)
```

lead to a MemoryError!

On the other hand, using:

```
machine.mem32[0x50000000+0x03c] = uctypes.addressof(self._frame)
```

works fine.

I'm really missing the use case of the `registers` attribute...

nickovs, 2024-11-19:
Without knowing the code that sets up `self._dmaFrame` or `self._frame` it's rather hard to say what the issue is.

nickovs, 2024-11-19:
Thanks. Are you using an RP2040 or an RP2350 (Pico 2)?

Also, could you please print out `uctypes.addressof(self._dmaFrame.registers)` and `self._dmaFrame.channel`?

fmafma, 2024-11-19:
It's a RP2040.
`uctypes.addressof(self._dmaFrame.registers): 0x50000000`
`self._dmaFrame.channel: 0`

nickovs, 2024-11-26:
@fmafma Having spent some time failing to reproduce your bug, I came back to it today and realised that you said that you are doing this:

> from the interrupt routine of another DMA

Looking again at your code I see that you set `hard=True` for the IRQ handler. While `hard` interrupts are faster, as detailed in the [MicroPython documentation](https://docs.micropython.org/en/latest/reference/isr_rules.html#creation-of-python-objects), you are not allowed to create any Python objects inside hard interrupt handlers, and that even includes the transient objects used by the language. As the documentation states:

>A consequence of this is that ISR’s can’t use floating point arithmetic; this is because floats are Python objects. Similarly an ISR can’t append an item to a list. In practice it can be hard to determine exactly which code constructs will attempt to perform memory allocation and provoke an error message: another reason for keeping ISR code short and simple.

The error you are getting is a consequence of this, since the `registers` object is created on the fly.

> I'm really missing the use case of the registers attribute...

The *use case* of this is specifically to make chaining code more readable. It is *not* expected that users will directly assign values into the register bank, since doing so obscures the meaning. When you're doing chaining you don't really have much choice than to dive deep into the guts of the register bank; in that case the `registers` attribute gives you a way to find the register bank without needing to know which channel you're on, how the DMA registers are mapped in the chip or even how many DMA channels there are. For what you want to do a better and clearer (and safer in an IRQ handler) way to write this would be:
```
self._dmaFrame.config(read=self._frame, trigger=True)
```
Aside from the fact that this will work from inside a hard interrupt routine, when compared to:
```
self._dmaFrame.registers[15] = self._frame
```
it's a great deal easier for someone reading your code (and even for you reading your code, in three years time) to see what the intent is.

fmafma, 2024-11-27:
> [...] since the registers object is created on the fly.

Ah, that something that needs to be documented! I'm aware of the object creation limitation in ISR, but I though the `registers` attributes was a standard class attribute...

> `self._dmaFrame.config(read=self._frame, trigger=True)`

My bad, I missed the point I was able to recall the `config()` method. Thanks!

But I'm still not convinced having `registers` an dynamic array-like is really usefull, as for chaining we need the **address** of the first register we want to write to. If there is no real use case to write to individual registers from Python code, I would prefer to have a real not dynamic `register_address` attribute array, so we don't need to bother with a slice in `config()` method, and we could be able to use it from an ISR.

nickovs, 2024-11-27:
> > [...] since the registers object is created on the fly.
> 
> Ah, that something that needs to be documented! I'm aware of the object creation limitation in ISR, but I though the `registers` attributes was a standard class attribute...

A great many things in Python and MicroPython are created on the fly, including many (if not most) properties that are not basic types. I can look at creating this at construction time; there would be a memory overhead to doing that and I need to work out how much overhead that will be before making such a change. Most of the time `registers` are used during the setup process but are not needed later, so making them when needed seemed to make more sense.

Note that it's easy enough to deal with though; you could just make the definition of the interrupt handle a closure that references some local variable into which you've placed the value of the `registers` attribute.

> > `self._dmaFrame.config(read=self._frame, trigger=True)`
> 
> My bad, I missed the point I was able to recall the `config()` method. Thanks!

You're welcome!
 
> But I'm still not convinced having `registers` an dynamic array-like is really usefull, as for chaining we need the **address** of the first register we want to write to. If there is no real use [...]

The class could give you an address, but then you'd have to add an offset instead of taking a slice, and for that you'd also need to know the register size and spacing. Addresses are not something native to Python, but arrays are. A memory view and the various classes like it *are* the way that Python represents addresses, along with information like the length and field size and stride.

Of course the joy of MicroPython being open source is that if you want it to be different you're welcome to write a patch and submit a pull request!