micropython/micropython #16088: Checksum Verification for nRF BLE Stack Downloads

nmccamish, 2024-10-26:
### Description

[The download script](https://github.com/micropython/micropython/blob/master/ports/nrf/drivers/bluetooth/download_ble_stack.sh) for the nRF BLE stack does not perform checksum verification of any of the downloads. Obviously the checksums would have to be updated every time we bump the stack versions needed, but this could add some extra safety against, for instance, botched downloads.

In addition, much of the bash code is duplicative, and could be refactored.

### Code Size

As the `sha256sum` etc commands are part of coreutils, I do not see this adding any additional dependencies, plus the refactoring should result in less code size / maintenance costs.

### Implementation

I intend to implement this feature and would submit a Pull Request if desirable

### Code of Conduct

Yes, I agree

andrewleech, 2024-10-27:
There's been a desire to move away from this script and the legacy softdevice stack as it's no longer maintained by Nordic. 

Also the Bluetooth support on nrf currently is not compatible with the newer BLE api used on other micropython ports or with aioble.

The alternatives which are both in progress (slowly) are to: 
* Migrate this port to use the nimble BLE stack (same as used on stm and esp) https://github.com/micropython/micropython/pull/8318
* Jump over to using the zephyr port (which is basically what Nordic do these days) 

My PR to use nimble is basically complete, but needs to be rebased and re-tested since some other large updates have been recently made to the nrf port. I haven't been using nrf chips much lately though so haven't had much time to dedicate to getting it finished.

patfrench, 2025-03-19:
Hello @andrewleech, 

I have a custom board based on NRF52840 but in QFN48 pin version (there is no USB).
I would like to know how to test the NRF52840_DONGLE port ?