micropython/micropython #16107: Attempting to use SPI(6) on STM32H747BIT6 does not seem to work.

picarro-rdorfner, 2024-10-29:
### Port, board and/or hardware

I am using a custom board with STM32H747BIT6

### MicroPython version

Micropython 1.23 on stm32h747BIT6 using a custom board.

### Reproduction

Spi device is created with..
spi1 = SPI(6, baudrate=1000000, polarity=0, phase=0, bits=8  )

later attempt to write to this device generates a timeout.

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "main.py", line 112, in <module>
  File "valve_control.py", line 84, in __init__
  File "valve_control.py", line 148, in drv_spi_write_buf
  File "valve_control.py", line 131, in drv_spi_write_reg
OSError: [Errno 110] ETIMEDOUT

So the device appears to be there, I have defined thee pins in mpconfigboard.h

#define MICROPY_HW_SPI6_SCK         (pyb_pin_SPI6_SCK)
#define MICROPY_HW_SPI6_MISO        (pyb_pin_SPI6_MISO)
#define MICROPY_HW_SPI6_MOSI        (pyb_pin_SPI6_MOSI)

### Expected behaviour

I was expecting micropython to perform the write to the SPI6 port on the stm32h747BIT6 device.

### Observed behaviour

However, I get the following instead.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "main.py", line 112, in <module>
  File "valve_control.py", line 84, in __init__
  File "valve_control.py", line 148, in drv_spi_write_buf
  File "valve_control.py", line 131, in drv_spi_write_reg
OSError: [Errno 110] ETIMEDOUT

It is almost as if the pins are not actually being configured for output via the spi, like the alternate function register is not being set up?

I'm guessing this might be a bug in configuration management when micropython starts up, but so far I've not been able to find any documentation that indicates the correct way to set up a new hardware spi port.. The code appears to be there for SPI device 6. It just does not seem to actually function, though.

### Additional Information

No, I've provided everything above.

Also, I THINK this is a bug, but it might be that I need to posts this some where else, if so, please provide me with a whack from a clue by four so I can put this in the right place.

Thanks!

### Code of Conduct

Yes, I agree

andrewleech, 2024-10-29:
The OSError is almost certainly being raised from here: https://github.com/micropython/micropython/blob/9ea8d2a03181681feebaede3add82ec4970e30d7/ports/stm32/spi.c#L673C9-L673C21

This is after the DMA is being used to manage sending the data over SPI, so timeout kind of makes sense there as a possible kind of error.
Is it possible for you to test any of the other SPI channels to see if they work?
Perhaps the dma configuration for that chip / spi channel isn't quite right, it would be worth a sanity check / ensure no conflicts of the settings for `dma_SPI_6_TX ` and `dma_SPI_6_RX` if you can figure out which definition in this file applies to you:  https://github.com/micropython/micropython/blob/master/ports/stm32/dma.c

Also from the repl, after you've configured the SPI but before trying to transfer anything you could just check the pin configuraion to ensure they look correct, eg. `print(machine.Pin.board.SPI6_SCK))` for each of them and see if they're in alternate mode for spi.

picarro-rdorfner, 2024-10-30:
Thank you for your help! Truly appreciated.

SO super clue number one is that those pins are NOT defined at all.  I inserted a print(dir(Pin.board)) and got a TONE of pins, but no SPI6_anything at all..

A few questions come to mind, are the pin names presupposedly a certain name? like SPI6_SCK ? so when I add them to pins.csv file they should have a certain name?  I have been simply duplicating the names used for SPI1 and SPI5 and just assigning them the pin id, but those apparently are not showing up at all. I’m not even seeing the pins assigned to SPI1 and SPI5.

For what it’s worth, we designed the board to follow the Arduino GIGA r1 minus the bga package and minus the sdram. The package changed, so I commented out the sdram defines and modified the pin assignments to reflect the custom board. However, it is looking like my assignments are not being picked up for some reason.

I’ve got a lot of experience in embedded C, but I’m just , for the first time starting to dig into the micropython interpreter.. and.. there is a LOT of stuff going on in there. Is the only documentation for the micropython internals in the github comments and forums? Or is there some other source of documentation for the inner workings of the interpreter?

In the mean time I shall try and figure out why my pin assignments are not being built into the interpreter for my board.

Thanks again!
Rick

andrewleech, 2024-10-30:
Ah yes, I actually only recently found out about the - in the CSV myself! They're intended to provide flexibility to describe pins by name in the C headers but then not expose pins to python if they're really not needed there; typically the spi pin names are not going to be needed in python if the hardware spi is used! 
Not exposing them in python saves since flash space.

picarro-rdorfner, 2024-10-31:
Oh! Interesting. And yes, I figured the spi device creation would do “All the right things” with the pin names and assignments and mapping via the AF registers..  And all of that does seem to be correct.  I’ve been trying to get the dma portion understood to see if maybe that’s where the problem is, but so far it’s been a bit of a puzzle. I’m too used to using an IDE and setting break points and then dumping registers that I need to examine, but micropython has been a learningcurve for me. I’m lacking a TON of tools (meaning understanding what tools I have available) to do any real low level debugging. So it’s been a bit of an uphill climb for me.

andrewleech, 2024-11-01:
Oh yeah it's a pretty steep ramp up certainly. 
I definitely use a C debugger for this of stuff myself, my go-to toolchain these days is vscode with the cortex-debug extension. 

For stm work I usually use either a stlink V3 or a jlink from work, though I also want to like my  cmsis-dap / daplink debuggers but they're a bit slower for me. 

There's also a probe-rs tool / extension that's even easier to set to in vscode, though I haven't used it as much yet.

To build with debug symbols you add DEBUG=1 on the end of the make command.

picarro-rdorfner, 2024-11-01:
Oh nice! So adding the suymbols should help me with setting breakpoints in areas that I need to and not have to look. Up the address manually. Does VSCODE know how to set those breakpoints with the stlink ?  If so, I know what I’ll be doing today. 😊

Thanks for the help!
Rick

picarro-rdorfner, 2024-11-01:
I noticed today that the SPI1 can ALSO be made to use PA5,6 and 7 on the stm32h747BIT6 chip.. I tried that today and found that it works without issue. For now, that’s the route I’ll be taking for this problem, however, I still think that SPI6 needs to be able to work with those pins as well

Thanks
Rick