micropython/micropython #16115: LittleFS2 upgrade crashing micropython

victorallume, 2024-10-30:
### Port, board and/or hardware

stm32 port PYBD_SF6 board

### MicroPython version

MicroPython v1.18 on 2022-01-17; PYBD-SF6W with STM32F767IIK 
and 
MicroPython v1.23.0 on 2024-06-02; PYBD-SF6W with STM32F767IIK

### Reproduction

```
#flash to v1.18
import pyb, os
b = pyb.Flash(start=0)
os.VfsLfs2.mkfs(b)

#soft reboot
import pyb, os
f=open('/flash/foo','w')
f.write('qwe')
f.close()
os.listdir('/flash')
open('/flash/foo').read()

#soft reboot
import pyb, os
os.listdir('/flash')
open('/flash/foo').read()

pyb.bootloader()

#flash to v1.23

import pyb, os
os.listdir('/flash')
open('/flash/foo').read()

f=open('/flash/bar','w')
f.write('abc')
f.close()
os.listdir('/flash')
open('/flash/bar').read()
open('/flash/foo').read()

#soft reboot
```

### Expected behaviour

Micropython upgrades the littlefs filesystem, and reboots without crashing

### Observed behaviour

```
MicroPython v1.18 on 2022-01-17; PYBD-SF6W with STM32F767IIK
Type "help()" for more information.
>>> import pyb, os
>>> f=open('/flash/foo','w')
>>> f.write('qwe')
3
>>> f.close()
>>> os.listdir('/flash')
['foo']
>>> open('/flash/foo').read()
'qwe'
>>> 

# [Ctrl-D]

MPY: sync filesystems
MPY: soft reboot
MicroPython v1.18 on 2022-01-17; PYBD-SF6W with STM32F767IIK
Type "help()" for more information.
>>> import pyb, os
>>> os.listdir('/flash')
['foo']
>>> open('/flash/foo').read()
'qwe'
>>> pyb.bootloader()
device disconnected

# Flash micropython 1.23

Connected to MicroPython at /dev/ttyACM0
Use Ctrl-] or Ctrl-x to exit this shell
MicroPython v1.23.0 on 2024-06-02; PYBD-SF6W with STM32F767IIK
Type "help()" for more information.
>>> import pyb, os
>>> os.listdir('/flash')
['foo']
>>> open('/flash/foo').read()
'qwe'
>>> f=open('/flash/bar','w')
>>> f.write('abc')
3
>>> f.close()
>>> os.listdir('/flash')
['bar', 'foo']
>>> open('/flash/bar').read()
'abc'
>>> open('/flash/foo').read()
'qwe'
>>> 

# [Ctrl-D]

MPY: sync filesystems
MPY: soft reboot
FATAL: uncaught exception 2001ba60
ValueError: 

FATAL ERROR:
```
At which point the Pyboard cycles through LED colours and is otherwise unresponsive. If I reset the filesystem using the button combo, I can get back to a REPL.

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

dpgeorge, 2024-10-31:
Thanks for the clear report.  I can reproduce the issue using the above steps.

The issue is very similar to #11944, although in the case here the filesystem is not corrupt it's just detected incorrectly.

Here is a diff of the binary data of the before (good) and after (crashing) LFS2 filesystem:
```diff
1,9c1,9
< 00000000: a261 0070 f00f fff7 6c69 7474 6c65 6673  .a.p....littlefs
< 00000010: 2fe0 0010 0000 0200 0010 0000 0002 0000  /...............
< 00000020: ff00 0000 ffff ff7f fe03 0000 701f fc08  ............p...
< 00000030: ad34 ddba ffff ffff ffff ffff ffff ffff  .4..............
< 00000040: ffff ffff ffff ffff ffff ffff ffff ffff  ................
< 00000050: ffff ffff ffff ffff ffff ffff ffff ffff  ................
< 00000060: ffff ffff ffff ffff ffff ffff ffff ffff  ................
< 00000070: ffff ffff ffff ffff ffff ffff ffff ffff  ................
< 00000080: ffff ffff ffff ffff ffff ffff ffff ffff  ................
---
> 00000000: a461 0070 f00f fff7 6c69 7474 6c65 6673  .a.p....littlefs
> 00000010: 0fe0 040b 666f 6f20 0000 0071 7765 1000  ....foo ...qwe..
> 00000020: 000b 80e4 1a44 811a b513 1000 0410 0100  .....D..........
> 00000030: 0200 0010 0000 0002 0000 ff00 0000 ffff  ................
> 00000040: ff7f fe03 0000 7fef fc10 2000 0000 f454  .......... ....T
> 00000050: 9300 0ff0 0002 e318 0a2c ffff ffff ffff  .........,......
> 00000060: 101f f80a 4000 0003 6261 7220 0000 037f  ....@...bar ....
> 00000070: eff8 0820 0000 00f4 5493 000f f000 295d  ... ....T.....)]
> 00000080: 91d7 a7ff ffff ffff ffff ffff ffff ffff  ................
11,13c11,13
< 000000a0: ffff ffff ffff ffff ffff ffff ffff ffff  ................
< 000000b0: ffff ffff ffff ffff ffff ffff ffff ffff  ................
< 000000c0: ffff ffff ffff ffff ffff ffff ffff ffff  ................
---
> 000000a0: 701f f822 6162 6310 0000 0b00 fecc 79af  p.."abc.......y.
> 000000b0: 1ab5 136f eff8 0020 0000 00f4 5493 000f  ...o... ....T...
> 000000c0: f000 15e2 5fa4 07ff ffff ffff ffff ffff  ...._...........
```
The reason the code crashes is because the first block is no longer a valid superblock even though it still has the "littlefs" tag.   The block size and number of blocks are overwritten with the data from the foo file and the code that auto-detects the filesystem uses this bad data and tries to construct a `pyb.Flash()` object that has out-of-bounds size.  That raises an exception, which is uncaught.

Note, this crash is not related to an upgrade of the LFS on-disk data from 2.0 to 2.1 (which was my initial guess).

Suggested fix:
- apply #11944; that will at least allow the board to boot up (without a filesystem)
- fix the littlefs auto-detection algorithm so it can handle the case of the first block not being a superblock even if it has the "littlefs" string in it; that will allow this filesystem to be correctly mounted

dpgeorge, 2024-11-01:
Upon further investigation of this issue I found the following things:
- the same bug is present when going from MicroPython 1.21.0 to 1.22.0 (and that's exactly when littlefs was updated to support the new on-disk version)
- the crashing filesystem is actually a littlefs 2.1 filesystem and is tagged as such in the superblock data

The cause of the crash and suggested fix is still the same, but let me refine how I think it should be fixed:
1. Mounting of the initial filesystem should be protected against raising exceptions in the case of errors (and the approach in #11944 looks good for that).
2. The littlefs auto-detection algorithm in stm32 needs to be fixed/improved to properly detect a superblock and search for its metadata.

RachidEddahaPrivate, 2024-11-12:
This bug also occurs on my device running MicroPython 1.22.* on the ESP32 port, which uses littlefs as the VFS.
The filesystem loads correctly on my PC, but the same filesystem appears corrupted on the ESP32.
After investigating, it seems that the issue is due to the superblock not being correctly detected.

 This suggests the error may not be isolated to the STM32 port and could affect other ports using littlefs.

dpgeorge, 2024-11-12:
@RachidEddahaPrivate the bug originally reported here is definitely due to the littlefs auto detection code in `ports/stm32/main.c`.  The issue you see must be due to some other bug.  Can you please describe your bug in more detail, ideally with a full reproduction, or at least with a dump of a "broken" filesystem.  Also, how is your filesystem being mounted, does it just use the default code in `_boot.py` or some custom code?

RachidEddahaPrivate, 2024-11-12:
Unfortunately, the filesystem contains sensitive information, so I cannot share a direct dump.
Regarding the setup, our filesystem uses the default code in `_boot.py`, but we made some modifications to the partition table as follows:
```plaintext
nvs,      data, nvs,     0x9000,   0x4000,
otadata,  data, ota,     0xd000,   0x2000,
phy_init, data, phy,     0xf000,   0x1000,
ota_0,    app,  ota_0,   0x10000,  0x1e0000,
ota_1,    app,  ota_1,   0x1f0000, 0x1e0000,
vfs,      data, fat,     0x3d0000, 0x020000
```
To bypass the issue, I modified the following line in [vfs.c](https://github.com/micropython/micropython/blob/3844733d604d83fbfec5592cee20aab065ebff48/extmod/vfs.c#L182), removing the conditional check that determines if the filesystem is LittleFS. This workaround allowed me to read the previously "broken" filesystem dump correctly, suggesting the problem may lie in the filesystem detection logic for LittleFS.

dpgeorge, 2024-11-13:
@RachidEddahaPrivate thanks for the extra info.  Are you able to post the first 16 bytes of each 4096-byte block of the littlefs2 filesystem?  That should hopefully be enough to debug the issue you see.

victorallume, 2024-11-13:
We have another bit of strangeness with LFS. I've not been able to reproduce it with a stock micropython build unfortunately, but will describe the situation and process in case it's relevant to this.

- Pyboard has 3 partitions; the third one ('/data') is an LFS2 partition (which is the one I've been testing on). 
- Micropython is compiled to _not_ mount anything on boot, as the partitions are mounted in our code. 
- /data gets populated with three folders on boot, one of these folders has some files in it. 
- Initially, the pyboard is running micropython v1.18
- I create a file `/data/foo` and write the string `abc`
- Now I upgrade to micropython v1.23
- The file `/data/foo` exists and is readable
- I delete `/data/foo`
- I create `/data/bar` and write the string `qwe` to that
- `os.listdir()` shows that `/data/foo` doesn't exist, and `/data/bar` does; I can read `/data/bar` and it has `qwe` as its contents (as you would expect)
- Now i flash micropython v1.18 back on
- `os.listdir()` shows that `/data/foo` _exists_ and `/data/bar` _does not_; I can read `/data/foo` and it has `abc` as its contents - I would expect the mount to have failed.

So the old filesystem and new one seem to be co-existing on the same partition. I've attached the first 350 bytes of the upgraded partition; the rest is empty. Presumably since the files are small, there's no data being over-written, and the filesystem auto-detection code is incorrect as previously suggested. However, this is not happening during the boot process, but rather in our own code, so I'm wondering if this is actually an issue beyond the stm32 port? It is the old micropython (1.18) which is playing up here.

[pyb_upgraded_lfs_data_partition_start.bin.gz](https://github.com/user-attachments/files/17725921/pyb_upgraded_lfs_data_partition_start.bin.gz)

dpgeorge, 2024-11-13:
Thanks @victorallume for the additional test and the details to reproduce it.

I can myself reproduce the same issue using unix MicroPython (1.18 and current master), and using a block size of 256 bytes and a filesystem size of 512 bytes (two blocks).  That gives a very simple reproduction.

dpgeorge, 2024-11-13:
@victorallume further debugging the issue described just above in https://github.com/micropython/micropython/issues/16115#issuecomment-2472327650 I conclude the following:
1. micropython-1.18 / littlefs 2.0 creates a filesystem and writes a file; block 0 and 1 are superblocks but it's block 1 that is the most recent one and contains the file
2. micropython-1.23 / littlefs 2.1 updates this filesystem with a new file, creating an updated superblock 0 with the new filesystem version and the new file; the old superblock 1 is left as-is
3. micropython-1.18 / littlefs 2.0 reads the new filesystem, sees superblock 0 as corrupt because it doesn't understand the format, and so considers superblock 1 as the most up-to-date block of the pair (0,1), and this block has the old file

The main point is, in step 3, the old littlefs 2.0 driver does not even see the update 2.1 version number so cannot report that the version is unsupported.  Instead it thinks it's a 2.0 filesystem with a corrupt superblock 0, eg due to power failure during a write of this block.  Then it happily parses block 1 as a valid 2.0 block because that was unchanged and is in fact still a valid 2.0 superblock.

Littlefs 2.0 cannot tell the difference between an updated filesystem and a partially corrupt 2.0 filesystem.  They look the same to it.

So I think what you are experiencing is expected behaviour for littlefs 2.

victorallume, 2024-11-13:
Thanks. This is probably outside the scope of micropython and more in littlefs territory, but do we know if there is a reason the upgrade only upgrades the first superblock? And does the updated littlefs handle an upgrade situation any better, or will this be a thing for the next upgrade as well?

RachidEddahaPrivate, 2024-11-14:
@dpgeorge Sure, I've attached the first 16 bytes of each 4096-byte block from the littlefs2 filesystem for reference. I've also included the data here for quick viewing:
```plaintext
3b 0c  00 00  9f e0  03 f7  14 00  00 00  16 00  00 00      ;...............
3c 0c  00 00  9f e0  03 f7  0c 00  00 00  0f 00  00 00      <...............
11 00  00 00  0d 00  00 00  61 70  70 20  32 2e  31 2e      ........app 2.1.
1d 00  00 00  6c 65  20 22  77 69  66 69  2e 70  79 22      ....le "wifi.py"
10 00  00 00  20 22  77 69  66 69  2e 70  79 22  2c 20      .... "wifi.py", 
03 00  00 00  1d 00  00 00  45 54  29 0a  32 30  30 30      ........ET).2000
11 00  00 00  0d 00  00 00  61 70  70 20  32 2e  31 2e      ........app 2.1.
8c 62  30 30  f0 0f  ff f7  6c 69  74 74  6c 65  66 73      .b00....littlefs
1b 00  00 00  29 0a  32 30  30 30  2d 30  33 2d  31 31      ....).2000-03-11
7b 22  6c 6f  6e 65  6c 79  5f 72  65 73  65 74  5f 74      {"lonely_reset_t
1a 00  00 00  61 69  6e 2e  70 79  22 2c  20 6c  69 6e      ....ain.py", lin
08 00  00 00  1b 00  00 00  10 00  00 00  1a 00  00 00      ................
8e 62  30 30  f0 0f  ff f7  6c 69  74 74  6c 65  66 73      .b00....littlefs
16 00  00 00  05 00  00 00  1d 00  00 00  20 36  34 2c      ............ 64,
11 00  00 00  0d 00  00 00  61 70  70 20  32 2e  31 2e      ........app 2.1.
8d 62  30 30  f0 0f  ff f7  6c 69  74 74  6c 65  66 73      .b00....littlefs
19 00  00 00  15 00  00 00  1a 00  00 00  6c 61  73 74      ............last
0d 00  00 00  20 32  2e 31  2e 30  2e 2e  2e 20  28 72      .... 2.1.0... (r
11 00  00 00  0d 00  00 00  61 70  70 20  32 2e  31 2e      ........app 2.1.
11 00  00 00  0d 00  00 00  61 70  70 20  32 2e  31 2e      ........app 2.1.
11 00  00 00  0d 00  00 00  61 70  70 20  32 2e  31 2e      ........app 2.1.
0a 00  00 00  1a 00  00 00  20 5f  63 6f  6e 6e  65 63      ........ _connec
05 00  00 00  6e 65  63 74  20 74  6f 20  6d 6d  70 0a      ....nect to mmp.
11 00  00 00  0d 00  00 00  61 70  70 20  32 2e  31 2e      ........app 2.1.
11 00  00 00  0d 00  00 00  61 70  70 20  32 2e  31 2e      ........app 2.1.
15 00  00 00  72 74  69 6e  67 20  43 53  20 61  70 70      ....rting CS app
32 30  30 30  2d 30  33 2d  31 31  20 30  39 3a  30 37      2000-03-11 09:07
04 00  00 00  10 00  00 00  2d 31  31 20  31 37  3a 32      ........-11 17:2
23 20  54 68  69 73  20 66  69 6c  65 20  69 73  20 65      # This file is e
32 30  30 30  2d 30  33 2d  31 31  20 32  30 3a  30 30      2000-03-11 20:00
11 00  00 00  0d 00  00 00  61 70  70 20  32 2e  31 2e      ........app 2.1.
11 00  00 00  0d 00  00 00  61 70  70 20  32 2e  31 2e      ........app 2.1.
```
[lfs_first_16_bytes_of_all_sectors.txt](https://github.com/user-attachments/files/17747451/lfs_first_16_bytes_of_all_sectors.txt)

dpgeorge, 2025-11-18:
I'll close this assuming it's fixed.  Feel free to reopen with additional information if it's not.

closed 2025-11-18 by dpgeorge