micropython/micropython #16129: _thread.stack_size(xxxx) not working on ESP32-S3

sebert007, 2024-11-01:
### Port, board and/or hardware

esp32, thread

### MicroPython version

MicroPython v1.24.0-dirty on 2024-10-31; Generic ESP32S3 module with ESP32S3

It's basically the official v1.24.0 that I compiled on my own using this description here:
https://github.com/micropython/micropython/blob/master/ports/esp32/README.md

### Reproduction

I have a medium complex project on my ESP32-S3 with 2 megs of RAM.
Until now, I was running everything in a main loop. Now I am using the `_thread` module.

I encountered the "maximum recursion dept exceeded" error multiple times. I wondered, since I am not using any extra recursions on my code. This post here brought me on the track that not the number of recursions, but the stack size might actually be the issue:
https://forum.micropython.org/viewtopic.php?f=16&t=3091&start=10

Using
```
_thread.stack_size(64*1024)
```
did not help any further and seems to have no effect at all.

### Expected behaviour

1. Expected an error message "Stack size too small for thread" instead of the message above.

2. Expected the command
```
_thread.stack_size(64*1024)
```
to be working.

### Observed behaviour

I made the following changes to **mpthreadport.c**:
```
define MP_THREAD_MIN_STACK_SIZE                        (4 * 1024)
to
define MP_THREAD_MIN_STACK_SIZE                        (64 * 1024)
```

I made the following changes to **mpconfigport.h**:
```
define MICROPY_TASK_STACK_SIZE             (16 * 1024)
to
define MICROPY_TASK_STACK_SIZE             (64 * 1024)
```

Now everything is working fine, and I don't get the error message anymore.

### Additional Information

If it is not possible to fix the `_thread.stack_size()` issue (e.g. you cannot define the stack size during runtime), could you at least increase the two numbers slightly by default (not necessarily as much as I did, but a little) and clarify the error message?

Do you have any idea, which made my code work? Was it changing `MICROPY_TASK_STACK_SIZE` or changing `MP_THREAD_MIN_STACK_SIZE`?

This issue might be related to all that:
https://github.com/micropython/micropython/pull/12415

### Code of Conduct

Yes, I agree

sebert007, 2024-11-01:
After playing around with the two different values, it seems that only the `MP_THREAD_MIN_STACK_SIZE` really matters. Setting it from the original value
```
MP_THREAD_MIN_STACK_SIZE                        (4 * 1024)
```
to
```
MP_THREAD_MIN_STACK_SIZE                        (16 * 1024)
```
works at least in my case. Running ``gc.mem_free()`` right before starting the thread and right after starting, it shows a memory consumption of about 5kB. So increasing the value for ``MP_THREAD_MIN_STACK_SIZE`` by default should not make the ESP run out of memory so fast.

sosi-deadeye, 2024-11-02:
Recursion is bad in stack based languages like C or Micropython. Use an iterative approach, and yes, this is not easy to approach.

dpgeorge, 2024-11-04:
> Using
> 
> ```
> _thread.stack_size(64*1024)
> ```
>
> did not help any further and seems to have no effect at all.

This should definitely work.  Did you call that function *before* creating the thread?

sebert007, 2024-11-04:
I tried `_thread.stack_size(64*1024)` with the official `v1.24.0` and put it _before_ creating the thread, like I did before.

It is working now as expected. The reason why it didn't work before, seems to be that I put `print(_thread.stack_size())` right after `_thread.stack_size(64*1024) `. This prints the stack size set, but seems to reset the specified stack size to the default value as well. As a consequence, the default stack size and not the one specified before will be used.

Could you change this behaviour? And also could you make the error message more clear? It still confuses that it says `RuntimeError: maximum recursion depth exceeded` although it's a stack size problem.

dpgeorge, 2024-11-04:
> It is working now as expected. The reason why it didn't work before, seems to be that I put `print(_thread.stack_size())` right after `_thread.stack_size(64*1024) `. This prints the stack size set, [...]

Yes, that's expected (and perhaps unfortunate) behaviour. `stack_size()` without any argument will set the stack size to the default.

> Could you change this behaviour?

Unfortunately not.  It's standard CPython behaviour: https://docs.python.org/3/library/_thread.html#thread.stack_size

> And also could you make the error message more clear?

This is also CPython behaviour, to report "maximum recursion depth exceeded" when the call depth gets too large (regardless of whether there is any recursion or not).  That is subtly different to stack overflow, but in MicroPython there's no counter for the number of nested calls and so any stack overflow is just reported as "recursion depth exceeded".

closed 2024-11-04 by sebert007