micropython/micropython #16131: ESP8266 is difficult to obtain an IP address from soft-routing WIFI

yfyungit, 2024-11-02:
### Port, board and/or hardware

esp8266

### MicroPython version

ESP8266_GENERIC-FLASH_1M-20241025-v1.24.0

### Reproduction

wlan = network.WLAN(network.STA_IF)
def connect_static_ip(ssid, password):
    wlan.active(True)
    while not wlan.isconnected():
        print("connecting...", _, wlan.isconnected(), wlan.status())
        time.sleep(1)
    print("IP :", wlan.ifconfig(), wlan.isconnected())

### Expected behaviour

I tried to use esp8266 to connect to wifi and found it difficult to connect, occasionally I can get the ip address

After that, I tried to use ARDUINO for programming, and connected to the same WIFI. I found that every time I connected to WIFI under Arduino, I could get the DHCP IP smoothly and correctly.

### Observed behaviour

I tried to use esp8266 to connect to wifi and found it difficult to connect, occasionally I can get the ip address

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

robert-hh, 2024-11-02:
I can confirm the behavior. ESP8266 device with larger flash work fine. That includes devices with 4Mflash, loaded with the 1M firmware.
A ESP01 with 1M flash shows sporadic connects to the AP. Interestingly, when it happens it may get different IP addresses, even when the IP configuration was set before calling connect(). Once a connection succeeded, it will do so on reconnect until the next power cycle or reset. Tested with:
`MicroPython v1.25.0-preview.10.g0e490b7c8 on 2024-11-02; ESP module (1M) with ESP8266
`

yfyungit, 2024-11-02:
> 我可以确认行为。ESP8266设备与较大的闪存工作正常。这包括设备与4M闪存，加载1M固件。带有1M闪光灯的ESP01显示到AP的零星连接。有趣的是，即使在调用connect()之前设置了IP配置，当发生这种情况时，它也可能获得不同的IP地址。一旦连接成功，它将重新连接，直到下一次重启或重置。使用测试： `MicroPython v1.25.0-preview.10.g0e490b7c8 on 2024-11-02; E

> 我可以确认行为。ESP8266设备与较大的闪存工作正常.这包括设备与4M闪存，加载1M固件.带有1M闪光灯的ESP01显示到AP的零星连接。有趣的是，即使在调用连接()之前设置了IP配置，当发生这种情况时，它也可能获得不同的IP地址.一旦连接成功，它将重新连接，直到下一次重启或重置。使用测试： `MicroPython v-preview.10.g

yfyungit, 2024-11-02:
> I can confirm the behavior. ESP8266 device with larger flash work fine. That includes devices with 4Mflash, loaded with the 1M firmware. A ESP01 with 1M flash shows sporadic connects to the AP. [...]

I used esp01s hardware for the test, which uses esp8266 chip. It is very difficult to connect to WIFI. Can this problem be optimized? Because I tested the same flash 1mb, the WIFI link under Arduino programming is very stable, and the link speed is also very fast. WIFI communication is a basic function and the value of this chip.

yfyungit, 2024-11-04:
@robert-hh I used esp01s hardware for the test, which uses esp8266 chip. It is very difficult to connect to WIFI. Can this problem be optimized? Because I tested the same flash 1mb, the WIFI link under Arduino programming is very stable, and the link speed is also very fast. WIFI communication is a basic function and the value of this chip.

robert-hh, 2024-11-04:
@yfyungit I have seen your response and can confirm the behavior, but I did not look into the code for possible reasons. That may take a while.

yfyungit, 2024-11-04:
> @yfyungit I have seen your response and can confirm the behavior, but I did not look into the code for possible reasons. That may take a while.

Ok, thank you very much

yfyungit, 2024-11-16:
[image]
@robert-hh    I have found the cause of this problem. It seems to be a compatibility issue between WIFI5 and WIFI6. When connecting to the router WIFI in Mircopython, the router must be set to a device mode compatible with WIFI5 so that ESP01S can connect to the router correctly. I have also tested that the AP access point in the iPhone 13 must be turned to the maximum compatible mode so that ESP01S can connect correctly to the AP access point of the iPhone 13. However, under Arduino programming, there is no need to set the compatibility of WIFI5 and WIFI6. The router works in WIFI6 mode, and ESP01S can also connect to the WIFI6 router correctly and quickly. I think if mircopython wants to solve this problem, it needs to update the API to support WIFI6 compatibility.

robert-hh, 2024-11-16:
Thank you for the information. I have two ESP8266 board now which connect not easily. No problem with all others. I did not look further into that topic. WiFi5 vs. WiFi6 is not the issue here. I use an pretty old AP which does not support WiFi6 at all, and I'm not sure if it supports even WiFi5. So it usually works fine with all devices.

yfyungit, 2024-11-17:
> Thank you for the information. I have two ESP8266 board now which connect not easily. No problem with all others. I did not look further into that topic. WiFi5 vs. WiFi6 is not the issue here. I use [...]

The old router WIFI ESP01S is now connected without any problems. I have determined the problem and can now start to complete the rest of the work. I was stuck on this problem before and couldn't proceed. Thank you very much