micropython/micropython #16135: Get machine resource by name in addition to by id

alonbl, 2024-11-02:
### Description

Hello,

It would be useful that similar to `machine.Pin` class, the Led, UART, SPI, I2C and other classes will be able to receive a string identifier, so that the logical name can be acquired.

I apologize if this was considered before, I am new to micropython and looked for similar cases but could not find any.

The logical name may be used to reduce synchronization between application and board settings and to allow the same application run on different boards and board variants without modifications.

For example:

```python
import machine
machine.UART("DEBUG").write("Hello\r\n");
machine.UART("UI").write("Hello\r\n");
machine.LED("OK").on()
machine.LED("CONNECTED").on()
```

This should be backward compatible as we can distinguish between string id and numeric id.

Another option is to provide board specific constants exposed to user as a pattern, in this case we may add a `machine.registry` object and the code will look like:

```python
import machine
machine.UART(machine.registry["UART_DEBUG"]).write("Hello\r\n");
machine.UART(machine.registry["UART_UI"]).write("Hello\r\n");
machine.LED(machine.registry["LEDOK"]).on()
machine.LED(machine.registry["CONNECTED"]).on()
```

The registry approach is less intrusive to current implementation, easier to add, but not as intuitive as expected, and error prone, as board configuration requires to refer the same entity in two different locations.

What do you think?

BTW: it would be handy to have a `machine.board_id()` which will return a machine friendly board class unique identifier similar to `os.uname().machine` human friendly identifier for cases in which this feature is rejected and registry should be constructed outside of the board support.

Thanks,

### Code Size

This is similar to Pin class, I would like to assume that if logical name for pin is acceptable, then logical name for lower number of peripherals is also acceptable.

### Implementation

I intend to implement this feature and would submit a Pull Request if desirable

### Code of Conduct

Yes, I agree

sosi-deadeye, 2024-11-02:
This change would invalidate all existing code. If you require a mapping, then use a dict.

```python
from machine import Pin

pins = {"Out1": Pin(5, mode=Pin.OUT, value=1)}
```

If you want to access the names as attribute, just use a class.

```python
from machine import Pin

class Pins:
    out1 = Pin(5, mode=Pin.OUT, value=1)

```

alonbl, 2024-11-02:
Hi @sosi-deadeye,

Thank you for responding.

I am unsure why it invalidate all existing code, it is backward compatible, 1st argument is currently an integer, if it is a string new functionality (aka lookup) is enabled.

I am unsure how pins provide a solution. How can I define a name for a specific board LED object? how can I define a name for a specific board UART using the pin mechanism? As far as I saw these objects are pre-allocated at boot without ability to provide a label.

Thanks,
Alon

robert-hh, 2024-11-02:
I cannot see how this suggestion makes things simpler or easier. Of course one can use other names or symbols for device numbers instead of numbers or in case of Pins the predefined strings, which are the names printed on the board itself. But that is specific for a certain use. The UART which you call "DEBUG" or "UI", will certainly used by other people for other purposes. So if you want in your application to give devices names, you can do so. Create a dictionary with these names, use the `cons()` mechanism, assign number to symbols, ...... You are free in your personal choices.

alonbl, 2024-11-02:
Hello @robert-hh,

A board may be a generic module or a specific device.

In a generic module, you are correct, most of the devices may be assigned to any purpose, there are exceptions, for example the debug UART or board led.

In specific devices, each pin and port have specific purpose. A series of boards that are compatible may expose the same labels for pins and peripherals allowing the software to run on different variants.

The pin module already have the capability of using labels, one can argue that this is inconsistent, as why not use numeric ids for pins having constants in application?

Per your logic each pin "will contently user by other people for other purposes".

```python
>>> machine.Pin.board.
<snip>
LED2            MCK             RX              SCK
SCK_RX          SCK_TX          SCL             SDA
SDI             SDO             SD_RX           SD_TX
```

What is the difference between this and UART in a specific purpose board?

BSP transformation should be consistent to all resources, shouldn't it?

Thanks,

robert-hh, 2024-11-02:
The pin names in Pin.board follow the printed names on the board. And obviously, a Pin with the name "RX" can be used for other purposes than RX as well.
The usual way to suggest a certain feature in this project is to implement it and make a pull request, and not to ask someone to do that. So if you think that this is useful, go ahead. If it is not accepted, you have it at least for your needs.

alonbl, 2024-11-02:
As written in comment#0:

```
I intend to implement this feature and would submit a Pull Request if desirable
```

I will do the work if the feature is desirable, I would like a maintainer to ACK it before I invest resources and implement a cross-product feature such as this. My intention is not to maintain a fork of micropython or any other project.

Logical names are important and should be consistent. I do not see why we cannot have logical names for all resources. I would love to work with maintainers to make this happen.

robert-hh, 2024-11-02:
So you have to wait for the maintainer to respond.
Out of interest: the two most favorite board families at the moment are RPI Pico and ESP32. Which names would you assign to the UART, I2C, SPI, etc.. devices at the boards? Would that be static or something configurable by the users?

alonbl, 2024-11-02:
Usually, once one finishes to play with modules for prototyping, he may design a custom board to reduce size/power/cost, there may be differences in pin assignment and peripheral assignment.

Once doing so, this feature implementation will make the software easier to maintainer and compatible with both setups, and also future revisions of the product family.

robert-hh, 2024-11-03:
Custom boards require custom builds, which will probably not make it's way into the official distribution. They have to be maintained by the board builders.

alonbl, 2024-11-03:
> Custom boards require custom builds, which will probably not make it's way into the official distribution. They have to be maintained by the board builders.

Correct, but the feature of having logical names for the peripheral should be available as part of the core, aka this feature request. Just line in the pin case, adding new board includes the wiring and labeling, not core functionality.

sosi-deadeye, 2024-11-03:
Just make a wrapper around existing functions to implement your suggestion. Then you can offer this a package and let the devs decide, if they want to use it or not. If your change is made in Micropython itself, it enforces all developers who do not want to use it, disabling this and compiling the firmware by themselves.

Sometimes we have to do this to enable functions like compression. This is by default not in the firmware to save space.

alonbl, 2024-11-03:
Hello @sosi-deadeye,

This change enforces nothing on developers, just like pin aliases micropython already support, if there are none, no alias is added. It is all a matter of having a `char *` somewhere at the peripheral table.

With alias:

```patch
 #define IOMUX_TABLE_UART \
-    { IOMUXC_GPIO_AD_24_LPUART1_TXD }, { IOMUXC_GPIO_AD_25_LPUART1_RXD }, \
-    { IOMUXC_GPIO_DISP_B2_10_LPUART2_TXD }, { IOMUXC_GPIO_DISP_B2_11_LPUART2_RXD }, \
+   "DEBUG", { IOMUXC_GPIO_AD_24_LPUART1_TXD }, { IOMUXC_GPIO_AD_25_LPUART1_RXD }, \
+   "AUX1", { IOMUXC_GPIO_DISP_B2_10_LPUART2_TXD }, { IOMUXC_GPIO_DISP_B2_11_LPUART2_RXD }, \
```

or without alias:

```patch
 #define IOMUX_TABLE_UART \
-    { IOMUXC_GPIO_AD_24_LPUART1_TXD }, { IOMUXC_GPIO_AD_25_LPUART1_RXD }, \
-    { IOMUXC_GPIO_DISP_B2_10_LPUART2_TXD }, { IOMUXC_GPIO_DISP_B2_11_LPUART2_RXD }, \
+   NULL, { IOMUXC_GPIO_AD_24_LPUART1_TXD }, { IOMUXC_GPIO_AD_25_LPUART1_RXD }, \
+   NULL, { IOMUXC_GPIO_DISP_B2_10_LPUART2_TXD }, { IOMUXC_GPIO_DISP_B2_11_LPUART2_RXD }, \
```

or we can support this using codegen similar to `pins.cvs`, aka `peripherals.csv`:

```
UART,1,DEBUG
UART,2,AUX1
```

Comparing this to compression or any functional code is incorrect, I keep reminding here that it is similar to the existing pin alias that is already supported.

Thanks,

jonnor, 2025-01-09:
Using variables or dictionaries in Python seems much more practical. No need to invent a mechanism inside Pin, which then requires codegen and a build step in order to define the names.

alonbl, 2025-01-09:
> Using variables or dictionaries in Python seems much more practical. No need to invent a mechanism inside Pin, which then requires codegen and a build step in order to define the names.

I agree.

Although this feature request is not about pins but about synchronize the peripherals of the BSP with the way pins **behave**, and this does not require codegen as I showed in https://github.com/micropython/micropython/issues/16135#issuecomment-2453446293.

peterhinch, 2025-01-10:
This discussion has me baffled - I don't understand the problem under discussion. 

Take the original example of a set of UARTs. In practice each will need to be instantiated with args (baudrate etc). Consider the following absolutely bog standard Python code:
```py
uart_debug = machine.UART(1, 9600)
uart_ui = machine.UART(6, 38400, parity=0)
uart_debug.write("Hello\n")
```
The customisation is done via the instance name.

alonbl, 2025-01-10:
Hello @peterhinch,

In many cases on has multiple variants of boards which is shared by the same application, newer revisions, different purpose.

Currently, the Pin mechanism enable the board to name each Pin with designated alias using `machine.Pin.board.XXX` this enables the application to be agnostic to the board, for example if a pin of button modified, this concern is hidden as in both machine variants the symbol would be `machine.Pin.board.ACTIVATE_BUTTON` which is mapped to the physical pin.

We do not have this mechanism of board aliases to peripherals, for example if the console UART is modified from, let's say UART0 to UART1, there is no [formal] mechanism for the board port to provide this information to the application, this requires the application to be aware of the change.

This feature request suggest to align the peripheral and Pin alias usages, so that in both environment we can write:

```python
import machine

print(machine.Pin("ACTIVATE_BUTTON").value())
machine.UART("CONSOLE").write("Hello\r\n");
# or:
print(machine.Pin.board.ACTIVATE_BUTTON.vaue())
machine.UART.board.CONSOLE.write("Hello\r\n");
# or whatever 
```

This will provide consistent handling of all board resources for both pins and peripherals.

Thanks,

peterhinch, 2025-01-11:
@alonbl Thanks for the explanation.

My approach to this problem is to have a [config file](https://github.com/peterhinch/micropython-mqtt/blob/master/mqtt_local.py) which detects the platform and instantiates objects accordingly. Alternatively one could have a separate config file for each platform.

It's arguably untidy but localises platform-specific code in a single module. The application uses instance names as an abstraction.

alonbl, 2025-01-11:
Thanks @peterhinch,

Of course I can use configuration file.
I believe there is still inconsistency in the way board resources are exposed in micropython.
There should be a consistent standard/common method to translate symbolic name to resource that works  work all resources.
In my initial comment you can also see a registry approach which is closer to formalizing a configuration, and allow extension.

```python
import machine
machine.UART(machine.registry["UART_DEBUG"]).write("Hello\r\n");
machine.UART(machine.registry["UART_UI"]).write("Hello\r\n");
machine.LED(machine.registry["LEDOK"]).on()
machine.LED(machine.registry["CONNECTED"]).on()
```

I guess I am the only one that is bothered by multiple methods to access resources in a single platform.

Josverl, 2025-01-12:
You state : 
> There should be a consistent standard/common method to translate symbolic name to resource that works work all resources.

I disagree with your objective for multiple reasons :
- it would require a single clear model of all pins, resources periphials and across all boards. In my experience that is hard to get even in a single MCU port and supplier
- the implementation of such a model will likely increase the firmware, and runtime size
- even when model can be defined, i do not believe that it will be simpler to use than a `config.py` file.
- most type checkers and IDEs will offer completion, and even docstrings for config files. While I'm not sure that is possible to implement for the model you propose.

Perhaps a similar goal can be achieved by adding documentation or wiki pages with examples on how config files could be setup and used.

alonbl, 2025-01-13:
@jonnor, 
This thread, including the 1st comment includes several suggestions of implementation, have you read it fully or addressed only the last comment?
Thanks,