micropython/micropython #16137: ESP32C6 WDT reset when using network module

Thecoolpeople, 2024-11-02:
### Port, board and/or hardware

esp32c6

### MicroPython version

MicroPython v1.24.0 on 2024-10-25; ESP32C6 module with ESP32C6

### Reproduction

Case 1:

```
import network
wlan = network.WLAN(network.STA_IF)
wlan.active(True)
wlan.connect('ssid', 'pw')
wlan.ipconfig('addr4')
```

Case 2:

```
import network
ap = network.WLAN(network.AP_IF)
ap.active(True)
ap.config(ssid='esp32')
```

### Expected behaviour

Connect to my WIFI or generate an AP. Both cases fail

### Observed behaviour

Case 1:
```
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x8 (TG1_WDT_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x20000800
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0xed4
load:0x4086c410,len:0xc18
load:0x4086e610,len:0x2b44
entry 0x4086c410
W (24) boot.esp32c6: PRO CPU has been reset by WDT.
```

Case 2:
```
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x8 (TG1_WDT_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4080ea24
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0xed4
load:0x4086c410,len:0xc18
load:0x4086e610,len:0x2b44
entry 0x4086c410
W (24) boot.esp32c6: PRO CPU has been reset by WDT.

```

Case 2, but with other error:
```
A fatal error occurred. The crash dump printed below may be used to help
determine what caused it. If you are not already running the most recent
version of MicroPython, consider upgrading. New versions often fix bugs.

To learn more about how to debug and/or report this crash visit the wiki
page at: https://github.com/micropython/micropython/wiki/ESP32-debugging

MPY version : v1.24.0 on 2024-10-25
IDF version : v5.2.2
Machine     : ESP32C6 module with ESP32C6

Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).

Core  0 register dump:
MEPC    : 0x4080ea28  RA      : 0x40812b7a  SP      : 0x40826580  GP      : 0x40818e34
TP      : 0x407e6ea4  T0      : 0x40022494  T1      : 0x4080ebda  T2      : 0x00000000
S0/FP   : 0x40825e70  S1      : 0x40822000  A0      : 0x00000001  A1      : 0x00000001
A2      : 0x00000004  A3      : 0x00000001  A4      : 0x600c2000  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x00000000  S2      : 0x40822000  S3      : 0x4081bc70
S4      : 0x00000001  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x00000000  T4      : 0x00000000  T5      : 0x00000000  T6      : 0x00000000
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000018  MTVAL   : 0x00000000
MHARTID : 0x00000000

Stack memory:
40826580: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
408265a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
408265c0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000158 0x40826500 0x00000000 0x4081bcc8 0x4081bcc8
408265e0: 0x408265d0 0x4081bcc0 0x00000019 0x00000000 0x00000000 0x408265d0 0x00000000 0x00000000
40826600: 0x40825fcc 0x454c4449 0x00000000 0x00000000 0x00000000 0x408265c0 0x00000000 0x00000000
40826620: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x40823b90 0x40823bf8 0x40823c60
40826640: 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x42007030 0x00000000
40826660: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40826680: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
408266a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
408266c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
408266e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40826700: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40826720: 0x00000000 0x00000200 0x0000000c 0x0e56000a 0x00000000 0x40826750 0x00000010 0x4081d888
40826740: 0x00000000 0x00000000 0x00000100 0x0000000c 0x0d580008 0x00000000 0x4082ae5c 0x00000054
40826760: 0x00000000 0x40826760 0x00000000 0x00000000 0x00000000 0x40826778 0xffffffff 0x40826778
40826780: 0x40826778 0x00000000 0x4082678c 0xffffffff 0x4082678c 0x4082678c 0x00000001 0x00000001
408267a0: 0x00000000 0xc000ffff 0x00000000 0xd54304b0 0x40826758 0x00000028 0x421b0850 0x4083db5c
408267c0: 0x4083e270 0x00000002 0x408269cc 0x408269cc 0x00000001 0x4082683c 0x4082683c 0x00000001
408267e0: 0x00000054 0x00000000 0x408267e4 0x00000000 0x00000000 0x00000000 0x408267fc 0xffffffff
40826800: 0x408267fc 0x408267fc 0x00000000 0x40826810 0xffffffff 0x40826810 0x40826810 0x00000001
40826820: 0x00000001 0x00000000 0x0000ffff 0x00000000 0xb86283b2 0x408267dc 0x0000001c 0x421b081c
40826840: 0x00000000 0x00000000 0x408268ec 0x408268ec 0x408230c4 0x40826834 0x0000002c 0x001c0000
40826860: 0x001c0000 0x00000000 0x00000000 0x00000000 0x00000012 0x00000000 0x00000000 0x00000000
40826880: 0x4082688c 0x4081d794 0x0000002c 0x00800000 0x00800000 0x00000000 0x00000000 0x00000000
408268a0: 0x00000012 0x00000000 0x00000000 0x00000000 0x00000000 0x40826880 0x0000001c 0x00000000
408268c0: 0x40826cbc 0x6373696d 0x00000000 0x00000000 0x00000000 0x40826801 0x0000000c 0x4083bca8
408268e0: 0x00000000 0x408268d4 0x00000034 0x4081ad44 0x00000001 0x00000002 0x00009000 0x00006000
40826900: 0x00001000 0x0073766e 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x40826924
40826920: 0x00000034 0x4081ad44 0x00000001 0x00000001 0x0000f000 0x00001000 0x00001000 0x5f796870
40826940: 0x74696e69 0x00000000 0x00000000 0x00000000 0x00000000 0x4082695c 0x00000034 0x4081ad44
40826960: 0x00000000 0x00000000 0x00010000 0x001f0000 0x00001000 0x74636166 0x0079726f 0x00000000

ELF file SHA256: efa4db093

Rebooting...
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0xed4
load:0x4086c410,len:0xc18
load:0x4086e610,len:0x2b44
entry 0x4086c410
```

### Additional Information

i already tried to set the wdt manually with
```
import machine
wdt = machine.WDT(timeout=1000000)
wdt.feed()
```
but still the same errors. What else can I try to do?
WLAN and Co is working perfectly in C/C++, and with circuitpy, but i want to use micropython.

Have a nice day!

### Code of Conduct

Yes, I agree

robert-hh, 2024-11-02:
I cannot replicate your observation. Station mode and Access Point mode work fine. Some notes on your scripts:

In station mode you have to wait until the device is connected. Then the IP address can be set with:
`wlan.ipconfig(addr4='addr4')`
Otherwise one gets an error about invalid IP address. Alternatively you can use wlan.ifconfig() to set all IP4 properties before calling wlan.connect().

MicroPython v1.25.0-preview.10.g0e490b7c8 on 2024-11-02; ESP32C6 module with ESP32C6

Thecoolpeople, 2024-11-05:
With version 1.24, it is not working, but with version
MicroPython v1.25.0-preview.10.g0e490b7c8 on 2024-11-02; ESP32C6 module with ESP32C6
everything works as expected!

Thanks for the help

closed 2024-11-05 by Thecoolpeople