micropython/micropython #16144: Wrong message on malloc fail

Tangerino, 2024-11-03:
### Port, board and/or hardware

esp32

### MicroPython version

MemoryError: memory allocation failed, allocating %u bytes
MicroPython v1.24.0 on 2024-10-25; Generic ESP32 module with ESP32

### Reproduction

Allocated a big list

### Expected behaviour

Fail with the right message

### Observed behaviour

MemoryError: memory allocation failed, allocating %u bytes
MicroPython v1.24.0 on 2024-10-25; Generic ESP32 module with ESP32

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

dpgeorge, 2024-11-03:
I'm not sure exactly what the bug report is here, what is the "right message"?

Here's what I get:
```
MicroPython v1.24.0 on 2024-10-25; Generic ESP32 module with ESP32
Type "help()" for more information.
>>> list(range(10000000))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
MemoryError: memory allocation failed, allocating 131072 bytes
```

stinos, 2024-11-04:
If the message literally is
```
MemoryError: memory allocation failed, allocating %u bytes
```
i.e. containing the `%u` then there's an issue substituting that

jonnor, 2025-03-03:
@Tangerino can you confirm you are seeing a literal %u in the message? 

Also, is this using the pre-built firmware from micropython.org, or did you build it yourself?