micropython/micropython #16165: wifi channel selection in access point mode is ignored

stanelie, 2024-11-06:
### Port, board and/or hardware

Waveshare ESP32-S3-ZERO

### MicroPython version

MicroPython v1.24.0 on 2024-10-25; Generic ESP32S3 module with ESP32S3

### Reproduction

Here is the code I use to test : 
```
import network

wlan = network.WLAN(network.AP_IF)
wlan.active(True)
wlan.config(ssid='potato',password='password',channel=6)

while True
    print(wlan.config('channel'))
```

### Expected behaviour

I expect this code to create an access point that will be on channel 6.

### Observed behaviour

The returned print command indicates the channel is "1", and a wifi scanner also confirms the access point is being broadcasted on channel 1.

### Code of Conduct

Yes, I agree

mdaeron, 2024-11-10:
Same here with micropython v1.25.0-preview.24.g5dfbd4371.dirty on a Lolin S2 mini board. Any idea when this bug appeared?

stanelie, 2024-11-11:
Hi @mdaeron. Does your comment imply that this used to work in the past?

mdaeron, 2024-11-11:
> Hi @mdaeron. Does your comment imply that this used to work in the past?

Yes, definitely. But that was perhaps a couple of years ago (micropython 1.18 or 1.19). I'm currently testing to see which combination of micropython + esp-idf versions restores this functionality. Will report here if I get somewhere.

dpgeorge, 2024-11-12:
Related to #8991

PR #16225 (2024-11-28): esp32: Fix AP mode channel configuration.
### Summary

* Seems at some point (probably following an ESP-IDF update), the channel setting code added in #8991 stopped working for the AP interface. This is because `esp_wifi_config()` is called after `esp_wifi_set_channel()`, the old channel gets re-configured and the AP doesn't change channels as expected.
* For AP mode, calling `esp_wifi_config()` seems to be all that's needed to get desired behaviour. The Wi-Fi stack configures 40MHz bandwidth and selects the secondary channel based on the available channels.
* However, the `esp_wifi_set_channe [... cut]
closed 2024-11-28 by projectgus