micropython/micropython #16171: SDCard os.mount(sd,'/sd') OSError: [Errno 5] EIO

FreddyBI, 2024-11-06:
### Port, board and/or hardware

ports esp32 build-ESP32_GENERIC Chip ESP32D0WDQ5 (revision 1)

### MicroPython version

MicroPython v1.25.0-preview.16.g919756cea on 2024-11-05; Generic ESP32 module with ESP32

### Reproduction

MicroPython v1.25.0-preview.16.g919756cea on 2024-11-05; Generic ESP32 module with ESP32

import machine
sd=machine.SDCard(slot=2)
import os
os.mount(sd,'/sd')

Traceback (most recent call last):
  File "<stdin>", line 4, in <module>
OSError: [Errno 5] EIO

### Observed behaviour

no

### Additional Information

with build with IDF 5.0.4 and v1.2'.0-preview "no problem"

### Code of Conduct

Yes, I agree

dpgeorge, 2024-11-13:
Thanks for the report.  See #16222 and #16223 for candidate fixes.

dpgeorge, 2024-11-18:
Fixed by 5dc9eda1953668eb6861be01ca85f147dcf8d406