micropython/micropython #16179: wifi long range mode (MODE_LR) is not working

stanelie, 2024-11-07:
### Port, board and/or hardware

Waveshare ESP32-S3-ZERO

### MicroPython version

MicroPython v1.24.0 on 2024-10-25; Generic ESP32S3 module with ESP32S3

### Reproduction

I have been unable to set the long range mode of wifi using micropython.
Either I can't read, or the documentation is lacking, or the feature does not work. Here is the code I tried : 

```
import network

sta = network.WLAN(network.STA_IF)
sta.active(True)
sta.config(protocol=network.MODE_LR)
```

### Expected behaviour

I expect this to set the LR mode that is available on Espressif devices.

### Observed behaviour

Nothing changes, the mode is not set.
print(network.phy_mode()) reports NONE

### Code of Conduct

Yes, I agree

marrelUnderscore, 2024-12-22:
Seems like this got even more broken/removed with version 1.24.1:
When trying to use `sta.config(protocol=network.MODE_LR)` (or any other mode) i just get `RuntimeError: Wifi Unknown Error 0x0102`. Also, MODE_LR seems to no longer be present at all (or I am doing something wrong)?

stanelie, 2024-12-23:
Have you tried completely removing the power from the ESP32 between each test? I think I remember seeing an error like that when I was just updating the code and running it without completely removing power.