micropython/micropython #16181: lightsleep on Pico W regression for 1.24.0

jeanlucmongrain, 2024-11-08:
### Port, board and/or hardware

PICO W

### MicroPython version

Micropython 1.24.0

### Reproduction

Micropython 1.23.0 can run that code:

```
import machine
import time

led = machine.Pin("LED", machine.Pin.OUT)
led.off()

while True:
    print("Loop instance")
    led.on()
    time.sleep(0.1)   
    led.off()
    machine.lightsleep(10000)
```

it will sleep for 10s.

if you try the same with 1.24.0 there is absolutely no sleep at all. the string `Loop instance` just appears every 0.1 seconds

### Expected behaviour

`machine.lightsleep` to do something on 1.24.0

### Observed behaviour

`machine.lightsleep` do something on 1.23.0 but not 1.24.0

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

jeanlucmongrain, 2024-11-08:
related to #16180 16180

CrabbyPete, 2024-12-08:
I got the same result using a regular PICO. Works on 1.23, not on 1.24.1

scruss, 2024-12-09:
* expected behaviour seen on a Raspberry Pi Pico (RP2040, **RP2-B1**, _no_ wifi) running v1.24.1
* observed behaviour seen on a Raspberry Pi Pico W (RP2040,  **RP2-B2**, _with_ wifi) running v1.24.1

I don't have a B2 revision Pico to try this on: all of mine are old.

scruss, 2024-12-09:
> I got the same result using a regular PICO. Works on 1.23, not on 1.24.1

Does your Pico say RP2-B1 on the chip, or RP2-B2?

projectgus, 2025-03-25:
I posted some explanation of this in one of the linked PRs that's since closed, but here's what is happening (mostly copied from #16431):

`lightsleep()` behaviour has always been to "sleep for the allotted time unless woken by an interrupt first". This issue was introduced in https://github.com/micropython/micropython/commit/74fb42aa82d537b6fefb463907274bd660c6ce8d (https://github.com/micropython/micropython/pull/13329) when we switched away from the pico-sdk alarm pool for soft timers and unintentionally made soft timer expiry a wakeup source.

Before https://github.com/micropython/micropython/commit/74fb42aa82d537b6fefb463907274bd660c6ce8d (including V1.23), both the pico-sdk alarm pool and the lightsleep wakeup timer use alarm 3. The lightsleep timer would quietly clobber alarm 3's timeout, meaning softtimer events wouldn't wake the chip from lightsleep.

After https://github.com/micropython/micropython/commit/74fb42aa82d537b6fefb463907274bd660c6ce8d (including V1.24), soft timer wakeup happens on timer alarm 2 so this interrupt can wake the chip from light sleep. On PICO-W builds this happens every lwIP tick (64ms), which is unexpected and prevents lightsleep from sleeping more than 64ms.

We could change the code back so that soft timer isn't a wakeup source, or make this configurable. However, I think the happy path for most users will be to stop the the lwIP tick (or timers) when there aren't any active network interfaces. That would mean lightsleep+wifi would work (lightsleep stills exit earlier, but wifi stays up with minimal power consumption), however if you instead disconnect wifi then lightsleep would run until something else woke it up or it hits the timeout.

This work has gotten intertwined with some related pico-sdk sleep problems, but now those are approaching resolution (see #16454) then it's probably a good time to revisit it.

madozu, 2025-04-30:
I consider the proposal from @projectgus to be a really good solution:

> However, I think the happy path for most users will be to stop the the lwIP tick (or timers) when there aren't any active network interfaces. That would mean lightsleep+wifi would work (lightsleep [...]

This would bring back the possibility to lightsleep for >64ms (with WiFi turned off) as it was with <=1.23.0. The Pico_W guys can go back to 1.23 ... but for the Pico2_W there is currently no solution, so this proposal is even more important for those wanting to use a Pico2_W in low power-consumption scenario.

Is there any work going on to stop the the lwIP tick/timers when there aren't any active network interfaces?

cpottle9, 2025-04-30:
I will weigh-in on @projectgus and @madozu recent comments.

I don't seem much point in calling lightsleep() while the wifi radio is active.
The wifi radio consumes much more power than lightsleep saves.

When I run the following fragment on a RPI Pico W power consumption (measured on USB) 
jumps from 19.2 milli-amps eventually stabilizing at 52.2 milli-amps.
```
from network import WLAN, STA_IF
from rp2 import country
from time import sleep_ms
country('ca')
wlan=WLAN(STA_IF)
wlan.active(True)
wlan.connect('my_ssid', 'my_password')
sleep_ms(10000)
```
Running on a RPI Pico (where lightsleep works) I measure a power consumption drop
of from 19.9 milli-amps to 4.96 milli-amps.

For an application running from battery it makes sense to shutdown WIFI and the CYW43 chip first.
Only after that call lightsleep.

Having said that, I agree with @projectgus about the 'happy path'.

PR #17272 (2025-05-21): rp2: Disable the LWIP tick timer when not needed.
### Summary

Closes #16181. Prevents lightsleep being woken up every 64ms to service LWIP timers, when:

1. No netif is up, and
2. No TCP sockets are active

The TCP socket check may not be strictly necessary, but without ticking the tcp timer they won't ever time out by themselves.

It means that it's possible to do:

```py
wlan.disconnect()
machine.lightsleep(5000)
wlan.connect(...)
```

... and get the full sleep time, but also possible to do 

```py
while some_condition():
    machine.lightsleep(5000)
```

... and Python will wake up every  [... cut]