micropython/micropython #16190: NeoPixel does not work on RP2350B with GPIO pin >=32

sfe-SparkFro, 2024-11-08:
### Port, board and/or hardware

rp2 port with RP2350B

### MicroPython version

v1.24.0-preview.461.gb8227a3f7.dirty

### Reproduction

```
import neopixel
import machine

pin = 32

p = machine.Pin(pin)
n = neopixel.NeoPixel(p, 1)

n[0] = (127, 0, 0)
n.write()
```

### Expected behaviour

A NeoPixel connected to `pin` should illuminate.

### Observed behaviour

A NeoPixel connected to `pin` only illuminates if `pin` is <32. If `pin` is >=32, the NeoPixel does not illuminate.

### Additional Information

The root problem appears to be here:

https://github.com/micropython/micropython/blob/5dfbd4371480c175e6f54e57e4e5eb5d579cbd00/ports/rp2/mphalport.h#L184-L190

`gpio_set_mask()` requires a 32-bit value, so the shifted bit gets truncated when using a pin >= 32. There is a `gpio_set_mask64()`, however for some reason it didn't work when I tried it (I didn't dig into it). However changing to `gpio_put()` works fine on my end.

### Code of Conduct

Yes, I agree

robert-hh, 2024-11-08:
Looks like mp_hal_pin_low() and mp_hal_pin_high() in rp2/mphalport.h have to be fixed to use gpio_set_mask64() and gpio_clr_mask64() instead of the 32bit versions. Someone with a RP2350B could implement and test it. It may have an impact on Neopixel timing.

sfe-SparkFro, 2024-11-08:
Turns out the reason the 64-bit versions didn't work is because you need to explicitly typecast to a `uint64_t`. Will update my PR, since I think the mask functions are faster than `gpio_put()` (though I didn't see any timing issues).

robert-hh, 2024-11-08:
> though I didn't see any timing issues

I will compare it to the previous state tomorrow. I still have the old test data from rp2040. The difference should be small, like a few clock cycles per output phase.

sfe-SparkFro, 2024-11-08:
Actually, would it even matter if it takes longer? The bitstream implementation (which the NeoPixel driver uses) just waits until it's time to toggle the pin state:

https://github.com/micropython/micropython/blob/5dfbd4371480c175e6f54e57e4e5eb5d579cbd00/ports/rp2/machine_bitstream.c#L83-L93

So as long as setting the pin high/low doesn't take longer than the time between pulses ([shortest is 400ns](https://github.com/micropython/micropython-lib/blob/01047889eb1acf115424fee293e03769f6916393/micropython/drivers/led/neopixel/neopixel.py#L20)), I don't think it matters if this change results in a couple extra clock cycles.

robert-hh, 2024-11-09:
Probably it will not matter. Especially not for neopixel, which has quite relaxed timing requirements.

robert-hh, 2024-11-09:
Testing gives sometimes unexpected results. I tested the bitstream timing with a unmodified RP2350A and the differ from the rp2040. So it needs a small adjustment. I made a change to the compensation value as of:
```
#if PICO_RP2350
#define MP_HAL_BITSTREAM_NS_OVERHEAD  (5)
#else
#define MP_HAL_BITSTREAM_NS_OVERHEAD  (9)
#endif
```
Maybe you could include this change into your PR, making it a single commit for about the same topic.

All timings made with a bit tuple of (1000, 1000, 1000, 1000) and \x55\x55 as pattern.
Timing with 9ns compensation at 125MHz clock:
[image]

Timing with 5ns compensation at 125MHz clock:
[image]

Timing with 5 ns compensation at 150MHz clock (default for RP2350):
[image]

For comparison a figure for RP2340:
Timing with 9ms compensation at 125MHz:
[image]

robert-hh, 2024-11-09:
For comparison the timing for RP2350, RISCV. The 150Mhz figures are good, the ones for 125MHz and 250Mhz are worse. They would need a larger compensation value. The value for 250 Mhz is surprising. Usually the timing got better at higher frequencies. 

150MHz, 5ns, compensation:
[image]

125MHz, 5 ns compensation:
[image]

sfe-SparkFro, 2024-11-09:
PR updated with your suggested timings!

PR #16191 (2025-01-11): ports/rp2/mphalport.h: Fix mp_hal_pin_xyz() for pin>=32
### Summary

Fixes #16190

### Testing

Tested on a board with an RP2350B, now NeoPixels on pins >=32 work.

### Trade-offs and Alternatives

~This uses `gpio_put()`, though `gpio_set_mask64()` might also work. The latter didn't work right away for me, and I did not bother testing further since `gpio_put()` worked fine, so it's possible I just did something wrong with `gpio_set_mask64()`.~
closed 2025-01-11 by dpgeorge