micropython/micropython #16199: PIO does not work on RP2350B with GPIO pin >=32

sfe-SparkFro, 2024-11-09:
### Port, board and/or hardware

rp2 port with RP2350B

### MicroPython version

v1.24.0-preview.461.gb8227a3f7.dirty

### Reproduction

Connect a quadrature encoder to any pair of GPIO pins. Then create an instance of this class, which implements a quadrature decoder using PIO: https://github.com/Open-STEM/XRP_MicroPython/blob/main/XRPLib/encoder.py

Then run the following while rotating the encoder:

```
pinA = 0
pinB = 1

myEncoder = Encoder(0, pinA, pinB)

while True:
    print(myEncoder.get_position_counts())
    time.sleep(0.1)
```

### Expected behaviour

A PIO encoder should work on any pair of pins, and the printed output should change as the encoder rotates.

### Observed behaviour

A PIO encoder only works if `pinA` and `pinB` are <32. If they're >=32, the printed output is fixed at 0.

### Additional Information

I'm not super familiar with MicroPython, but this appears to be where the state machine is created:

https://github.com/micropython/micropython/blob/5dfbd4371480c175e6f54e57e4e5eb5d579cbd00/ports/rp2/rp2_pio.c#L726-L743

Which eventually calls [this Pico SDK function](https://github.com/raspberrypi/pico-sdk/blob/efe2103f9b28458a1615ff096054479743ade236/src/rp2_common/hardware_pio/include/hardware/pio.h#L381-L383):

https://github.com/micropython/micropython/blob/5dfbd4371480c175e6f54e57e4e5eb5d579cbd00/ports/rp2/rp2_pio.c#L666-L669

Not sure whether the root problem is with MicroPython or the Pico SDK, but I suspect it may be related to https://github.com/raspberrypi/pico-sdk/issues/2030 TLDR; need to set `pio->gpiobase` to 16 if the pin is >=32.

### Code of Conduct

Yes, I agree

lurch, 2024-11-11:
> ### Expected behaviour
> A PIO encoder should work on any pair of pins, and the printed output should change as the encoder rotates.

I guess (even after this bug is fixed) they'd still need to be in the same "PIO GPIO range"? So both 0-31 inclusive, or both 16-47 inclusive?

lurch, 2024-11-25:
Note that there's been several fixes for using pins >= 32 with PIO in the recently-released Raspberry Pi Pico SDK 2.1.0
https://github.com/raspberrypi/pico-sdk/releases/tag/2.1.0

Gadgetoid, 2024-11-25:
*snip*

```
hardware_pio

Added pio_sm_set_pins64(), pio_sm_set_pins_with_mask64() and pio_sm_set_pindirs_with_mask64() to allow setting of >32 pins.
Much improved documentation of how GPIO numbers > 32 are handled.
Fixed a bug in the use of a "jmp pin" > 32.
Fixed implementation of sm_config_set_in_pin_count()
Renamed sm_config_set_clkdiv_int_frac() to sm_config_set_clkdiv_int_frac8() to be clear that it takes an 8-bit fraction; the old name is still supported. Note that "int" part in the new method is 32-bit not 16-bit for consistency with other clkdiv methods.
Renamed pio_calculate_clkdiv_from_float() to pio_calculate_clkdiv8_from_float() to be clear that it produces an 8-bit fraction; the old name is still supported. Note that "int" part in the new method is 32-bit not 16-bit for consistency with other clkdiv methods.
Added PICO_PIO_CLKDIV_ROUND_NEAREST for controlling rounding of floating-point clock dividers
```
--

I think it's probably worth getting an SDK bump into MicroPython ahead of #16057 since I think there may be some more implications and there's certainly the new `hardware_xip_cache` stuff we should be using in #15620.

It's way too Monday morning and I'm still getting my head around things 🫠

maholli, 2025-06-16:
Bumping this to say PIO functionality on pins > 32 is certainly still needed. Wish I was up to speed enough on the pico-sdk to help here.

Gadgetoid, 2025-07-26:
Most of the PIO I work with is in C and handily avoids the quirks but there are a couple of related issues- setting up PIO pins on RP2350 is definitely a lot more fraught than RP2040.

I’ll set up a test case (won’t be for a few days, mind) and see if I can repro and if any of my changes thus far accidentally fix this.