micropython/micropython #16215: Can Support CH32V103 ?

Witawat, 2024-11-12:
### Description

Product mcu : https://www.wch-ic.com/products/CH32V103.html

Features
Qingke V3A processor, max 80MHz system clock frequency.
Single-cycle multiplication and hardware division.
20KB SRAM, 64KB CodeFlash.
2.7V to 5.5V supply voltage, supplies to GPIO simultaneously.
Low-power modes: sleep/stop/standby.
Power-on/power-down reset (POR/PDR).
Programmable voltage detector (PVD).
7-channel DMA controller.
16-channel TouchKey detection.
16-channel 12-bit ADC.
7 timers.
One USB2.0 host/device interface (full-speed and low-speed).
Two I2C interfaces (support SMBus/PMBus).
3 USARTs.
2 SPIs (Master mode and Slave mode).
51 I/O ports, all I/O ports can be mapped to 16 external interrupts.
CRC calculation unit, 96-bit unique ID.
Serial debug interface (SDI).
Packages: LQFP64M, LQFP48, QFN48.

I hope there will be python language to support writing code with mcu wch.

### Implementation

I hope the MicroPython maintainers or community will implement this feature

### Code of Conduct

Yes, I agree

mattytrentini, 2024-11-12:
It may be _possible_, but would be of limited use due to the small amount of RAM. 

It's likely that, to fit on a device with such tight constraints, core features like the compiler, REPL and other major components will need to be excluded from the build. It might be possible to freeze all code and build to this device but it's a lot of effort to maintain a port that limits the benefits that MicroPython provides...

I think it'd be better to focus on the larger cousin, the WCH CH32V307. There's already a good start for this micro, see #8637.

agatti, 2024-11-12:
Following on what @mattytrentini said, the V208 can also be a valid MicroPython target (I'm also working on a more general port for the WCH family of RV32 processors - I do have a version running for that MCU as well), but for the V208 unless you strip it down to remove a fair amount of functionality you will be forced to load code from the SDCard in that case.

I'd say anything besides V208, V303 w/64k SRAM, V307, and V317 is out of the question for that MCU family at this time.

yfyungit, 2024-11-12:
@agatti This is great, I also hope that mircopython supports the CH32V208 chip

yfyungit, 2024-11-15:
@slorquet I did not find relevant information about Bluetooth in English, but I found relevant instructions for Chinese information for you. You can download it by entering the link below.
https://www.wch.cn/downloads/CH32V20xEVT_ZIP.html

yfyungit, 2024-11-15:
> @slorquet I did not find relevant information about Bluetooth in English, but I found relevant instructions for Chinese information for you. You can download it by entering the link below.我没有找到蓝牙的相关英文资料，但是我给你找到了中文资料的相关说明。你可以通过下面的链接下载。 https://www.wch.cn/downloads/CH32V20xEVT_ZIP.html

He exists in the following directory in the compressed package 

CH32V20xEVT\EVT\EXAM\BLE