micropython/micropython #16233: Can't open https url on picow - micropython changes it to http

vinceskahan, 2024-11-13:
### Port, board and/or hardware

picoWH

### MicroPython version

RPI_PICO_W-20241025-v1.24.0.uf2 from https://micropython.org/download/RPI_PICO_W/

### Reproduction

```
import requests
import json

# connect to wifi
import network
import time
wlan = network.WLAN(network.STA_IF)
wlan.active(True)
wlan.connect('MYSSIDHERE', 'MYWIFIPASSHERE')
while not wlan.isconnected() and wlan.status() >= 0:
     print("Waiting to connect:")
     time.sleep(1)
print();print("ifconfig = ",wlan.ifconfig())

url="https://api.weather.gov/gridpoints/SEW/121,54/forecast"
response = requests.get(url)

print("trying url: ", url)

forecast = None

try:
    forecast = response.json()
except Exception as e:
    print(e)
    print()
    print(response.content)
    print()
    
print(forecast)
```

### Expected behaviour

Expect to see the correct JSON response that the https url above returns from a browser

### Observed behaviour

````
MPY: soft reboot

ifconfig =  ('192.168.1.98', '255.255.255.0', '192.168.1.1', '192.168.1.171')
trying url:  https://api.weather.gov/gridpoints/SEW/121,54/forecast
syntax error in JSON

b'<HTML><HEAD>\n<TITLE>Access Denied</TITLE>\n</HEAD><BODY>\n<H1>Access Denied</H1>\n \nYou don\'t have permission to access "http&#58;&#47;&#47;api&#46;weather&#46;gov&#47;gridpoints&#47;SEW&#47;121&#44;54&#47;forecast" on this server.<P>\nReference&#32;&#35;18&#46;10853417&#46;1731534552&#46;a43648\n<P>https&#58;&#47;&#47;errors&#46;edgesuite&#46;net&#47;18&#46;10853417&#46;1731534552&#46;a43648</P>\n</BODY>\n</HTML>\n'

None
````

### Additional Information

Yes - if you look carefully at the output (sorry about the scrollbar) you'll see that I am trying to open a 'https' url, yet micropython actually is trying to open a 'http' url for some reason.  Is this a bug, or do I need to add something to get https support to be able to request a https url ?

I might note that I'm running connected via USB to a mac mini that 'can' get to the https url just fine, if that matters.

### Code of Conduct

Yes, I agree

dpgeorge, 2024-11-14:
It looks like the `api.weather.gov` site rejects requests that do not include the `User-Agent` header.

You can add this header yourself, eg:
```py
response = requests.get(url, headers={"User-Agent": "micropython"})
```
That should work (I tested it on the unix port of MicroPython).

> if you look carefully at the output (sorry about the scrollbar) you'll see that I am trying to open a 'https' url, yet micropython actually is trying to open a 'http' url for some reason

It's accessing the HTTPS site, it's just the error message in that 403 response that is wrong.

vinceskahan, 2024-11-14:
It works.  Thanks so much  !!!!!

Just for my info, how did you figure that out ?

dpgeorge, 2024-11-14:
> how did you figure that out ?

Using CPython + requests with plain HTTP (which gave a valid response) and Wireshark to view the HTTP packets, I copied all the headers it was sending and worked out which one was required.

vinceskahan, 2024-11-14:
Thanks again !!

-----  ***@***.*** ----

sosi-deadeye, 2024-11-14:
Many APIs refuse to work, if the `User-Agent` is missing. Some APIs even check, if the `User-Agent` is a valid browser. Requests sends also a `User-Agent`. This is why it works with some APIs. But there are also APIs which refuse to work with requests, if you do not change the `User-Agent`.

Here a list of well known `User-Agent`: https://deviceatlas.com/blog/list-of-user-agent-strings