micropython/micropython #16239: Maximum recursion depth

Vishal-Birajdar, 2024-11-14:
### Port, board and/or hardware

Pi Pico w 

### MicroPython version

MicroPython v1.22.2 on 2024-02-22; Raspberry Pi Pico W with RP2040

### Reproduction

import network
import json
from umqtt.robust import MQTTClient
import machine
from esecurity import call_loop
import ssl
import time
import shared
import micropython
#from ota import OTAUpdater

s1_pin= machine.Pin(6,machine.Pin.IN,machine.Pin.PULL_UP)
s2_pin= machine.Pin(7,machine.Pin.IN,machine.Pin.PULL_UP)
s3_pin= machine.Pin(8,machine.Pin.IN,machine.Pin.PULL_UP)
s4_pin= machine.Pin(9,machine.Pin.IN,machine.Pin.PULL_UP)
sensor_pin=[s1_pin,s2_pin,s3_pin,s4_pin]

f = open('config.json',"r")
config = json.load(f)

interval = 1000
previousMillis = time.ticks_ms()
wlan = 0
max_attempts=30
retry_delay=6
check_interval=0.5
x = call_loop()

#-------------------------------------___________POWER_ON____________---------------------------------------------------------------
shared.pix[4] = shared.RED
shared.pix.write()
time.sleep(10)
shared.pix[4] = shared.GREEN
shared.pix.write()

shared.event("\tlog Data\n")
#---------------------------------------------_________________CONNECTION__________________--------------------------------------------

def wifi_connect(max_attempts=30, retry_delay=6, check_interval=0.5):
    global wlan
    wlan = network.WLAN(network.STA_IF)
    wlan.disconnect()
    wlan.active(True)
    
    print('Connecting to network....')
    
    for attempt in range(max_attempts):
        wlan.connect(config['wifi']['ssid'],config['wifi']['password'])
        # Wait for connection to establish, checking periodically
        elapsed_time = 0
        while elapsed_time < retry_delay:
            if wlan.isconnected():
                print("Connected")
                print("IP", wlan.ifconfig())
                return True
            time.sleep(check_interval)
            elapsed_time += check_interval
        print(f"\n Attempt {attempt + 1}/{max_attempts} failed, retrying...")
    else:
        print("wifi connection failed !")
        shared.event("wifi connection failed !\n")
        machine.reset()

#---------------------------------------------____________READ_CERTIFICATE_______________--------------------------------------------
def read_certificate(file_path):
    try:
        with open(file_path, 'rb') as f:
            data = f.read()
        if not data:
            print(f"Error: {file_path} is empty.")
            return None
        print(f"{file_path} read successfully.")
        return data
    except OSError as e:
        print(f"Error reading {file_path}: {e}")
        return None
#------------------------------------------------------------_______________MQTT_CONNECTION________________--------------------------------------------------------

#-----------------------------------------------------------______________________________--------------------------------------------------------
def mqtt_connect():
    print("Attempting to connect to MQTT broker...")
    shared.client = MQTTClient(config['device_info']['device_id'],config['mqtt']['broker'],user=config['mqtt']['user'],password=config['mqtt']['password'],keepalive=60,ssl=True)
    shared.client.set_callback(x.callback)
    shared.client.set_last_will(shared.topic_hb,"-1",False,qos=0)
    shared.client.connect()
    print('Connected to %s MQTT Broker' % config['mqtt']['broker'])
    
    shared.client.subscribe(shared.topic_cmd+"/D")
    shared.client.subscribe(shared.topic_cmd+"/FD")
    shared.client.subscribe(shared.topic_cmd+"/R1")
    shared.client.subscribe(shared.topic_cmd+"/R2")
    shared.client.subscribe(shared.topic_cmd+"/R3")
    shared.client.subscribe(shared.topic_cmd+"/R4")
    shared.client.subscribe(shared.topic_cmd+"/RR")
    shared.client.subscribe(shared.topic_cmd+"/MR")
    shared.pix[3] = shared.GREEN
    shared.pix.write()
    #shared.client.publish(shared.topic_hb,"1")
    #return shared.client  
    """except Exception as e:
        mqtt_connect()
        return shared.client
        print('Failed to connect to MQTT broker. Reconnecting...',e)
        shared.event('\n Failed to connect to MQTT broker. Reconnecting...')
        shared.client = None
        return shared.client
        machine.reset()"""
#----------------------------------------------------__________VERSION_UPDATE_______________--------------------------------------------
  
#p1 = Connection()       

"""def check_updates():
    try:
        firmware_url = "https://github.com/Vishal-Birajdar/Micropython-/"
        files = ["shared.py", "esecurity.py", "config.json"]
        ota_updater = OTAUpdater(config['wifi']['ssid'],config['wifi']['password'] , firmware_url, files)
        ota_updater.download_and_install_update_if_available()
    except:
        shared.event("Error in updating device \n")
        machine.reset()"""
        

#--------------------------------------------------------_______________________________--------------------------------------------------------
key_data = read_certificate('privkey4.pem')
cert_data = read_certificate('cert4.pem')
ca_data = read_certificate('chain4.pem')

ssl_params = {
    "certfile": cert_data,
    "keyfile": key_data,
    "cadata": ca_data,
    "server_hostname": config['mqtt']['broker'],
    "cert_reqs": ssl.CERT_REQUIRED
}
#check_updates()
#p1.mqtt_connect()
try:
    wifi_connect()
    mqtt_connect()
except Exception as e:
    mqtt_connect()
    time.sleep(0.3)
    print('Failed to connect to MQTT broker. Reconnecting...',e)
    shared.event('\n Failed to connect to MQTT broker. Reconnecting...')
    
    

#-----------------------------------------------------------__________CONNECTION_CHECKING________________---------------------------------------------

while True:
    try:
        shared.client.publish(shared.topic_hb,'1')
        currentMillis = time.ticks_ms()
        for i, s_pin in enumerate(sensor_pin):
            shared.buffer["S"+str(i+1)] = str(s_pin.value())+str(i+1)
            if s_pin.value()==1:shared.active_sensor[i]=1
            else:shared.active_sensor[i]=0
        shared.act_sensor_cnt=shared.active_sensor.count(1)
        #print("active sensor",shared.act_sensor_cnt)
        print("sensor buffer",shared.active_sensor)
        time.sleep(0.1)
        x.main()
        if time.ticks_diff(currentMillis,previousMillis)>=interval:
            previousMillis = currentMillis
            if not wlan.isconnected():
                shared.pix[1] = shared.YELLOW
                shared.pix.write()
                wifi_connect()
                #time.sleep(.4)
            """if shared.topic_hb == "1":
                shared.pix[1] = shared.WHITE
                shared.pix.write()
                pass
            else:
                shared.pix[1] = shared.RED
                shared.pix.write()
                wlan.disconnect()
                time.sleep(0.3)
                print("Retrying wifi..")
                wlan.connect(config['wifi']['ssid'],config['wifi']['password'])
                time.sleep(0.5)
                if wlan.isconnected():
                    print("Connected")
                    print("IP", wlan.ifconfig())
                    shared.client.reconnect()
                    print('client reconnected')
                    shared.pix[1] = shared.WHITE
                    shared.pix.write()
                    break"""
                        #shared.event("wifi and client Reconnected\n")
                
    except Exception as e:
        print("Error with ",e)
        shared.event("\n Error in the main flie in loop")
        machine.reset()

### Observed behaviour

I am not using any recursion or thread in my code but running a code after some time occuremaximum recursion depth error .

### Additional Information

I am working on Iot project there is three .py file one is for main logic and other is varibale declared which used in all three file. and third file is main.py here i impleted wifi,mqtt and in loop I check the 4 gpio input (for any detection) and then call the main logic function which is define in another file,and checking the wifi connection status continously.

 after running a code 10-15 min working  ok then i got two error
1) maximum recursion depth
2) mqtt connection lost

### Code of Conduct

Yes, I agree

sosi-deadeye, 2024-11-14:
Do you ever read your own posts after you've sent them?

You are not the first or the last person to format code incorrectly. The code must be enclosed with code tags.
https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code

You can also edit your original post.

Vishal-Birajdar, 2024-11-14:
> Do you ever read your own posts after you've sent them?
>
> You are not the first or the last person to format code incorrectly. The code must be enclosed with code tags. [...]

I posted issue first time on gitbub . sorry for mistakes

sosi-deadeye, 2024-11-14:
Just put your code in Code Tags, that other users could help.

projectgus, 2024-11-15:
@Vishal-Birajdar Thanks for the clear description of the problem. I've changed the `"""` in your post to <code>```py</code> so the code highlights correctly.

It's not possible to investigate a bug report without the code to reproduce it, and the code posted above is not complete or runnable  - the `ota`, `shared`, and `x` modules are missing, for example - and there are references to third party modules like `OTAUpdater`

My best guess is there are two likely possibilities for what is happening here:

* One of the callbacks (for example `x.message_callback` is calling back around into some other function which eventually recurses.
* The call stack is too deep without any actual recursion (i.e. too many functions calling into other functions). This is unlikely but it's possible.

If that's not enough to solve the problem, suggest you remove or disable code in your program and re-run until you find a version which doesn't crash. If you do this carefully, you may find the root cause of the problem yourself. If you don't, you may end up with a minimal working example that we can use to diagnose the bug.

projectgus, 2024-11-15:
Oh, I looked again and there is actually recursion here:

```py
def mqtt_connect():
    try:
        print("Attempting to connect to MQTT broker...")
        # (removed some lines)  
    except Exception as e:
        mqtt_connect()
        # (removed more lines)
```

The exception handler for `mqtt_connect()` calls `mqtt_connect()`, so each time the connection fails you are pushing a frame on the stack. If the network goes down then it could easily call into itself many times. Try restructuring the code with a while loop around the outside of the try/except instead of using recursion.

Josverl, 2024-11-15:
That suggested structure would look something like

```py
def mqtt_connect():
    while True:
        try:
            print("Attempting to connect to MQTT broker...")
            # (removed some lines)
            break  # Exit the loop if the connection is successful
        except Exception as e:
            print(f"Connection failed with error: {e}. Retrying...")
            # (removed more lines)
```

Vishal-Birajdar, 2024-11-15:
this is my another file where i declare callback function.
'''
class call_loop:
    def __init__(self):
        pass
    def callback(self,topic_cmd,msg):
            decoded_topic = topic_cmd.decode('utf-8').strip()
            decoded_msg = msg.decode('utf-8').strip()
            #print("decoded_topic:"+decoded_topic)
            #print("decoded_msg:"+decoded_msg)
            
            if decoded_topic.endswith("FD"):
                print("decoded_topic:"+decoded_topic)
                if decoded_msg == "1105":
                    shared.buffer["FD"] = "1105"
                    shared.FG_Arm = True
                    print("shared.FG_Arm ",shared.FG_Arm)
                    shared.string_json = json.dumps(shared.buffer)
                elif decoded_msg == "0105":
                    shared.buffer["FD"] == "0105"
                    shared.FG_Arm = False
                    print("shared.FG_Arm ",shared.FG_Arm)
                    shared.string_json = json.dumps(shared.buffer)
                else:
                    print("Invalid Input !")
     shared.string_json = json.dumps(shared.buffer)
      shared.client.publish(shared.topic_json,shared.string_json)
'''

I am remov the try  and exception but same issue

sosi-deadeye, 2024-11-18:
> I am remove the try and exception but same issue again Is this error hardware related?, because I have soldered the controller on my I/O board.

What your `mqtt_connect` function does:
```
def foo():
    foo()
```

This is called recursion. In this special case, a recursion without limit. The only limit are the resources on the Microcontroller/SoC/Computer. Stack based languages requires memory for the stack of a function. After the function is left, the memory is freed. But if the function calls another function, the memory is still in use. On a Microcontroller, you can hit the recursion limit very fast because they do not have much RAM.

If you remove the `mqtt_connect()` call from the mqtt_connect function, it should solve the recursion issue.
Modified function:
```python

import machine
import time
import sys

from umqtt.robust import MQTTClient

import shared

# the next two functions should be implemented in shared
def led_set_green():
    """
    Illuminate green
    """
    shared.pix[3] = shared.GREEN
    shared.pix.write()

def led_set_red():
    """
    Illuminate red
    """
    shared.pix[3] = shared.RED
    shared.pix.write()

def mqtt_connect():
    # from where does config come?!

    shared.client = MQTTClient(
        config["device_info"]["device_id"],
        config["mqtt"]["broker"],
        user=config["mqtt"]["user"],
        password=config["mqtt"]["password"],
        keepalive=90,
        ssl=True,
    )

    # x does not exist
    # shared.client.set_callback(x.message_callback)
    shared.client.set_last_will(shared.topic_hb, "-1", False, qos=0)

    try:
        print("Attempting to connect to MQTT broker...")
        shared.client.connect()
        print(f'Connected to {config["mqtt"]["broker"]} MQTT Broker')

        for topic in ("D", "FD", "R1", "R2", "R4", "RR", "MR"):
            shared.client.subscribe(f"{shared.topic_cmd}/{topic}")

        # use simple functions to show the state
        led_set_green()

    except Exception as e:
        time.sleep(0.2)
        print("Failed to connect to MQTT broker. Reconnecting...")
        sys.print_exception(e) # print the original exception
        shared.event("\n Failed to connect to MQTT broker. Reconnecting...")
        # use simple functions to show the state
        led_set_red()
        time.sleep(0.3)
        machine.reset()

```

Vishal-Birajdar, 2024-11-19:
Thanks for the support.
I have tested your mqtt_connect function which is fine.
Actually it's a hardware related problem, I have uploaded the code in both controllers (one is IO board soldered controller and other is development pico w board), IO board is showing error after some time and development board is not giving any error (Test 4- 5 hours).

And also checked the input gpio voltage, out of 4 gpio input pins 2 gpio is showing 3.3 volts and 2 gpio is showing 2.89 volts
[image]