micropython/micropython #16259: Multiple complications with Termux (Android, armv7a, clang)

CardealRusso, 2024-11-17:
### Port, board and/or hardware

Unix, armv7a, android

### MicroPython version

1.24

### Reproduction

I had to go through a series of hacks and even change the micropython source code to get it to work in Termux. Having a deformed, broken, probably deficient Micropython in the end.  Hopefully, this experience can be resolved in the future?
Termux 0.119, clang 19.1, Galaxy a01

### “make” on a fresh/unmodified ports/unix:
The first error I get is this:
```
../../py/emitglue.c:118:47: error: passing 'uint8_t *' (aka 'unsigned char *') to parameter of type 'char *' converts between pointers to integer types where one is of the unique plain 'char' type and the other is not [-Werror,-Wpointer-sign]
  118 |     __builtin___clear_cache((void *)fun_data, (uint8_t *)fun_data + fun_len);
```
at

https://github.com/micropython/micropython/blob/611d8f9ce82ab5e04fb86ab6cfc28d5ed98f33bf/py/emitglue.c#L118
Which I was able to “ bypass” by exchanging for:

```c
__builtin___clear_cache((char *)fun_data, (char *)fun_data + fun_len);
```

### gchelper_generic:
```
CC ../shared/runtime/gchelper_generic.c
../shared/runtime/gchelper_generic.c:114:14: error: variable 'r4' is uninitialized when used here [-Werror,-Wuninitialized]
  114 |     arr[0] = r4;
      |              ^~
../shared/runtime/gchelper_generic.c:104:21: note: initialize the variable 'r4' to silence this warning
  104 |     register long r4 asm ("r4");
      |                     ^
      |                      = 0
...(10 or more similar hidden)
../shared/runtime/gchelper_generic.c:123:14: error: variable 'r13' is uninitialized when used here [-Werror,-Wuninitialized]
  123 |     arr[9] = r13;
      |              ^~~
../shared/runtime/gchelper_generic.c:113:22: note: initialize the variable 'r13' to silence this warning
  113 |     register long r13 asm ("r13");

```

fixed by setting #define MICROPY_GCREGS_SETJMP (1) at top of both mpconfigport.h from ports/unix and mpy-cross

### pthreads
Despite being present and functional, it refuses to compile with pthreads.

```
CC mpthreadport.c
mpthreadport.c:148:9: error: call to undeclared function 'pthread_cancel'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
  148 |         pthread_cancel(th->id);
      |         ^
mpthreadport.c:148:9: note: did you mean 'pthread_once'?
/data/data/com.termux/files/usr/include/pthread.h:264:5: note: 'pthread_once' declared here
  264 | int pthread_once(pthread_once_t* _Nonnull __once, void (* _Nonnull __init_routine)(void));
      |     ^
mpthreadport.c:206:5: error: call to undeclared function 'pthread_setcanceltype'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
  206 |     pthread_setcanceltype(PTHREAD_CANCEL_ASYNCHRONOUS, NULL);
      |     ^
mpthreadport.c:206:5: note: did you mean 'pthread_setname_np'?
/data/data/com.termux/files/usr/include/pthread.h:341:5: note: 'pthread_setname_np' declared here
  341 | int pthread_setname_np(pthread_t __pthread, const char* _Nonnull __name);
      |     ^
mpthreadport.c:206:27: error: use of undeclared identifier 'PTHREAD_CANCEL_ASYNCHRONOUS'
  206 |     pthread_setcanceltype(PTHREAD_CANCEL_ASYNCHRONOUS, NULL);

```

Fixed by disabling threads at mpconfigport.mk

And finally I have a micropython with unexpected behavior thanks to the modified __builtin___clear_cache, slow because of MICROPY_GCREGS_SETJMP 1 and no threads.
[image]

### Expected behaviour

make without errors

### Observed behaviour

errors

### Code of Conduct

Yes, I agree

agatti, 2024-11-17:
> ### pthreads
> 
> Despite being present and functional, it refuses to compile with pthreads.

The problem with Android's pthreads implementation is that `pthread_cancel` is simply not there, the rest is.

This diff makes MicroPython build without changes on an Android AArch64 tablet with Termux, and makes thread-related tests pass:

```diff
diff --git a/ports/unix/mpthreadport.c b/ports/unix/mpthreadport.c
index 16ac4da8b..720b1232b 100644
--- a/ports/unix/mpthreadport.c
+++ b/ports/unix/mpthreadport.c
@@ -145,7 +145,7 @@ void mp_thread_deinit(void) {
     while (thread->next != NULL) {
         mp_thread_t *th = thread;
         thread = thread->next;
-        pthread_cancel(th->id);
+        pthread_kill(th->id, SIGUSR1);
         free(th);
     }
     mp_thread_unix_end_atomic_section();
@@ -203,7 +203,6 @@ void mp_thread_start(void) {
     }
     #endif
 
-    pthread_setcanceltype(PTHREAD_CANCEL_ASYNCHRONOUS, NULL);
     mp_thread_unix_begin_atomic_section();
     for (mp_thread_t *th = thread; th != NULL; th = th->next) {
         if (th->id == pthread_self()) {
```

@dpgeorge is there interest in a PR that fixes this issue on Android under termux?  Since there's no explicit "ANDROID" define (or variants thereof) this will have to be put behind a define in `ports/unix/mpconfigport.h`.

dpgeorge, 2024-11-18:
> is there interest in a PR that fixes this issue on Android under termux

Yes!

agatti, 2024-11-18:
> ### gchelper_generic:
>
> ```
> CC ../shared/runtime/gchelper_generic.c
> ../shared/runtime/gchelper_generic.c:114:14: error: variable 'r4' is uninitialized when used here [-Werror,-Wuninitialized]
>   114 [...]

@CardealRusso this should help compilation on Clang for armv7a:

```diff
diff --git a/shared/runtime/gchelper_generic.c b/shared/runtime/gchelper_generic.c
index 093723137..ef3054efc 100644
--- a/shared/runtime/gchelper_generic.c
+++ b/shared/runtime/gchelper_generic.c
@@ -111,6 +111,18 @@ static void gc_helper_get_regs(gc_helper_regs_t arr) {
     register long r11 asm ("r11");
     register long r12 asm ("r12");
     register long r13 asm ("r13");
+    #ifdef __clang__
+    asm ("" : "=r" (r4));
+    asm ("" : "=r" (r5));
+    asm ("" : "=r" (r6));
+    asm ("" : "=r" (r7));
+    asm ("" : "=r" (r8));
+    asm ("" : "=r" (r9));
+    asm ("" : "=r" (r10));
+    asm ("" : "=r" (r11));
+    asm ("" : "=r" (r12));
+    asm ("" : "=r" (r13));
+    #endif
     arr[0] = r4;
     arr[1] = r5;
     arr[2] = r6;
```

Can you please try that change without enabling `MICROPY_GCREGS_SETJMP`?  All my Android devices are on AArch64 so the 32-bits helper isn't compiled.

agatti, 2024-11-18:
> ```
> CC ../shared/runtime/gchelper_generic.c
> ../shared/runtime/gchelper_generic.c:122:10: error: write to reserved register 'R11'
>   122 |     asm ("" : "=r" (r11));
>       |          ^
> ```
> 
> However, I'm using micrpython 1.24. Are these patches only for the most recent commit?

They are, but that should not matter - the gchelper code for Arm hasn't been touched in a while.

The error message is a bit odd, can you please paste the output of `clang --version` and `cat /proc/cpuinfo`?

CardealRusso, 2024-11-18:
> The error message is a bit odd, can you please paste the output of `clang --version`, `cpp -dM < /dev/null`, and `cat /proc/cpuinfo`?

[output.txt](https://github.com/user-attachments/files/17805287/output.txt)

agatti, 2024-11-18:
Well, first off it looks like you're running a 32-bits only compiler on a 64-bits CPU.  That in itself isn't the problem but using a 64-bits compiler would have solved a few headaches as the pthreads patch alone might have solved your issues.  Maybe you picked the 32-bits termux package by accident?

Regarding the error message, I can reproduce the issues you found with `MICROPY_FORCE_32BIT` set to 1 in `ports/unix/mpconfigport.mk` - I'll sort them out myself.

Still, if you want something more suited to your environment maybe you can try installing a 64-bits version of termux in the meantime.  Keep in mind that the 64-bits Arm version of MicroPython won't support native code generation (ie. `@micropython.native`, `@micropython.viper`, and inline assembler support), everything else should work the same.

CardealRusso, 2024-11-18:
> Well, first off it looks like you're running a 32-bits only compiler on a 64-bits CPU. That in itself isn't the problem but using a 64-bits compiler would have solved a few headaches as the pthreads patch alone might have solved your issues. Maybe you picked the 32-bits termux package by accident?
> 
> Regarding the error message, I can reproduce some of the issues you found with `MICROPY_FORCE_32BIT` set to 1 in `ports/unix/mpconfigport.mk` - I'll sort them out myself.
> 
> Still, if you want something more suited to your environment maybe you can try installing a 64-bits version of termux in the meantime. Keep in mind that the 64-bits Arm version of MicroPython won't support native code generation (ie. `@micropython.native`, `@micropython.viper`, and inline assembler support), everything else should work the same.

my android is running on 32bit mode for unknown reasons (maybe they consider it enough, since it doesn't have 4gb of ram.)

agatti, 2024-11-18:
This should fix the compile error about R11.  It's a hack but it works...

```diff
diff --git a/shared/runtime/gchelper_generic.c b/shared/runtime/gchelper_generic.c
index 093723137..d24a83387 100644
--- a/shared/runtime/gchelper_generic.c
+++ b/shared/runtime/gchelper_generic.c
@@ -101,6 +101,8 @@ static void gc_helper_get_regs(gc_helper_regs_t arr) {
 // Fallback implementation, prefer gchelper_thumb1.s or gchelper_thumb2.s
 
 static void gc_helper_get_regs(gc_helper_regs_t arr) {
+    #pragma clang diagnostic push
+    #pragma clang diagnostic ignored "-Wuninitialized"
     register long r4 asm ("r4");
     register long r5 asm ("r5");
     register long r6 asm ("r6");
@@ -121,6 +123,7 @@ static void gc_helper_get_regs(gc_helper_regs_t arr) {
     arr[7] = r11;
     arr[8] = r12;
     arr[9] = r13;
+    #pragma clang diagnostic pop
 }
 
 #elif defined(__aarch64__)
```

CardealRusso, 2024-11-18:
>...can you please try to run "make test" and give me the output?

this patch worked

> 938 tests performed (28422 individual testcases)
> 936 tests passed
> 40 tests skipped: builtin_next_arg2 builtin_range_binop cexample_class cexample_module cexample_subclass class_inplace_op2 cmd_parsetree cryptolib_aes128_ctr deflate_compress deflate_stream_error extra_coverage ffi_int_base ffi_int_long32 ffi_int_long64 ffi_int_types heap_locked import_mpy_native_gc io_buffered_writer machine_disable_irq machine_i2s_rate machine_rtc machine_soft_timer machine_spi_rate machine_uart_irq_txidle
> machine_uart_tx memoryview_itemsize namedtuple_asdict nanbox_smallint re_debug re_groups re_span repl_sys_ps1_ps2 ssl_ioctl ssl_poll sys_getsizeof sys_settrace_features sys_settrace_generator sys_settrace_loop sys_tracebacklimit unicode_ure
> 2 tests failed: select_poll_fd vfs_posix
>

agatti, 2024-11-18:
Great, thanks!  I'll submit the changes as a PR.

`select_poll_fd` and `vfs_posix` in theory shouldn't fail, but I suspect those are related due to the slightly non-standard environment the code runs on.  If they occur on AArch64 I'll see how hard they are to fix.

CardealRusso, 2024-11-18:
Everything is working properly now, but I just want to note here that the executable has indeed slowed down.

```py
l = 10000000
s = 0
for i in range(l):
  s = s + 1

print(s)
```

The old binary (with MICROPY_GCREGS_SETJMP):
```
time micropython test.py
10000000

real    0m3.360s
user    0m3.310s
sys     0m0.010s
```

Now (i even used -O3 and -flto):
```
 time ./micropython test.py
10000000

real    0m4.402s
user    0m4.370s
sys     0m0.000s

```

dpgeorge, 2024-11-18:
> ```python
> l = 10000000
> s = 0
> for i in range(l):
>   s = s + 1
> 
> print(s)
> ```

This is not the best way to measure performance in MicroPython because it's subject to caching effects.  If you put all that in a function and call the function then that would be better, because `s` will be a local variable rather than a dict entry.

You can also use `test/run-perfbench.py` to run the performance benchmark test suite.

CardealRusso, 2024-11-19:
Sorry. I actually forgot to remove MICROPY_GCREGS_SETJMP 1
>write to reserved register 'R11'  

This error persist, but now at link time.

```
CC ../shared/runtime/gchelper_generic.c
LINK build/mpy-cross
ld.lld: error: write to reserved register 'R11' at line 4257
gcc: error: linker command failed with exit code 1 (use -v to see invocation)

```

agatti, 2024-11-19:
It looks like you forgot to revert those changes and apply the diff in https://github.com/micropython/micropython/issues/16259#issuecomment-2484083265

CardealRusso, 2024-11-19:
I skipped your first path in gchelper_generic and manually added the pragmas in gc_helper_get_regs 
It compiled without any errors. I assume it's working now? Although there is no noticeable change in performance, this now seems to be enough.

agatti, 2024-11-19:
If it compiles and links without errors and `make test` doesn't report any more failed tests than it used to, I'd say it's all for now.  Unless something else shows up, all that's left to do is to wait for #16261 and #16264 to be merged :)

CardealRusso, 2024-11-19:
Thank you for your help. Using pragma added some flaws to the test, I think in the end I'll stick with MICROPY_GCREGS_SETJMP

`5 tests failed: asyncio_get_event_loop asyncio_new_event_loop builtin_hasattr select_poll_fd vfs_posix
`

However, there was a slight improvement in performance. Although small, it was noticeable.
Using 
> 
> ```python
> @micropython.native
> def f():
>   l = 10000000
>   s = 0
>   for i in range(l):
>     s = s + 1
>   print(s)
> 
> f()
> ```

Old:
```
real    0m1.097s
user    0m1.040s
sys     0m0.030s
```

Now:
```
real    0m0.963s
user    0m0.950s
sys     0m0.000s
```

agatti, 2024-11-19:
No worries, glad everything worked out in the end.

dpgeorge, 2025-04-22:
This issue should now be fully resolved:
- GC helper build issue fixed by 5784714f734caf3713ff880ffe100bbbec59216e
- cache build warning fixed by 2a8f6047ff5c94bba7ace738a14b572cc8553251
- pthread cancel issue fixed by 38a3873310c571c32e52edc2a7d4de7c80ff35d5

closed 2025-04-22 by dpgeorge