micropython/micropython #16266: f-string too eager ?

WebReflection, 2024-11-19:
### Port, board and/or hardware

Arduino Nano ESP32 or MicroPython WASM

### MicroPython version

MicroPython v1.24.0

### Reproduction

```python
import json

b = "b"

# throws
f'f{json.dumps(f"a{b}c")}'

# works
value = json.dumps(f"a{b}c")
f'f{value}'
```

### Expected behaviour

I wasn't expecting the need to create a reference for a value passed as interpolation and I've thought `f` was weird in that regard, then I've tested on my Linux Python and everything works as expected where it *throws* in MicroPython instead.

### Observed behaviour

It looks like the `f` in MicroPython is a bit too greedy ... I believed that scanning for nested `{` and `}` would solve the issue as it needs to ignore those curly braces but then again, even trying to make some curly explicit:

```python
f'f{json.dumps("a{{b}}c")}'
```

there is a Syntax error, but also that would produce an invalid f string in C-Python: `'f"a{{b}}c"'` so that such string can't be evaluated anymore.

### Additional Information

I've no idea if this is a duplicate but the example is fairly simple and somehow weird to reason about ... the goal of the code is to create strings that can be evaluated without needing to enforce `f"..."` around content

### Code of Conduct

Yes, I agree

andrewleech, 2024-11-19:
I personally didn't realise nested f-strings are supported in cpython! The micropython f-string implementation is optimised for code size so doesn't support all features in desktop / cpython. This case isn't actually listed here but these documented difference give some idea of similar limitations: https://docs.micropython.org/en/latest/genrst/core_language.html#f-strings-don-t-support-concatenation-with-adjacent-literals-if-the-adjacent-literals-contain-braces

WebReflection, 2024-11-20:
@andrewleech the workaround is fairly trivial but thanks for sharing that page, I never looked at it and it's full of good advices / use cases where uPy is different, which is a great-to-know resource.

You can close this if part of the same group of incompatibilities, or maybe that page could add this case too?

andrewleech, 2024-11-20:
Ah yes! Iirc that docs page is directly generated from the tests in this dir: https://github.com/micropython/micropython/tree/master/tests/cpydiff

Eg the doc paragraph I linked to previously is from this test: https://github.com/micropython/micropython/blob/master/tests/cpydiff/core_fstring_concat.py