micropython/micropython #16267: range() objects don't have method index()

mirko, 2024-11-19:
### Port, board and/or hardware

board unspecific

### MicroPython version

MicroPython 1.21

### Reproduction

`range(5,50,5).index(25)`

### Expected behaviour

Expected:
```
>>> range(5,50,5).index(25)
4
```

### Observed behaviour

Actual:
```
>>> range(5,50,5).index(25)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'range' object has no attribute 'index'
>>>
```

### Additional Information

Difference to CPython - and as far as I can see undocumented.

### Code of Conduct

Yes, I agree

scruss, 2024-12-09:
Yes, this is an undocumented difference.

`list(range(5,50,5)).index(25)` is a workaround which also works on CPython

mirko, 2024-12-09:
> Yes, this is an undocumented difference.
> 
> `list(range(5,50,5)).index(25)` is a workaround which also works on CPython

It only works if you can afford it - e.g. memory wise. The whole idea to use it on a `range()`-object is *not* to expand it.