micropython/micropython #16283: docs:  I2C Scan on ESP32-S3 Returns No Devices

hardikofficial963, 2024-11-21:
### Description

**Issue:**
 I2C Scan on ESP32-S3 Returns No Devices

**Description:**
I am trying to scan for I2C devices on the ESP32-S3 using MicroPython but getting no devices detected, even though the hardware works with the Arduino Wire Scan example.

**MicroPython Version:**
v1.24.0 (2024-10-25)

**Hardware Details:**
- Microcontroller: ESP32-S3
- I2C Device: PCA9685
- Pins Used:
     - SCL: GPIO 19
     - SDA: GPIO 20

**Code:**

        from machine import Pin, SoftI2C
        import time
        
        def scan_i2c_addresses():
            # Initialize I2C
            i2c = SoftI2C(sda=Pin(19), scl=Pin(20), freq=400000)
            
            print("Scanning I2C addresses...")
            devices = i2c.scan()
            
            if len(devices) == 0:
                print("No I2C devices found!")
            else:
                print(f"Found {len(devices)} I2C devices:")
                for device in devices:
                    print(f"Device address: {hex(device)}")
        
        if __name__ == '__main__':
            scan_i2c_addresses()

    
**Observations:**

- Expected Behavior: Detect the PCA9685 at address 0x40.
- Actual Behavior: "No I2C devices found!" message is printed.

**Troubleshooting Performed**
1. Tested the same hardware and wiring configuration using the Arduino Wire Scan example. The device at 0x40 is detected successfully.
2. Verified the wiring connections:
    SCL → GPIO 19
    SDA → GPIO 20
3. Proper pull-up resistors in place.

**Request:**

- Is there a known issue with SoftI2C on ESP32-S3 in MicroPython?
- Are there any additional configurations or debugging steps I should try?

Any insights would be appreciated.

[image]

### Code of Conduct

Yes, I agree

hardikofficial963, 2024-11-21:
> Have you got pullup resistors on the I2C lines?

Yes, I have pullup resistors on the I2C lines.

peterhinch, 2024-11-22:
I can't replicate this. I had to change the pin numbers as my ESP32-S3 boards don't bring out pins 19 and 20. Also I don't have a PCA9685 so I used another I2C target. However with the same firmware build soft I2C does work correctly.

I would suspect hardware. Have you checked that the PCA9685 is receiving power? Some ESP32-S3 boards have a switched 3.3V supply that needs to be turned on in code.

hardikofficial963, 2024-11-23:
> Trou

Sir, I have checked the same hardware with Arduino Wire Scan example it is working there. I am sure Hardware and wiring configuration is fully functioning.

yoann-darche, 2024-11-27:
Can you give a schema of the hardware, how do you power the PCA ?

Malplaced, 2024-11-27:
Have you just swapped sda and scl ?

I see:

Pins Used:
SCL: GPIO 19
SDA: GPIO 20
and
Verified the wiring connections:
SCL → GPIO 19
SDA → GPIO 20

**But** then in program:
sda=Pin(19), scl=Pin(20)

Best reg. Soren

jonnor, 2024-12-27:
Issue not reproduced and the above seems to be a likely bug. Proposing to close.

closed 2025-01-09 by jonnor