micropython/micropython #16284: User C module fails to link in esp32 port

stinos, 2024-11-21:
### Port, board and/or hardware

esp32 port

### MicroPython version

Latest master

### Reproduction

Change micropython/examples/usercmodule/cppexample/example.cpp to

```
extern "C" {
#include <examplemodule.h>
#include <py/objstr.h>
}

#include <cstring>
#include <string>

namespace fun {
    inline mp_obj_t RaiseException(const mp_obj_type_t *exc_type, const char *msg) {
        mp_obj_exception_t *o = mp_obj_malloc_var(mp_obj_exception_t, args, mp_obj_tuple_t, 0, exc_type);
        o->traceback_data = nullptr;
        o->args = reinterpret_cast < mp_obj_tuple_t * > (MP_OBJ_TO_PTR(mp_obj_new_tuple(1, nullptr)));
        o->args->items[ 0 ] = mp_obj_new_str(msg, std::strlen(msg));
        nlr_raise(MP_OBJ_FROM_PTR(o));
    }
}

struct F {
    static const char *Foo() {
        return "Foo";
    }
};

extern "C" {
void foo() {
    fun::RaiseException(&mp_type_TypeError, (std::string("oops") + F::Foo()).data());
}

// Here we implement the function using C++ code, but since it's
// declaration has to be compatible with C everything goes in extern "C" scope.
mp_obj_t cppfunc(mp_obj_t a_obj, mp_obj_t b_obj) {
    // The following no-ops are just here to verify the static assertions used in
    // the public API all compile with C++.
    MP_STATIC_ASSERT_STR_ARRAY_COMPATIBLE;
    if (mp_obj_is_type(a_obj, &mp_type_BaseException)) {
    }

    // Prove we have (at least) C++11 features.
    const auto a = mp_obj_get_int(a_obj);
    const auto b = mp_obj_get_int(b_obj);
    const auto sum = [&]() {
        return mp_obj_new_int(a + b);
    } ();
    // Prove we're being scanned for QSTRs.
    mp_obj_t tup[] = {sum, MP_ROM_QSTR(MP_QSTR_hellocpp)};
    return mp_obj_new_tuple(2, tup);
}
}
```

So this is like the original code, with `foo` and `RaiseException` added. There's nothing very special here and this builds fine on a couple of Makefile-based ports I tried, e.g.

```
>> cd ports/unix
>> make USER_C_MODULES=../../examples/usercmodule/cppexample/micropython.cmake
...
LINK build-standard/micropython
   text	   data	    bss	    dec	    hex	filename
 764743	  69328	   7088	 841159	  cd5c7	build-standard/micropython
```

### Expected behaviour

Should build on all ports supporting C++ compilation.

### Observed behaviour

Building following the standard instructions from https://docs.micropython.org/en/latest/develop/cmodules.html#basic-example this does not build on the ESP32 port using docker:

```
>> docker run --rm -v ${PWD}:/project -w /project -it espressif/idf:v5.2
>> cd ports/esp32
>> make USER_C_MODULES=../../../examples/usercmodule/cppexample/micropython.cmake
...
/opt/esp/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32/libmain_esp32.a(example.cpp.obj): in function `cppfunc':
/project/micropython/examples/usercmodule/cppexample/example.cpp:34: multiple definition of `cppfunc'; CMakeFiles/micropython.elf.dir/project/micropython/examples/usercmodule/cppexample/example.cpp.obj:example.cpp:(.text+0xf8): first defined here
/opt/esp/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main_esp32/libmain_esp32.a(example.cpp.obj): in function `foo':
/project/micropython/examples/usercmodule/cppexample/example.cpp:28: multiple definition of `foo'; CMakeFiles/micropython.elf.dir/project/micropython/examples/usercmodule/cppexample/example.cpp.obj:example.cpp:(.text+0x1c): first defined here
collect2: error: ld returned 1 exit status
```

It's also fairly surprising that for all user C module files there are 2 object files, all build logs and compile_commands.json etc seem to show these files getting built twice, whereas the rest of the micropython source tree is built only once.

@Gadgetoid FYI

### Code of Conduct

Yes, I agree

Tobinator32, 2024-11-26:
Hey,
ive tried looking into the makefile and added `message(STATUS "${SOURCES_VARNAME}")` and message(STATUS "${lib_sources} ls") after line 15 to get all the sources.
There are no duplicates in here.
Also this would not explain why a file compiles normally and when adding the same line again it fails linking with all functions etc. in this translation unit are duplicated.

ned-pcs, 2024-12-01:
Just out of curiosity, why did you need `-fexceptions` ?

stinos, 2024-12-01:
To reproduce the issue with with this self-contained sample code :)

Originated from building https://github.com/stinos/micropython-wrap into a user C module on ESP32. But even after removing the need for `-fexceptions` for it, the linking issue persists.

stinos, 2024-12-05:
Did some more testing and managed to build a working binary, once. Steps were something like (I'm going to use micropython-wrap here because that always reproduces the issue, whereas the one I originally posted doesn't always):
```
cd /path/to/micropython/..
git clone git@github.com:stinos/micropython-wrap.git
git -C micropython-wrap checkout origin/32bitlong
cd ./micropython/ports/esp32
```
then add to esp32_common.cmake
```
list(APPEND MICROPY_SOURCE_QSTR
    ${MICROPY_DIR}/../micropython-wrap/tests/cmodule.c
)
```
then `make clean && make`, which will fail with `undefined reference to 'upywraptest_module'` because that's indeed not in the build, but this step is needed to find the root pointer definition so it is in genhdr/ already, because the next step will build the actual source but for that'll run before qstr generation and I haven't figured out how to not do that. Now add in esp32_common.cmake:
```
add_library(upywraptest OBJECT
    ${MICROPY_DIR}/../micropython-wrap/tests/cmodule.c
    ${MICROPY_DIR}/../micropython-wrap/tests/module.cpp
)
get_property(UP_INC_DIRS TARGET ${MICROPY_TARGET} PROPERTY INCLUDE_DIRECTORIES)
target_include_directories(upywraptest PRIVATE
    ${MICROPY_DIR}/../micropython-wrap
    ${UP_INC_DIRS}
    /opt/esp/idf/components/driver/i2s/include
    /opt/esp/idf/components/esp_wifi/include
    /opt/esp/idf/components/esp_event/include
)
target_compile_definitions(upywraptest PRIVATE -DUPYWRAP_FULLTYPECHECK=0)
target_link_libraries(${MICROPY_TARGET} upywraptest)
```
and this'll build and link and run.

So: using `OBJECT` (`STATIC` also works btw) and `PRIVATE` (instead of `INTERFACE` and `PUBLIC` as the micropython user C module examples do) seems to be one way to make sure that the files added are only built and linked once. However
- using `OBJECT` also means the include directories from the 'parent' i.e. micropython target are not available so I added a workaround by getting them from that target, but as can be seen that is somehow not sufficient and I had to add 3 directories manually
- running `make` again, even though nothingh changed, it'll now fail with
```
ninja: error: dependency cycle: esp-idf/main_esp32/CMakeFiles/upywraptest.dir/project/projects/micropython-wrap/tests/cmodule.c.obj -> /project/projects/micropython/ports/esp32/build-ESP32_GENERIC/genhdr/moduledefs.h -> genhdr/moduledefs.collected -> genhdr/moduledefs.split -> esp-idf/main_esp32/upywraptest -> esp-idf/main_esp32/CMakeFiles/upywraptest.dir/project/projects/micropython-wrap/tests/cmodule.c.obj
```

<br/><br/>
I searched around for what `add_library` is actually supposed to do, and it all seems to make sense, but since it doesn't work here and I don't actually know CMake I figured a simpler more straightforward way would probably work, i.e. adding sources in the same way as the core sources since that all just works:
- change micropython-wrap/micropython-wrap.cmake to
```
list(APPEND MICROPY_SOURCE_LIB
    ${CMAKE_CURRENT_LIST_DIR}/tests/cmodule.c
    ${CMAKE_CURRENT_LIST_DIR}/tests/module.cpp
)

target_compile_definitions(usermod INTERFACE -DUPYWRAP_FULLTYPECHECK=0)
```
- build with `make USER_C_MODULES=../../../../micropython-wrap/micropython-wrap.cmake`: just works

<br/><br/>
I'm going to leave it at that, hopefully someone with CMake knowledge can chime in and compile one of the presented methods into something generic.

Tobinator32, 2024-12-11:
Hey,
this seems promising to me. Ive read several other cmake related issues regarding linking against an INTERFACE target but it seems the user c modules requiring this. Maybe this is one part of the problem. But until it is sorted out why it has to be done that way your workaround seems viable. Thanks for your effort. Will try it the next days and see it its working. Maybe some ESP Port responsible will loop in and can explain why this is neccessary.
Kind Regards Tobi

projectgus, 2026-02-10:
No worries, thanks for the update. I'm going to close this issue then.