micropython/micropython #16308: ESP32: link fails on 78d017fc4 master branch generic build because of lwip_patch

ned-pcs, 2024-11-26:
### Port, board and/or hardware

ESP32 port, ESP32_S3_GENERIC-SPIRAM_OCT

### MicroPython version

Build environment: Mac OS 15.1 on Apple M1, ESP-IDF v5.2.2, Python 3.13

### Reproduction

Started from a fresh clone of master (at 78d017fc4).
```
cd micropython
git submodule update --init
make -C mpy-cross
cd ports/esp32
make BOARD=ESP32_S3_GENERIC BOARD_VARIANT=SPIRAM_OCT submodules
make BOARD=ESP32_S3_GENERIC BOARD_VARIANT=SPIRAM_OCT
```
Here's the resultant log file:
[idf_py_stdout_output_91679.txt](https://github.com/user-attachments/files/17925048/idf_py_stdout_output_91679.txt)

### Expected behaviour

Expected to successfully build.

### Observed behaviour

Link failed, apparently missing `__wrap_memp_malloc` and `__wrap_memp_free` from `lwip_patch.c`. However, this file was in fact compiled:
```
[ned@Neds-mini ports/esp32] % xtensa-esp32-elf-nm build-ESP32_S3_GENERIC-SPIRAM_OCT/esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/__/lwip_patch.c.obj
         U __real_memp_free
         U __real_memp_malloc
00000000 T __wrap_memp_free
00000000 T __wrap_memp_malloc
00000000 b active_tcp_pcbs
```
It was also linked into the static library `build-ESP32_S3_GENERIC-SPIRAM_OCT/esp-idf/main_esp32s3/libmain_esp32s3.a`.
This library was included in the link command.

### Additional Information

Not that this has anything to do with the build failure, but I did have to do the initial `git submodule update --init` at the top level,  as doing `make submodules` in the `ports/esp32` directory did nothing after a fresh clone.

### Code of Conduct

Yes, I agree

projectgus, 2024-11-26:
Thanks for the very clear report, @ned-pcs. Is there a `micropython.map` file in the build directory? If so, could you attach it here please?

ned-pcs, 2024-11-26:
Here's the `micropython.map` file.
[micropython.map.zip](https://github.com/user-attachments/files/17926919/micropython.map.zip)

projectgus, 2024-11-27:
Thanks @ned-pcs. *Flips "days since I misunderstood linker behaviour" sign back to zero*.

I can't reproduce the build failure here, but you could please try applying this patch and see if the link succeeds?

```diff
diff --git i/ports/esp32/esp32_common.cmake w/ports/esp32/esp32_common.cmake
index 9d51a03aa8..6a0b18a331 100644
--- i/ports/esp32/esp32_common.cmake
+++ w/ports/esp32/esp32_common.cmake
@@ -246,6 +246,7 @@ target_link_libraries(${MICROPY_TARGET} usermod)
 idf_build_set_property(LINK_OPTIONS "-Wl,--wrap=esp_panic_handler" APPEND)
 
 # Patch LWIP memory pool allocators (see lwip_patch.c)
+idf_build_set_property(LINK_OPTIONS "-u memp_malloc" APPEND)
 idf_build_set_property(LINK_OPTIONS "-Wl,--wrap=memp_malloc" APPEND)
 idf_build_set_property(LINK_OPTIONS "-Wl,--wrap=memp_free" APPEND)
```

ftylitak, 2024-11-29:
Hi @ned-pcs,

hopping that you will be able to reproduce the issue, in my case i get this linker error when I try to deactivate MDNS support with the following change: 

```diff
diff --git a/ports/esp32/boards/ESP32_GENERIC_S3/mpconfigboard.h b/ports/esp32/boards/ESP32_GENERIC_S3/mpconfigboard.h
index c714fe2c2..5a6e3069c 100644
--- a/ports/esp32/boards/ESP32_GENERIC_S3/mpconfigboard.h
+++ b/ports/esp32/boards/ESP32_GENERIC_S3/mpconfigboard.h
@@ -9,3 +9,10 @@
 
 #define MICROPY_HW_I2C0_SCL                 (9)
 #define MICROPY_HW_I2C0_SDA                 (8)
+
+// disable default mDNS
+#define LWIP_MDNS_RESPONDER (0)
+
+#define MICROPY_HW_ENABLE_MDNS_RESPONDER 0
+#define MICROPY_HW_ENABLE_MDNS_QUERIES 0
+
```

Your proposed solution did not change the behaviour.

ned-pcs, 2024-11-29:
> Thanks @ned-pcs. _Flips "days since I misunderstood linker behaviour" sign back to zero_.
> 
> I can't reproduce the build failure here, but could you please try applying this patch and see if the link succeeds?
> 
> ```diff
> diff --git i/ports/esp32/esp32_common.cmake w/ports/esp32/esp32_common.cmake
> index 9d51a03aa8..6a0b18a331 100644
> --- i/ports/esp32/esp32_common.cmake
> +++ w/ports/esp32/esp32_common.cmake
> @@ -246,6 +246,7 @@ target_link_libraries(${MICROPY_TARGET} usermod)
>  idf_build_set_property(LINK_OPTIONS "-Wl,--wrap=esp_panic_handler" APPEND)
>  
>  # Patch LWIP memory pool allocators (see lwip_patch.c)
> +idf_build_set_property(LINK_OPTIONS "-u memp_malloc" APPEND)
>  idf_build_set_property(LINK_OPTIONS "-Wl,--wrap=memp_malloc" APPEND)
>  idf_build_set_property(LINK_OPTIONS "-Wl,--wrap=memp_free" APPEND)
> ```

I get the same link errors with this change. Also tried adding `-u memp_free` but that didn't help either.

ned-pcs, 2024-11-29:
> Hi @ned-pcs,
> 
> hopping that you will be able to reproduce the issue, in my case i get this linker error when I try to deactivate MDNS support with the following change:

I also will be deactivating MDNS support but the linker errors I'm getting are even with the stock `ESP32_GENERIC_S3` build options.

ned-pcs, 2024-11-29:
OK, I looked into it a bit more. @ftylitak I did indeed have my own board definition where I had disabled `LWIP_MDNS_RESPONDER` as you showed. @projectgus the build succeeds with the stock build options but fails with the MDNS responder disabled as @ftylitak showed.

projectgus, 2024-12-02:
Thanks @ftylitak and @ned-pcs for the extra information. Was able to reproduce here as well, and have opened #16349. Are you able to confirm the fix, please?

ned-pcs, 2024-12-05:
@projectgus link succeeds. Thanks for fixing this!

PR #16349 (2024-12-10): esp32: Fix link failure due to library link order.
### Summary

When a wrapped symbol is provided in its own file, it's possible for the linker to skip that file entirely and not return to it depending on the order of libraries passed on the linker command line.

This is because these wrapped symbols create linker cycles (libmain_espXX depends on liblwip but liblwip now also depends on libmain for the wrapped functions in lwip_patch.c, for example.)

Linker failure for symbols in lwip_patch.c was reproducible if mDNS was disabled in the board configuration. Closes #16308.

This commit adds an  [... cut]
closed 2024-12-05 by ned-pcs