micropython/micropython #16312: 32bit unix port build fails to find module in same directory

stinos, 2024-11-27:
### Port, board and/or hardware

unix port on WSL, source code on a mounted windows directory

### MicroPython version

Latest master

### Reproduction

- clone and cd to unix port directory
- add a 32bit directory under variants containing

micropython/ports/unix/variants/32bit/mpconfigvariant.h:
```
#define MICROPY_CONFIG_ROM_LEVEL (MICROPY_CONFIG_ROM_LEVEL_EXTRA_FEATURES)
#define MICROPY_DEBUG_PRINTERS (1)
#define MICROPY_FLOAT_IMPL (MICROPY_FLOAT_IMPL_DOUBLE)
#define MICROPY_LONGINT_IMPL (MICROPY_LONGINT_IMPL_MPZ)
```

micropython/ports/unix/variants/32bit/mpconfigvariant.mk:
```
MICROPY_FORCE_32BIT = 1
MICROPY_PY_FFI = 0
MICROPY_PY_BTREE = 0
```
- build standard and 32bit variants, so that's `make` and `make VARIANT=32BIT`
- create a directory frob containing

micropython/ports/unix/frob/foo.py:
```
# empty
```
micropython/ports/unix/frob/bar.py:
```
import foo
```
- run `./build-standard/micropython ./frob/bar.py` -> no output, exepected
- run `./build-32bit/micropython ./frob/bar.py` -> output:

```
Traceback (most recent call last):
  File "./frob/bar.py", line 1, in <module>
ImportError: no module named 'foo'
```

### Expected behaviour

Behaviour should be the same as standard variant

### Observed behaviour

`import foo` fails to find `foo` even though it's in the same directory as the module where the import statement is

### Additional Information

Could be related to other features in this 32bit variant, though setting `MICROPY_FORCE_32BIT = 0` does make the problem go away. I'm caught up in getting other things working at the moment so don't have time to look into thi. Could just as well be a WSL bug.

### Code of Conduct

Yes, I agree

agatti, 2024-11-29:
It seems to not be a problem on my Linux machine, unless I followed your steps incorrectly:

```
~/src ➜ cat micropython/ports/unix/variants/32bit/mpconfigvariant.h
#define MICROPY_CONFIG_ROM_LEVEL (MICROPY_CONFIG_ROM_LEVEL_EXTRA_FEATURES)
#define MICROPY_DEBUG_PRINTERS (1)
#define MICROPY_FLOAT_IMPL (MICROPY_FLOAT_IMPL_DOUBLE)
#define MICROPY_LONGINT_IMPL (MICROPY_LONGINT_IMPL_MPZ)

~/src ➜ cat micropython/ports/unix/variants/32bit/mpconfigvariant.mk
MICROPY_FORCE_32BIT = 1
MICROPY_PY_FFI = 0
MICROPY_PY_BTREE = 0

~/src ➜ ls micropython/ports/unix/frob/
bar.py  foo.py

~/src ➜ cat micropython/ports/unix/frob/foo.py                                                                                     
# empty

~/src ➜ cat micropython/ports/unix/frob/bar.py
import foo

~/src ➜ cd micropython/ports/unix && rm -rf build-*

~/src/micropython/ports/unix ➜ make -j
# ...
LINK build-standard/micropython
   text    data     bss     dec     hex filename
 751581   69312    7088  827981   ca24d build-standard/micropython

~/src/micropython/ports/unix ➜ make VARIANT=32bit -j
# ...
LINK build-32bit/micropython
   text    data     bss     dec     hex filename
 629482   30712    5736  665930   a294a build-32bit/micropython

~/src/micropython/ports/unix ➜ git log -1 | head -n 1
commit eb80b04944e01d437efc2fa29189ad1e2bdd997f

~/src/micropython/ports/unix ➜ ./build-standard/micropython ./frob/bar.py

~/src/micropython/ports/unix ➜ ./build-32bit/micropython ./frob/bar.py

~/src/micropython/ports/unix ➜
```

I don't have a windows machine to test that myself under WSL, unfortunately.

stinos, 2024-11-30:
> It seems to not be a problem on my Linux machine, unless I followed your steps incorrectly:

That's correct, but it's not a general unix port problem apparently: the CI also builds and tests a 32bit unix coverage variant and that works fine.. So it's getting more likely this is a WSL issue.

stinos, 2024-12-02:
Turns out the C `stat` call in vfs_posix.c fails, on a clearly existing file, so really seems some WSL issue. And it's not just `stat` but other filesystem APIs as well, for example:

```
> cd ./frob
> ../build-standard/micropython
>>> import os
>>> os.listdir('.')
['bar.py', 'foo.py']
```

but

```
> ../build-32bit/micropython
>>> import os
>>> os.listdir('.')
[]
>>> os.listdir('/mnt/c')
[]
```

so closing this.

closed 2024-12-02 by stinos

dpgeorge, 2024-12-04:
> Note: solution is using `-D_FILE_OFFSET_BITS=64` which is fairly standard and even included in the manpages for stat, and by the looks of it will become the default at one point. So while this could only be reproduced on WSL so far, there might be other systems where it occurs as well.

Oh wow, that's exactly what I need!  See https://github.com/micropython/micropython/pull/16354