micropython/micropython #16335: Found 1 MCU chip W801 W806 with very rich peripheral resources

yfyungit, 2024-11-30:
### Description

I think this chip has high development potential and value. If the mircopython community can support this MCU, it will be a good choice.

MCU official website : [https://www.winnermicro.com/html/1/156/158/569.html](url)

`Chip appearance
QFN56 package, 6mm x 6mm
MCU features
Integrated 32-bit XT804 processor, operating frequency 240MHz, built-in DSP, floating-point unit and security engine
Built-in 2MB Flash, 288KB RAM Integrated PSRAM interface, supports up to 64MB external PSRAM memory
Integrated 6-way UART high-speed interface
Integrated 4-way 12-bit ADC, maximum sampling rate 1KHz
Integrated 1 high-speed SPI interface, supports up to 50MHz
Integrated 1 master/slave SPI interface
Integrated 1 SDIO_HOST interface, supports SDIO2.0, SDHC, MMC4.2
Integrated 1 SDIO_DEVICE, supports SDIO2.0, maximum throughput 200Mbps
Integrated PSRAM expansion interface, supports up to 64Mbyte
Integrated 1 I2C controller
Integrated GPIO controller, supports up to 44 GPIOs
Integrated 5-way PWM interface
Integrated 1 Duplex I2S controller
Integrated LCD controller, supports 4x32 interface
Integrated 1 7816 interface
Integrated 15 Touch Sensors
Security features
MCU has built-in Tee security engine, the code can distinguish between secure world/non-secure world
Integrated SASC/TIPC, memory and internal modules/interfaces can configure security attributes to prevent non-secure code access
Enable firmware signature mechanism to achieve secure Boot/upgrade
With firmware encryption function to enhance code security
Firmware encryption keys are distributed using asymmetric algorithms to enhance key security
Hardware encryption module: RC4, AES128, DES/3DES, SHA1/MD5, CRC32, 2048 RSA, true random number generator
Wi-Fi features
Support GB15629.11-2006, IEEE802.11 b/g/n
Support Wi-Fi WMM/WMM-PS/WPA/WPA2/WPS
Support EDCA channel access method
Support 20/40M bandwidth working mode
Support STBC, GreenField, Short-GI, support reverse transmission
Support AMPDU, AMSDU
Support IEEE802.11n MCS 0~7, MCS32 physical layer transmission rate gear, the maximum transmission rate is 150Mbps
Support Short Preamble when sending at 2/5.5/11Mbps rate
Support HT-immediate Compressed Block Ack, Normal Ack, No Ack response mode
Support CTS to self
Support Station, Soft-AP, Soft-AP/Station functions
Bluetooth features
Integrated Bluetooth baseband processor/protocol processor, support BT/BLE dual-mode working mode, support BT/BLE4.2 protocol
Support Bluetooth network configuration
Low power mode
3.3V single power supply
Support Wi-Fi energy-saving mode power consumption management
Support working, sleeping, standby, and shutdown working modes
Standby power consumption is less than 10uA`

### Implementation

I hope the MicroPython maintainers or community will implement this feature

### Code of Conduct

Yes, I agree

robert-hh, 2024-11-30:
Support for new devices is usually driven by the community. The maintainers have their primary focus on the MicroPython core mechanisms. So If you like to support this device, start working. Usually a condition for a device to be integrated into the core set of devices it needs good and continuing support by the MCU manufacturing company, someone supporting the port and a good offering of boards with that MCU. Besides the ports listed on the web site there are quiet a few port for other devices made by board vendors (e.g. XBEE) of people from the community. Personally I maintain together with others a port for the W60x device, the predecessor of the W80x (https://github.com/robert-hh/micropython/tree/w60x and https://github.com/robert-hh/Shared-Stuff/tree/master/w600_firmware). I had a short glance at the W80x a while ago bit decided not to start an implementation. Do you know if English support documentation for the MCU and the library is available from the vendor?

yfyungit, 2024-11-30:
@robert-hh The chip supplier is in China, and I also live in China, so I can easily assist in obtaining relevant data files. If necessary, I can also assist in obtaining the circuit board used in the development of the device.