micropython/micropython #16370: Problem with 'machine.Pin()' on Teensy 4.1 for the GPIOs

MicroControleurMonde, 2024-12-06:
### Port, board and/or hardware

Platform : Teensy 4.1 

### MicroPython version

MicroPython v1.24.0 on 2024-10-25; Teensy 4.1 with MIMXRT1062DVJ6A

### Reproduction

from machine import Pin

p0 = Pin('D0', Pin.OUT) # create output pin on GPIO0
p0.on()                 # set pin to "on" (high) level
p0.off()                # set pin to "off" (low) level
p0.value(1)             # set pin to on/high

p2 = Pin('D2', Pin.IN)  # create input pin on GPIO2
print(p2.value())       # get value, 0 or 1

p4 = Pin('D4', Pin.IN, Pin.PULL_UP) # enable internal pull-up resistor
p5 = Pin('D5', Pin.OUT, value=1) # set pin high on creation

p6 = Pin(pin.cpu.GPIO_B1_15, Pin.OUT) # Use the cpu pin name.

### Expected behaviour

The code shoud create and configures several GPIO pins on the MCU in input and output mode.
The output should be visible on the console for the value of 'print(p2.value())'.

### Observed behaviour

1
Traceback (most recent call last):
  File "<stdin>", line 14, in <module>
NameError: name 'pin' isn't defined

### Additional Information

I have tried many other scripts using 'machine.Pin () ' and systematically I come across this error : **'pin' isn't defined**
I have never seen this with ESP32, RP2040 or Pyboard...

The problem is specific to the use of machine.Pin() for the  GPIOs.
Direct access to the registers via the memory addresses of these same GPIOs poses no problem.
I went to search the NXP doc and I pulled out some info about GPIO.

| Start address     | End address     | Size | Region | NIC port |
|-------------------|-----------------|------|--------|----------|
|         400C_0000 | 400C_3FFF       | 16KB | AIPS-1 |  GPIO5   |
|         401C_4000 | 401C_7FFF       | 16KB | AIPS-2 |  GPIO4   |
|         401C_0000 | 401C_3FFF       | 16KB | AIPS-2 |  GPIO3   |
|         401B_C000 | 401B_FFFF       | 16KB | AIPS-2 |  GPIO2   |
|         401B_8000 | 401B_BFFF       | 16KB | AIPS-2 |  GPIO1   |
|         4200_C000 | 4200_FFFF       | 16KB | AIPS-5 |  GPIO9   |
|         4200_8000 | 4200_BFFF       | 16KB | AIPS-5 |  GPIO8   |
|         4200_4000 | 4200_7FFF       | 16KB | AIPS-5 |  GPIO7   |
|         4200_0000 | 4200_3FFF       | 16KB | AIPS-5 |  GPIO6   |

### Code of Conduct

Yes, I agree

robert-hh, 2024-12-06:
Use Pin.cpu.GPIO_B1_15 I stead of pin.cpu.GPIO_B1_15

MicroControleurMonde, 2024-12-06:
And it works ...
Thanks Robert.

MicroControleurMonde, 2024-12-06:
I just figured it out !

The management/call of ADCs and GPIOs seems different between the ESP32, the Teensy or the RP2040.
Am I not mistaken?

robert-hh, 2024-12-06:
GPIO via the Pin class should be basically the same, ADC is slightly different.

closed 2024-12-06 by robert-hh

MicroControleurMonde, 2024-12-06:
Could you tell me where it is documented (the differences) if possible ?

robert-hh, 2024-12-06:
https://docs.micropython.org/en/latest/esp32/quickref.html#adc-analog-to-digital-conversion and
https://docs.micropython.org/en/latest/mimxrt/quickref.html#adc-analog-to-digital-conversion

MicroControleurMonde, 2024-12-06:
Lovely man.
Danke sehr !