micropython/micropython #16373: UART Ch 1 GPIO 4&5 on RP2040 Pico stopped working with 1.24 releases

gobbyo, 2024-12-07:
### Port, board and/or hardware

RP2040 Pico UART CH1 GPIO4&5

### MicroPython version

The following have been tested and UART CH1 does not work on GPIO 4&5 (I didn't try other pins) on the RP2040 Pico:
[v1.24.1 (2024-11-29) .uf2](https://micropython.org/resources/firmware/RPI_PICO-20241129-v1.24.1.uf2)
[v1.24.0 (2024-10-25) .uf2](https://micropython.org/resources/firmware/RPI_PICO-20241025-v1.24.0.uf2)
[v1.25.0-preview.81.g2c80d3699 (2024-12-06) .uf2](https://micropython.org/resources/firmware/RPI_PICO-20241206-v1.25.0-preview.81.g2c80d3699.uf2)

### Reproduction

def testUART(channel, txpin, rxpin, payload="Hello from the Pico!"):
    uart = UART(channel)
    uart.init(channel, 9600, rx=Pin(rxpin), tx=Pin(txpin))
    uart.write(payload.encode('utf-8'))
    time.sleep(.2)
    if uart.any() > 0:
        b = bytearray(payload, 'utf-8')
        size = uart.readinto(b)
        if b != None:
            s = b.decode('utf-8')
            print("UART{0}: {1}(size={2})".format(channel, s, size))
    uart.deinit()

def main():
        testUART(0,0,1)
        testUART(1,4,5)

### Expected behaviour

Per the example code above, I expected the UART TX of "Hello from the Pico" text to be received.  The code works for all releases prior to 1.24.

### Observed behaviour

I didn't create a debug build to see the log output.

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

dpgeorge, 2024-12-07:
> uart.init(channel, 9600, rx=Pin(rxpin), tx=Pin(txpin))

When you use `UART.init` the first argument is the baudrate, not the channel.  The channel is implicit in the `uart` instance.

Using `uart.init(9600, rx=Pin(rxpin), tx=Pin(txpin)))`, it should work.

robert-hh, 2024-12-08:
All arguments of uart.init() can also be added to the constructor call. like:
`uart = UART(channel, 9600, rx=Pin(rxpin), tx=Pin(txpin))`. 
For the setting of rx and tx this is preferred. Otherwise rx and tx ould be set to the default pins with the constructor and then maybe changed to other pins with the init call. Sometimes this is intended, sometimes if may reconfigure pins being set before differently.

gobbyo, 2024-12-12:
Interesting, I use the call `uart.init(channel, 9600, rx=Pin(rxpin), tx=Pin(txpin))` on any release 1.23 and earlier and it works as expected. When I attempt to use release 1.24 or any preview release, it doesn't work.  Has the API changed?

dpgeorge, 2024-12-12:
Nothing has changed in the API.  On v1.23.0 it also has the same behaviour as v1.24.0:
```py
MicroPython v1.23.0 on 2024-06-02; Raspberry Pi Pico with RP2040
Type "help()" for more information.
>>> from machine import Pin, UART
>>> rxpin=5
>>> txpin=4
>>> uart = UART(1)
>>> uart.init(1, 9600, rx=Pin(rxpin), tx=Pin(txpin))
>>> uart
UART(1, baudrate=1, bits=128, parity=None, stop=1, tx=4, rx=5, txbuf=256, rxbuf=256, timeout=0, timeout_char=13001, invert=None)
```
You can see that it initialises the UART to a baudrate of 1 baud and 128 bits.

gobbyo, 2025-01-31:
I tried it on the latest build and everything seems to be working fine.

closed 2025-01-31 by gobbyo