micropython/micropython #16388: RP2350 lv_micropython support

Octopus1633, 2024-12-11:
### Description

Hello, is there any plan to update lv_micropython to support RP2350? The version of PICO SDK also needs to be updated.

### Implementation

I hope the MicroPython maintainers or community will implement this feature

### Code of Conduct

Yes, I agree

andrewleech, 2024-12-11:
lv_micropython is not supported by the core micropython team, but by the LVGL team. I however have been working on a submission to LVGL to allow it to be built against official micropython which would mean the lv_micropython would no longer be needed, or at least wouldn't require manual maintenance.
I haven't had a chance to test my updates on a RP2350 however I expect it should work fine there.
You can follow my changes here https://github.com/lvgl/lv_binding_micropython/pull/341 and/or request directly on the lv_micropython repo for an update to support this chip.

jonnor, 2024-12-27:
Request seems to have been answered well, closing :)

closed 2024-12-27 by jonnor