micropython/micropython #16391: Seeed Studio XIAO RP2350 doesn't boot

briskspirit, 2024-12-11:
### Port, board and/or hardware

RP2 2350

### MicroPython version

MicroPython 1.24.1 ; XIAO RP2350

### Reproduction

BOOTSEL XIAO RP2350 board, copy uf2 file, it reboots and never enumerates again

### Expected behaviour

should enumerate REPL

### Observed behaviour

doesn't enumerate

### Additional Information

On Seeed Studio XIAO RP2350 page there is a note about incompatibility related to flash chip they use (found after struggling for a few hours...) and that they work with MicroPython team to solve this issue. Couldn't find any PRs or issues here so would like to ask if and when it should be fixed and what's the real issue is? 

### Code of Conduct

Yes, I agree

thestumbler, 2024-12-13:
Hey @briskspirit , I'm having a similar problem with the RP2350 Waveshare 1.28 in Round LCD module. But, it DOES work when loading the Waveshare-provided UF2 file. Do you have a Seeed studio-provided UF2 for that module?

briskspirit, 2024-12-14:
@thestumbler here you go [Experimental Firmware 1.24.0](https://files.seeedstudio.com/wiki/XIAO-RP2350/res/RPI_PICO2-20240809-v1.24.0-preview.201.g269a0e0e1.uf2)

thestumbler, 2024-12-15:
Also, you might try the workaround I mention in the linked issue above. Basically grab an older commit of the RP2350 PR from back before it was merged into the mainstream branch. That worked for me and I'm using that until we can figure out what broke. 

I wonder if it's related to the RP2350 A vs B chip, but I don't have an official Pico 2 module to test with (I have a couple on order, should arrive this week)

briskspirit, 2024-12-15:
Seeed Studio says that it’s related to spi flash chip they are using. It’s different from original Pico 2 and something different about it I guess. Like some fast read command or setting to qspi command? Haven’t looked into it too deep

thestumbler, 2024-12-17:
Two things. Could you figure out what memory chip your board actually uses? Perhaps by visual inspection (if it's non destructive) or maybe by firmware inquiry to the chip? Just curious. 

And if you want to build your own UF2 now and not wait, then you should be able to make it from an old PR commit pre-merge.

briskspirit, 2024-12-22:
So @Lesords PRs both for SDK and for MicroPython work when cherry picked, hope SDK PR will be merged soon

charkster, 2025-01-06:
> I wonder if it's related to the RP2350 A vs B chip, but I don't have an official Pico 2 module to test with (I have a couple on order, should arrive this week)

I am having the same issue trying to build MicroPython v1.25.0-preview.172. The SEEED schematic shows that the RP2350-A chip (60pin QFN with 30 gpios) is used. The external non-volatile memory is listed as a 2MB psram (it has the same quad-spi interface as the flash memories, just less capacity)

I copied the RPI_PICO2 board files and made the following modifications:
(1) mpconfigboard.cmake
changed PICO_NUM_GPIOS from 48 to 30 (48 is for QFN-80 part, 30 is for QFN-60)
(2) mpconfigboard.h
changed MICROPY_HW_FLASH_STORAGE_BYTES to (1 * 1024 * 1024)

It should be noted that PICO_FLASH_SIZE_BYTES is set to 4 * 1024 * 1024 in the SDK library, and picotool is being built with that size... I suspect that required picotool build could be a root cause as it was never required with building for the RP2040 boards in older micropython versions.

mattytrentini, 2025-01-06:
This comment loosely explains how to build for the XIAO RP2350:

https://github.com/micropython/micropython/issues/16410#issuecomment-2558680530

Let me know how you go.

jonnor, 2025-03-03:
This looks like it should be resolved by the above comment?