micropython/micropython #16410: Waveshare RP2350 1.28 in Round LCD, Won't Run 1.24.0 or 1.24.1

thestumbler, 2024-12-13:
### Port, board and/or hardware

rp2 / rp2350A / Waveshare RP2350 1.28 in Round LCD

### MicroPython version

This works:
`b9f6698d3 - (HEAD -> rp2-add-2350) tools/ci.sh: Add RPI_PICO2 to CI. (4 months ago) <Damien George>`
Merged versions like 1.24.0, 1.24.1, etc., do not work.

### Reproduction

Build firmware image the usual way, program the chip by copying to the USB drive of the device.

### Expected behaviour

Expected the device to enumerate on the USB port and be able to start talking over REPL or other means. 

### Observed behaviour

Nothing happens. Device is flashed with some code, because I can perform a `picotool info` on a "dead" board and get data back.

### Additional Information

I've been struggling with this for a couple of days now. I have the Waveshare board which runs just fine with a UF2 provided by Waveshare. But I could not get it to run using MicroPython which I built locally. As a sanity check, I successfully built and ran MicroPython on an RP2040 Pico 1 and an Pimoroni Pico Plus 2. Interestingly I could run the Waveshare UF2 on the Pimoroni board no problem. I tried version 1.24.1 and also 1.24.0 (the version which Waveshare was using). A request to Waveshare to get a copy of their board configuration files has been unanswered for several days now.

Finally at the suggestion of @mattytrentini, I rolled back to the PR version from like 4 months ago (see above). **Bingo!** Everything suddenly works now. So something happened between the PR and the final merge that caused this particular board to "break". I have gone over the schematics several times, and can't find any reason to explain the failure.

Just tonight, a few hours before solving the problem, I realized that the Waveshare board uses the RP2350A chip, but the Pimoroni board uses the RP2350B chip. Then I found issue #16391, which I noticed also uses the A chip (but so does the Raspberry Pi Pico 2 module, so that seems like a dead end). 

If additional dumps will be helpful, I can provide OTP data, pico tool info text, version numbers, etc. 

### Code of Conduct

Yes, I agree

thestumbler, 2024-12-17:
Did some more research on this, noting information provided by @briskspirit in his similar issue. Found some interesting things. Seeeq uses PSRAM and not flash, interesting. The UF2 that Seeeq distributes is EXACTLY the same as Waveshare. These provided binaries are from around the same week of the PR that I had on my computer, which I used a few day ago to build an image that  works with my Waveshare module.  Seeeq says they have been collaborating on this issue with engineers from the MicroPython project since 10 Nov. **_Could any of those folks give us an update here?_** Waveshare says they don't have any such board support files, and now that I've dug deeper, I think I believe them. They're probably just as baffled as I am about why it stopped working between August and October. Details below.

_Note, the reason this is of interest to me is not just mere curiosity, but I need to rebuild the firmware to include graphics libraries, and possible other features unique to my project._ 

# Firmware versions:

### Versions built from main branch after RP2350 merge don't work

### Waveshare provided, RP2350 1.28 in Round LCD module, works:
* download link:
  - https://files.waveshare.com/wiki/common/RPI_PICO2-20240809-v1.24.0.zip
* Bootup string:
  - MicroPython v1.24.0-preview.201.g269a0e0e1 on 2024-08-09; Raspberry Pi Pico2 with RP2350
* Date:
  - commit: ???
  - built: 9 Aug 2024
  - file: 28 Aug 2024

### My version, built from old PR, works:
* GIT commit:
  - b9f6698d3 - (HEAD -> rp2-add-2350) tools/ci.sh: Add RPI_PICO2 to CI. (4 months ago) <Damien George>
    commit b9f6698d304c925ada4335bcf88f3f374faa4857
    Author: Damien George <damien@micropython.org>
    Date:   Fri Aug 9 12:32:48 2024 +1000
        tools/ci.sh: Add RPI_PICO2 to CI.
        Signed-off-by: Damien George <damien@micropython.org>
* Bootup string:
  - MicroPython v1.24.0-preview.203.gb9f6698d3.dirty on 2024-12-14; Waveshare RP2350-LCD-1.28 with RP2350
* Date 
  - commit: 9 Aug 2024 
  - built: 14 Dec 2024

### SEEEQ provided firmware, works:
* Product: XIAO RP2350 module (postage stamp size)
  - https://wiki.seeedstudio.com/getting-started-xiao-rp2350/
* download link:
  - https://files.seeedstudio.com/wiki/XIAO-RP2350/res/RPI_PICO2-20240809-v1.24.0-preview.201.g269a0e0e1.uf2
  - this is the EXACT same file being distributed by Waveshare

### Note from the SEEEQ studio website about this problem:

  MICROPYTHON FIRMWARE ISSUE

  As of November 10, 2024, the stable MicroPython firmware version
  1.24.0 available for download at MicroPython.org for RPI_PICO2 is
  currently incompatible with certain devices due to variations in the
  flash chip.
  
  The Seeed Team is actively working to resolve this issue in
  collaboration with the official MicroPython maintainers. In the
  meantime, you can use a preview version of the MicroPython firmware as
  a temporary solution: RP2350 MicroPython Firmware Preview

### Flash chips in use:

* Waveshare:
  - schematics: P25Q32SH-UXH-IR or W25Q16JVUXIQ
    * notice different capacities
  - on module: P25Q32SH-UXH-IR
  - Puya Semiconductors datasheet link:
    * https://www.puyasemi.com/download_path/数据手册/Flash/P25Q32SH_Datasheet_V1.8.pdf

* Seeeq Studios:
  - not specified on schematics
    * https://files.seeedstudio.com/wiki/XIAO-RP2350/res/Seeed-Studio-XIAO-RP2350-v1.0.pdf
  - not visible in photos (covered by a lid)
  - specified as 2 MiB of PSRAM (?)

* Pimoroni Pico Plus 2
  - this board works with main branch builds of MicroPython
  - schematics call out W25Q128JVPIQ (16 MiB chip)
    * https://cdn.shopify.com/s/files/1/0174/1800/files/Pimoroni_Pico_Plus_2_Schematic.pdf

* Raspberry Pi Pico 2 module
  - W25Q32RV, 4 MiB chip

thestumbler, 2024-12-20:
A brief update. Over on the Pico SDK repo, [this issue](https://github.com/raspberrypi/pico-sdk/pull/2098) suggests that the problem with the Seeed Xiao RP2350 board doesn't boot is related to the Pico SDK (an SPI clock speed issue).  This made me think that the Pico SDK version number must have changed between now and when building MicroPython back in August (when a normal build would work on the Waveshare, and presumably the Seeed, boards). I went back and checked this, and it seems not. Back in August, it was v 2.0 of the SDK, and only yesterday the SDK version used in MicroPython was bumped to v 2.1. This hints at the problem being unrelated to the SDK, but some things are confusing.

It seems from the issue comments on the Pico SDK, there really was a problem as it relates to the Seeed board, and that issue has been fixed (but not merged into the SDK yet). I tested with this corrected (but not merged) Pico SDK, and it made no difference on my Waveshare board.

So, it's interesting that the UF2 distributed by Seeed and Waveshare back in August are exactly the same, and work on their respective modules. New firmware builds don't work on the Waveshare (and apparently the Seeed). Supposedly this SDK PR does make the Seeed board work. That means there are different errors associated with each board, and yet both boards ran perfectly with identical firmware in August. I'm baffled.

thestumbler, 2024-12-22:
I asked someone with a Seeed board to test with the SDK PR changing the SPI clock speed. That change, while maybe needed, did not make the XIAO module boot up into MicroPython. Same result as my testing with the Waveshare module.

briskspirit, 2024-12-22:
As I replied you in another topic both PRs work. Think that person who you asked to check built it wrong? Set wrong target? idk

thestumbler, 2024-12-22:
The second PR is MicroPython or the Pico SDK?

It's interesting that the XIAO and Waveshare both ran okay back in August when presumably the SPI was still "too fast". 

Anyway, I'll try this again when I find the second PR. 

Thanks

mattytrentini, 2024-12-23:
Yes, I was able to get a working build for the Xiao RP2350. 

As discussed, it takes two PRs. I built #16299 (which uses the pico-sdk board definition for the Xiao) as well as manually making the flash timing change described in [#2098](https://github.com/raspberrypi/pico-sdk/pull/2098) (which modifies the board definition in the pico-sdk to adjust the timing).

So we need to wait for RPi to merge [#2098](https://github.com/raspberrypi/pico-sdk/pull/2098) and publish a new pico-sdk version - and then MicroPython can update to the new pico-sdk and merge in #16299....

thestumbler, 2024-12-23:
I'm still confused. I understand there is an issue in the SDK regarding the SPI clock. That's being corrected. But.... how is it that building MicroPython back in August, with the same problematic v2.0 SDK, result in a workable SPI clock frequency?  Between August and last week, when the SDK didn't change at all, something did change in MicroPython which "broke" these two boards. 

Perhaps in the early versions MicroPython was forcing a certain SPI clock speed, ignoring / overwriting whatever the SDK did? And after August, this changed and MicroPython stopped doing that and left configuration of the SPI clock entirely to the SDK?

Sorry for my continued curiosity, but the reported solutions, while maybe being important, don't make sense in light of the experimental observations.

mattytrentini, 2024-12-23:
It's important to note that MicroPython wasn't _using_ the seeed_xiao_rp2350 board definition in the pico-sdk back in August. That only occurred with #16299.

It was built against the pico2 definition so, I haven't tracked down exactly where, but I suspect the SPI flash rate changed for _that board_ somewhere (either in the pico-sdk or overridden in MicroPython). 

It's still a little confusing, I just didn't want you looking in the wrong place! 😉

thestumbler, 2024-12-23:
But the initial problem can't be in the Pico SDK, since that didn't change (or rather, the version 2.0 that MicroPython used didn't change) from August until a couple days ago.

Octopus1633, 2025-01-15:
@thestumbler 
Hi, I'm also curious why the August MicroPython firmware works but the latest one doesn't. They use the same PICO_FLASH_SPI_CLKDIV, did you find the problem?