micropython/micropython #16415: Missing board.json files

Josverl, 2024-12-14:
I noted that there appear to be a few boards missing a board.json file

The boards are defined in the following files

```
LAUNCHXL, ports\cc3200\boards\LAUNCHXL\mpconfigboard.h
PYBV3, ports\stm32\boards\PYBV3\mpconfigboard.h
PYBV4, ports\stm32\boards\PYBV4\mpconfigboard.h
STM32H573I_DK, ports\stm32\boards\STM32H573I_DK\mpconfigboard.h 
```
Is this intentional, 
or is it an oversight and should be corrected?

See also: https://github.com/mattytrentini/mpbuild/issues/56

robert-hh, 2024-12-14:
The first three are definitely old ones, nit on sale anymore and possibly obsolete.

Josverl, 2024-12-14:
Is there a way to indicate the boards deprecation in the board.json?

Is there an established schema for it that could be formalised or extended ?

dpgeorge, 2024-12-16:
> Is this intentional,
> or is it an oversight and should be corrected?

It's intentional to leave out `board.json` for all these boards except maybe `STM32H573I_DK`.  That could have one added.

> Is there a way to indicate a boards deprecation in the board.json?

Not at the moment.  The absence of a `board.json` means the board should not be built for the `micropython.org/download` page.

closed 2025-03-03 by Josverl