micropython/micropython #16424: USER_C_MODULES for ESP32

straga, 2024-12-16:
### Port, board and/or hardware

esp32

### MicroPython version

Master

### Reproduction

For building the example modules which come with MicroPython, set USER_C_MODULES to the examples/usercmodule directory for Make, or to examples/usercmodule/micropython.cmake for CMake.

The recent commit b65e89107c0194bfdaf92720786341a2047f36b6 enforces that USER_C_MODULES must point to a directory in the Make system. However, this causes confusion with the documentation, which mixes usage of relative paths and file references for both Make and CMake.

but that's half the battle. We can comment on this committee. 
Also, they no longer work as they used to.

### Observed behaviour

#### Examples:

- The following command no longer works with Make after the commit:

```
cd micropython/ports/esp32
make USER_C_MODULES=../../examples/usercmodule/micropython.cmake
```

- Instead, absolute paths are required:

`make USER_C_MODULES=/dev_iot/opt/upy/micropython/examples/usercmodule/micropython.cmake BOARD=ESP32_GENERIC all`

### Code of Conduct

Yes, I agree

dpgeorge, 2024-12-16:
> * The following command no longer works with Make after the commit:

This command should work, it's used in the CI to test building user C modules with the esp32 port.  Maybe you just need an extra `../`:
```
cd ports/esp32
make USER_C_MODULES=../../../examples/usercmodule/micropython.cmake
```

straga, 2024-12-16:
@dpgeorge 

```
[1045/1396] Generating /Users/dev/Documents/dev_iot/opt/upy/micropython/mpy-cross/build/mpy-cross
FAILED: /Users/dev/Documents/dev_iot/opt/upy/micropython/mpy-cross/build/mpy-cross
cd /Users/dev/Documents/dev_iot/opt/upy/micropython/ports/esp32/build-ESP32_GENERIC/esp-idf/main_esp32 && make -C /Users/dev/Documents/dev_iot/opt/upy/micropython/mpy-cross
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
../py/py.mk:38: *** Directory containing USER_C_MODULES doesn't exist: ../../../examples/usercmodule/.  Stop.
```

[image]

dpgeorge, 2024-12-16:
Ah, OK, so the issue is building mpy-cross with user C modules, not the esp32 port.

For a workaround, build mpy-cross separately, before building the esp32 board (from the top-level of this repo):
```
make -C mpy-cross
```

uraich, 2024-12-17:
Having the same problem here. It looks like USER_C_MODULES=... is passed to compiling mpy-cross.
**make -C mpy-cross** works fine but the problem remains when compiling micropython. (mpy-cross is re-built)

**make USER_C_MODULES=/opt/ucc/micros/esp32/user_modules/micropython.cmake -C mpy-cross** reproduces the problem.

straga, 2024-12-18:
@dpgeorge 
Rebuil mpy-cross and comment https://github.com/micropython/micropython/commit/b65e89107c0194bfdaf92720786341a2047f36b6, and works with relative path. **make USER_C_MODULES=../../../examples/usercmodule/micropython.cmake**
 
```
# Confirm the provided path exists, show abspath if not to make it clearer to fix.
#$(if $(wildcard $(USER_C_MODULES)/.),,$(error USER_C_MODULES doesn't exist: $(abspath $(USER_C_MODULES))))
```

Need rewrite or remove like was before -  "Confirm the provided path exists ... "

cnadler86, 2024-12-22:
Same here. commenting [b65e891](https://github.com/micropython/micropython/commit/b65e89107c0194bfdaf92720786341a2047f36b6) solved the problem.

closed 2025-03-03 by jonnor