micropython/micropython #16441: submodules command doesn't work

kdschlosser, 2024-12-18:
### Port, board and/or hardware

esp32

### MicroPython version

1.24.1

### Reproduction

make BOARD=ESP32_GENERIC_S3 submodules

### Expected behaviour

for it to install the needed submodules

### Observed behaviour

the submodules don't install. 

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

dpgeorge, 2024-12-18:
It works for me:
```
$ make BOARD=ESP32_GENERIC_S3 submodules
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
make[1]: Entering directory 'ports/esp32'
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Updating submodules: lib/berkeley-db-1.xx lib/tinyusb lib/micropython-lib
Synchronizing submodule url for 'lib/berkeley-db-1.xx'
Synchronizing submodule url for 'lib/micropython-lib'
Synchronizing submodule url for 'lib/tinyusb'
# If available, do blobless partial clones of submodules to save time and space.
# A blobless partial clone lazily fetches data as needed, but has all the metadata available (tags, etc.).
# Fallback to standard submodule update if blobless isn't available (earlier than 2.36.0)
make[1]: Leaving directory 'ports/esp32'
```
Make sure you are in the `ports/esp32` directory, and have sourced the IDF's `export.sh`.

kdschlosser, 2024-12-18:
are you using the release or the master branch? I am using the release version.

[image]

dpgeorge, 2024-12-18:
I'm using master branch.  What does `idf.py --version` print for you?

kdschlosser, 2024-12-18:
5.2.3

kdschlosser, 2024-12-18:
if I manually init the submodules it compiles without an issue. The submodules build command doesn't work.

I even ran the install.sh and also the export.sh scripts for the esp-idf thinking that maybe I forgot to set it up. It still didn't work after I did that. The ESP-IDF is properly installed otherwise the build would not work if I manually initialize the submodules.

dpgeorge, 2024-12-18:
The CI uses `make submodules`, and it works for me, so I'm not sure what to do here.

Are you on windows?  Do you have spaces in your path?

kdschlosser, 2024-12-18:
I am running Windows but  I am compiling inside of WSL on Ubuntu. No spaces in the path either.

Just out of question I noticed that some of the ports will install submodules as part of the build and other ports like the ESP32 require running a separate command to install the submodules. Is there a reason for this? I also noticed that there are checks that are made to see if a submodule has been initialized. If it has not been initialized then why not initialize it right then and there?? The way the build system works with running the IDF for the sole purpose of collecting the submodules that should be initialized is rather overkill.. With cmake create a function that will initialize a submodule and call that function if a submodule that is neededm and it has not yet been initialized . Needing to run the build system 2 times ends up polluting the build because it creates a second build folder.

VsevolodVasin, 2024-12-21:
Same error 

`vsevolodvasin@VsevolodVasinPC:~/lytko-tests/micropython/ports/esp32$ make submodules
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
make[1]: Entering directory '/home/vsevolodvasin/lytko-tests/micropython/ports/esp32'
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Updating submodules: lib/berkeley-db-1.xx lib/tinyusb lib/micropython-lib
Synchronizing submodule url for 'lib/berkeley-db-1.xx'
Synchronizing submodule url for 'lib/micropython-lib'
Synchronizing submodule url for 'lib/tinyusb'
# If available, do blobless partial clones of submodules to save time and space.
# A blobless partial clone lazily fetches data as needed, but has all the metadata available (tags, etc.).
# Fallback to standard submodule update if blobless isn't available (earlier than 2.36.0)
usage: git submodule [--quiet] [--cached]
   or: git submodule [--quiet] add [-b <branch>] [-f|--force] [--name <name>] [--reference <repository>] [--] <repository> [<path>]
   or: git submodule [--quiet] status [--cached] [--recursive] [--] [<path>...]
   or: git submodule [--quiet] init [--] [<path>...]
   or: git submodule [--quiet] deinit [-f|--force] (--all| [--] <path>...)
   or: git submodule [--quiet] update [--init] [--remote] [-N|--no-fetch] [-f|--force] [--checkout|--merge|--rebase] [--[no-]recommend-shallow] [--reference <repository>] [--recursive] [--[no-]single-branch] [--] [<path>...]
   or: git submodule [--quiet] set-branch (--default|--branch <branch>) [--] <path>
   or: git submodule [--quiet] set-url [--] <path> <newurl>
   or: git submodule [--quiet] summary [--cached|--files] [--summary-limit <n>] [commit] [--] [<path>...]
   or: git submodule [--quiet] foreach [--recursive] <command>
   or: git submodule [--quiet] sync [--recursive] [--] [<path>...]
   or: git submodule [--quiet] absorbgitdirs [--] [<path>...]
make[1]: Leaving directory '/home/vsevolodvasin/lytko-tests/micropython/ports/esp32'`

kdschlosser, 2024-12-21:
if I run the following commands from the micropython folder

```shell
git submodule sync lib/tinyusb
'git submodule update --init --depth=1 lib/tinyusb'

git submodule sync lib/mbedtls
'git submodule update --init --depth=1 lib/mbedtls'

git submodule sync lib/berkeley-db-1.xx
git submodule update --init --depth=1 lib/berkeley-db-1.xx

git submodule sync lib/micropython-lib
git submodule update --init --depth=1 lib/micropython-lib
```

and then compile for the ESP32 everything compiles properly.

kdschlosser, 2024-12-27:
I want to say the RP2 and also the STM builds automatically collect the SDK's

kdschlosser, 2024-12-27:
Here it is in STM32.
https://github.com/micropython/micropython/blob/c73204128e16cfa62abd42b07981bd8e5adeb321/ports/stm32/Makefile#L497

I am pretty sure there are one or 2 other ports that do this same kind of a thing. I didn't check any of the other ones but it shows you that not all the ports have the submodules downloaded using only the submodules command.

kdschlosser, 2024-12-27:
There are 2 locations in the stm32 port. the other one is in the mboot/makefile. There is also one for the renesas-ra port as well.

I made a mistake when I said the RP2 port it was the renesas-ra port, I knew there was another one. The 2 other ports was because of the STM32 having 2 locations where it is done. I knew there was a total of 3 places, didn't realize that 2 of them were in the same port.

kdschlosser, 2024-12-27:
I don't think that checking the readme files would impact the build time in any noticeable way. not with how fast SSD's are and most machines are running those these days. Python has to be one of the slowest things in terms of drive access and I am not able to get an accurate measurement of how long it takes to see if a file exists. It's too fast and the timekeeping on a Windows OS doesn't allow for measurements that small. I am sure that makefile is going to be very fast at testing if a file exists or not. 

What you do need to add is the `--depth 1` parameter to the git commands when updating a submodule. This makes collecting the submodules leaps and bounds faster. 

this

```shell
git submodule update --init lib/stm32lib
```

would become this

```shell
git submodule update --depth 1 --init lib/stm32lib)
```

That stops git from collecting all of the git related data like the commit history. If you were checking out a different branch or commit then you would need to have the depth set to where that is visible. Since that is not being done because the submodule is set to a specific release or commit there is no reason to collect the commit history or the rest of the git data. It wastes drive space, bandwidth and time.

I was having an issue with some router on the internet for almost a year and it caused me to only be able to get 500k a sec download speeds from GitHub. It was horrible. It would take 10 minutes for micropython to collect the submodules. adding that `--depth 1` trimmed the time down to only a minute or so. My speed issue has finally corrected itself thank god. It really showed me how much additional data was being downloaded because of GitHub and it was stuff that was never being used at all.

kdschlosser, 2024-12-27:
> Oh yeah and regarding make submodules not working for you? I really don't know why this is the case, it's always worked for me and as already mentioned it works on fresh clones in CI.

I am not able to put my finger on why it's not working, It should be working but it's not. It only seems to not be working for ESP32. Maybe the IDF version? I am using 5.2.3. I know that MicroPython works with 5.2 so there should be no issue.. Perhaps there is some kind of change made in the IDF from 5.2.2 to 5.2.3 that is causing the problem. I am not sure.

If I update and init the submodules from the command line and then start the build as usual it compiles fine and without error. For whatever reason running the submodules command the build system is not able to enumerate the needed modules. The printed out `GIT_SUBMODULES` is either not being printed out when the `ECHO_SUBMODULES` variable is set to 1 or there is something not working correctly in the makefile code that that collects the information. I don't know enough about makefile to find out the exact cause. It could be in the cmake side of things that is causing the issue. 

This doesn't make any sense to me..

```cmake
if(ECHO_SUBMODULES)
    # If cmake is run with GIT_SUBMODULES defined on command line, process the port / board
    # settings then print the final GIT_SUBMODULES variable and exit.
    # Note: the GIT_SUBMODULES is done via echo rather than message, as message splits
    # the output onto multiple lines
    execute_process(COMMAND ${CMAKE_COMMAND} -E echo "GIT_SUBMODULES=${GIT_SUBMODULES}")
    message(FATAL_ERROR "Done")
endif()
```