micropython/micropython #16448: Unable to configure pin 18 & 19 as GPIO output in ESP32S3

MetroMushrooms, 2024-12-19:
### Port, board and/or hardware

esp32 Port, Generic S3

### MicroPython version

MicroPython V1.24 

### Reproduction

1. Copy the code below into a .py file and run it (assuming you have a neopixel strip handy) and you get no data out on GPIO 19. 

`import machine
import neopixel

# Constants for IO MUX
IO_MUX_BASE = 0x60009000  # Base address for IO MUX
GPIO19_MUX_OFFSET = 0x4C  # Offset for GPIO19
IO_MUX_GPIO19_REG = IO_MUX_BASE + GPIO19_MUX_OFFSET

# Function to remap GPIO19
def remap_gpio19():
    """Remap GPIO19 to standard GPIO functionality."""
    # Read the current value of the register
    current_val = machine.mem32[IO_MUX_GPIO19_REG]
    print(f"Initial IO MUX register value for GPIO19: 0x{current_val:08X}")

    # Clear the MCU_SEL bits (bits [2:0]) and set to 0 (GPIO function)
    machine.mem32[IO_MUX_GPIO19_REG] = (current_val & ~0b111) | 0

    # Verify the change
    new_val = machine.mem32[IO_MUX_GPIO19_REG]
    print(f"Updated IO MUX register value for GPIO19: 0x{new_val:08X}")

# Call the remap function
remap_gpio19()

# Reconfigure GPIO19 as output
pin19 = machine.Pin(19, machine.Pin.OUT)
pin19.value(0)  # Set GPIO19 to LOW
print("GPIO19 reconfigured as output and set to LOW.")

# Initialize NeoPixel
NUM_LEDS = 24  # Number of LEDs in your NeoPixel ring
try:
    np = neopixel.NeoPixel(pin19, NUM_LEDS)
    print("NeoPixel object initialized successfully.")
except Exception as e:
    print(f"Error initializing NeoPixel: {e}")

# Set all LEDs to white
try:
    print("Setting LEDs to white...")
    for i in range(NUM_LEDS):
        np[i] = (255, 255, 255)  # White
    print("Writing to NeoPixel...")
    np.write()
    print("NeoPixel write completed.")
except Exception as e:
    print(f"Error writing to NeoPixel: {e}")

`

### Expected behaviour

Ideally this would would set the leds of a neopixel to white.

### Observed behaviour

No data or standard GPIO functionality is coming from GPIO 19. 

### Additional Information

I imagine this is very similar to [ #12479 ](https://github.com/micropython/micropython/issues/12479). I've tried to remap the GPIO, I've gone into ESP idf and disabled usb/jtag I've tried mirroring the solution of #12479  with no luck. Other pins work just fine and I tested pin 19 Using C and it worked. So there's something happening here I'm not grasping and would greatly appreciate a solution as pin 19 is what is baked in on my pcb. 

### Code of Conduct

Yes, I agree

garywill, 2024-12-22:
GPIO 19 is USB Pin on S3. I encountered problem on ESP32-C3 USB pins, then solved by https://github.com/micropython/micropython/pull/16236 . Maybe they're related issues (not sure)

projectgus, 2025-01-07:
As @garywill mentioned, currently it's not possible to re-assign the USB pins to other functions in MicroPython. I'm working on a fix to allow this. The same problem is already tracked in #14217 so I'm going to close this issue in favour of that one, please subscribe there to see updates.

closed 2025-01-07 by projectgus