micropython/micropython #16453: mpremote: Reading a mounted file can be extremely slow

peterhinch, 2024-12-19:
### Port, board and/or hardware

RP2

### MicroPython version

MicroPython v1.24.1 on 2024-11-29; Raspberry Pi Pico with RP2040

### Reproduction

Copy the script to the RP2 as `rats.py`
```py
from time import ticks_ms, ticks_diff
t = ticks_ms()
with open("m00", "r") as f:
    while z := f.readline():
        print(".", end="")
print()
print(f"Elapsed time {ticks_diff(ticks_ms(), t)}ms")
```
Place the ~8KiB text file [m00](https://github.com/peterhinch/micropython-touch/blob/master/optional/bitmaps/m00) in the PC current directory. Issue
```bash
$ mpremote mount . exec "import rats"
```

### Expected behaviour

I would expect this to complete in under 1s.

### Observed behaviour

It takes over 90s. Tested on two PC's.

### Additional Information

This seems to be down to the use of `.readline()`. A similar test where the file was opened "rb" and accessed using repeated calls to `read(128)` completed in under 1s

### Code of Conduct

Yes, I agree

andrewleech, 2024-12-20:
Oh drats, I've fixed this before in the past but now I'm not sure if it made it into a PR.... the virtual file object created by mpremote on the device implemts readline locally (on device) by reading single byte by byte from the host computer: https://github.com/micropython/micropython/blob/4e76acc88d8673ce4f43698fbb4c2913e63b0135/tools/mpremote/mpremote/transport_serial.py#L573

I tested previously adding an extra command in the command table https://github.com/micropython/micropython/blob/4e76acc88d8673ce4f43698fbb4c2913e63b0135/tools/mpremote/mpremote/transport_serial.py#L385 for `READLINE` so that the file command could be forwarded directly to host rather than implemented internally - this fixed the problem.

andrewleech, 2024-12-20:
Found the commit stacked on top of an old mpremote branch, now rebased here: https://github.com/micropython/micropython/compare/master...andrewleech:micropython:mpremote_mount_readline?expand=1

dpgeorge, 2024-12-21:
See fix in #16461.

closed 2024-12-21 by peterhinch