micropython/micropython #16457: 1.24 breaks USB OTG support

dcmcshan, 2024-12-19:
### Port, board and/or hardware

ESP32S3

### MicroPython version

1.23 / 1.24

### Reproduction

Instll kernel, reset, see if it boots

### Expected behaviour

Boots

### Observed behaviour

Loading firmware 1.24 does not boot using USB-OTG.  I don't have serial port.  Just USB-OTG, so need kernel to default that on.

Loading firmware 1.23 works as expected.

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

dpgeorge, 2024-12-19:
Please specify the firmware you used, eg `ESP32_GENERIC_S3`, and also the exact hardware.

dcmcshan, 2024-12-31:
https://micropython.org/download/ESP32_GENERIC_S3

1.23 boots, 1.24, 1.25 beta do not

Custom board using usb-otg, no serial interface

dcmcshan, 2025-01-02:
I think it's probably booting, just not enabling usb otg.  Is there a way to enable in boot.py?

projectgus, 2025-01-07:
I [downloaded](https://micropython.org/download/ESP32_GENERIC_S3/) and flashed both `ESP32_GENERIC_S3-20241129-v1.24.1.bin` and `ESP32_GENERIC_S3-20241025-v1.24.0.bin` release binaries to an ESP32-S3-DevKitC board using the native USB port, and they both work as expected.

As @dpgeorge mentioned, it's possible there is some issue with the specific hardware you have.

The other thing I'd suggest double checking is the flashing parameters and binary type are exactly the same for the working and non-working binaries.