micropython/micropython #16482: Pico W port Wifi Problems

edmcman, 2024-12-25:
### Port, board and/or hardware

RP2040 on Raspberry Pi Pico W

### MicroPython version

(sysname='rp2', nodename='rp2', release='1.24.1', version='v1.24.1 on 2024-11-29 (GNU 13.2.0 MinSizeRel)', machine='Raspberry Pi Pico W with RP2040')

### Reproduction

1. Clone https://github.com/edmcman/rpi-temp-server/tree/debug
2. Edit SSID/password in main.py
3. Run `pipkin --port /dev/ttyACM0 --install -r requirements.txt`
4. Edit the IP address in test.bash to batch the Pico W's IP
5. Run test.bash a bunch of times.

### Expected behaviour

The test.bash script tests whether the web server remains accessible after progressively longer periods of times with no requests.  It is expected that the web server will remain available and the script will print "Test succeeded".

### Observed behaviour

When run 20 times with a limit of 2048 second periods, 8 out of the 20 runs failed, meaning that the web server became inaccessible.  Here is a breakdown of the power modes during failures:

```
fgrep -l -R Fail *.log | xargs fgrep Random | awk '{print $4}' | sort | uniq -c
      6 network.WLAN.PM_POWERSAVE
      2 None
```

And the overall distribution of tested pm values:
```
      2 0xa11140
      8 network.WLAN.PM_NONE
      1 network.WLAN.PM_PERFORMANCE
      7 network.WLAN.PM_POWERSAVE
      3 None
```

I am not sure why they are so uneven.

### Additional Information

# High level problems

I think there are two problems:
1. The default PM value may not be ideal.  My testing suggests that `network.WLAN.PM_NONE` may be more reliable.
2. The fact that the Pico W reports it is still connected and can't reconnect is very problematic.

# Other Information

When the Pico W becomes inaccessible, the logs contain the following messages repeated very often:
```
[ 2509685] ASYNC(0000,49,0,0,0)
```

It is unclear what the 49 message is.  Under normal operation, it appears to be correlated with PSK_SUP messages.  But as soon as it starts repeating by itself the Pico W is offline.

Also, an asynchronous task that calls `WLAN.isconnected()` and `WLAN.status()` shows that the Pico W is connected, even though it does not seem to be:
```
[snip]
[ 3505634] ASYNC(0000,49,0,0,0)
[ 3507681] ASYNC(0000,49,0,0,0)
[ 3508397] ASYNC(0000,49,0,0,0)
Still checking Wi-Fi connection...
Debug: connected: True status: 3
Actual PM: 0xa11142
```

Here is a zip file of my collected logs: 
[logs-2048.zip](https://github.com/user-attachments/files/18247013/logs-2048.zip)

I am currently conducting more and longer tests to hopefully provide more information.  In theory, if `PM_NONE` proves more reliable, changing the default to help a lot in practice.  But even the current behavior for other power management modes is very concerning as it means long term connections are basically impossible.

### Code of Conduct

Yes, I agree

scruss, 2024-12-27:
I've attempted to reproduce your result, but unfortunately can't.

Same MicroPython version:
`(sysname='rp2', nodename='rp2', release='1.24.1', version='v1.24.1 on 2024-11-29 (GNU 13.2.0 MinSizeRel)', machine='Raspberry Pi Pico W with RP2040')`

I ran 21 tests of random PM settings with main.py modified to include:
```py3
network.country("CA")
network.hostname("picowX")
```
with no failures. These are the counts of the passed tests:
```
      8 0xa11140
      3 network.WLAN.PM_NONE
      5 network.WLAN.PM_PERFORMANCE
      3 network.WLAN.PM_POWERSAVE
      2 None
```
I ran five further tests with country and hostname removed, and PM hardcoded to `network.WLAN.PM_POWERSAVE`. I had no disconnections.

Log files: [rpi-temp-server-debug-scruss.zip](https://scruss.com/wordpress/wp-content/uploads/2024/12/rpi-temp-server-debug-scruss.zip)

edmcman, 2024-12-27:
Thanks for attempting to reproduce @scruss.  There must be some component of the problem that is related to the AP, which is a Actiontec Single Dual-Band Wireless Network Extender and Ethernet Over Coax Adapter (WCB3000N).

This is the first device that has had any issues with it.

scruss, 2024-12-27:
If you have a board like an ESP32, maybe check if it has disconnection problems too? It would be a bit of work to port your test:

* network.WLAN has fewer power options: PM_NONE, PM_PERFORMANCE, PM_POWERSAVE.
* internal temperature is reported by `esp32.raw_temperature()` — but in Fahrenheit
* gpiozero isn't a thing for ESP32 boards, and some have LEDs on different pins.

That way, you could see if it was a MicroPython problem or an AP problem.

This must be very frustrating for you. Side quests of gloom are no fun when you want to get on with the main thing!

edmcman, 2024-12-28:
I don't have another board like an ESP32, but I am chasing some other leads at the moment, such as sniffing the wifi when the problem happens.

edmcman, 2024-12-28:
I think I found out what the mysterious "49" event is:

https://github.com/embassy-rs/cyw43/blob/master/src/events.rs#L107

    /// WEP ICV error occurred
    ICV_ERROR = 49,

This suggests to me that there is a problem with the WPA keying.  (Although the comment says WEP, I use WPA2 PSK.)  @scruss Do you have WPA enabled?

An interesting experiment might be for me to disable WPA on my AP and see if the problem occurs.

It's hard to tell since most of the documentation for the CYW43 does not seem public, but it seems like the cyw43-driver that micropython uses should probably handle this event, but it does not.

edmcman, 2024-12-28:
My current theory is that because of power management, the Pico W becomes more likely to miss a rekeying message of some sort, and then can't decrypt any messages.

scruss, 2024-12-28:
> @scruss Do you have WPA enabled?

Yes, I do. I've never seen a 49 event, though

edmcman, 2024-12-28:
Here is a pcap while the problem is happening: 
[pico-w.zip](https://github.com/user-attachments/files/18267351/pico-w.zip)

There is not a smoking gun, but there are some more clues pointing towards rekeying.

* The first ~7200 seconds is a successful run.
* At t=7233, the Pico successfully connects to the network.
* At t=7235, the 4-way handshake occurs.  The last message gets sent twice.
* At t=8554, the AP sends a Disassociate message to Pico. The "reason" is "4-way handshake timeout".

edmcman, 2024-12-28:
Looking at the serial numbers, it seems like a lot of wlan frames are dropped.  When I captured the pcap, the laptop was within a foot or two of the pico.  So if the laptop did not receive a frame, it is very conceivable that the pico did not as well.  It looks like a group rekeying occurs a few seconds before the disassociation.  So my hypothesis is:

1. AP starts group rekey
2. All rekey frames are lost at the pico / sniffer
3. AP sends disassociate message to pico
4. This message may be lost too, since it does not seem to be acknowledged
5. Pico continues to think it is connected

edmcman, 2024-12-28:
Here is an additional recording, with the sniffer laptop in the same room as the AP.  
[pico-w-sniffer-near-ap.zip](https://github.com/user-attachments/files/18267657/pico-w-sniffer-near-ap.zip)

Unlike the other pcap, I also included eapol traffic in this pcap, in addition to any traffic to or from the pico.

* t=729 Another device receives a group rekey message
* t=735 AP sends a bunch of disassociate messages to pico

We are still missing a bunch of frames, if frame SNs are supposed to be incremented by one each time.  Sometimes the SN skips by as much as 10.  I'm not sure what is going on with that.  Maybe this is a very noisy channel, which is why the problem does not happen for everyone.

edmcman, 2024-12-29:
On a whim, I decided to try the [pico_w https](https://github.com/raspberrypi/pico-examples/tree/master/pico_w/wifi/httpd) example to see if it has the same problem.  It appears to.  Testing is a little bit more difficult because I need to manually restart each test, but it disconnected after 1024 seconds and 4096 seconds.  I did not manage to get debug output working, but it seems very likely to be the same problem.

With that in mind, I am going to close this issue and re-file it against pico-sdk.

PR #16914 (2025-03-13): Update cyw43-driver to v1.1.0, and add WPA3 constants to network.WLAN for cyw43
### Summary

This updates the cyw43-driver to the latest v1.1.0 version.

This is a minor release of cyw43-driver, some bugs have been fixed and some features have been added.  Those things were all tested upstream for that driver.

A few configuration constants have been updated here to work with the new driver version (very minor changes).

And, now that cyw43-driver supports WPA3, the WPA3 security constants have been added to `network.WLAN` for that driver's bindings.  These user-facing Python constants match t [... cut]
closed 2024-12-29 by edmcman