micropython/micropython #16484: WeACTF411 preview .hex file will not run

davefes, 2024-12-26:
### Port, board and/or hardware

WeActF411

### MicroPython version

WEACT_F411_BLACKPILL-V31_FLASH_8M-20241223-v1.25.0-preview.160.gc73204128.hex

### Reproduction

Upload the .hex file using a STM32Cube Programmer application
dave@davef:~$ rsusb

### Expected behaviour

That the .hex file should run.

### Observed behaviour

Can not run the .hex file for `V3.1 and 8MB SPI flash
v1.25.0-preview.160.gc73204128`
using STM32Cube Programmer and get the following:

dave@davef:~$ rsusb
Using buffer-size of 30
Connecting to /dev/ttyUSB0 (buffer-size 30)...
Trying to connect to REPL  connected
Retrieving sysname ... b'FATAL: uncaught exception 200039d0\r\nOSError: 16\r\n\nFATAL ERROR:\n'
Traceback (most recent call last):
  File "/home/dave/.local/lib/python3.10/site-packages/rshell/main.py", line 1381, in connect
    ip_address = socket.gethostbyname(port)
socket.gaierror: [Errno -2] Name or service not known

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

andrewleech, 2024-12-26:
I've never used rshell so don't know what it's trying to do. Did you test with mpremote?

Part of that error message includes socket interactions, I do know that the F411 doesn't have any networking connection stack enabled.

davefes, 2024-12-26:
```
dave@davef:~$ mpremote fs ls
b'FATAL: uncaught exception 200039d0\r\nOSError: 16\r\n\nFATAL ERROR:\n'
Traceback (most recent call last):
  File "/home/dave/.local/bin/mpremote", line 8, in <module>
    sys.exit(main())
  File "/home/dave/.local/lib/python3.10/site-packages/mpremote/main.py", line 528, in main
    handler_func(state, args)
  File "/home/dave/.local/lib/python3.10/site-packages/mpremote/commands.py", line 110, in do_filesystem
    state.ensure_raw_repl()
  File "/home/dave/.local/lib/python3.10/site-packages/mpremote/main.py", line 470, in ensure_raw_repl
    self.transport.enter_raw_repl(soft_reset=soft_reset)
  File "/home/dave/.local/lib/python3.10/site-packages/mpremote/transport_serial.py", line 170, in enter_raw_repl
    raise TransportError("could not enter raw repl")
mpremote.transport.TransportError: could not enter raw repl
```

davefes, 2024-12-29:
Tried to .dfu file:
`sudo dfu-util --alt 0 -D WEACT_F411_BLACKPILL-V31_FLASH_8M-20241223-v1.25.0-preview.160.gc73204128.dfu`
and get the same error when I try to run it using `rshell` or `mpremote`.

davefes, 2024-12-29:
The board has a faulty 8MB flash, I could program it with the default (non-external flash variant).

davefes, 2024-12-29:
The default firmware runs on the board.

davefes, 2025-01-05:
Same problem with `WEACT_F411_BLACKPILL-V31_FLASH_8M-20250102-v1.25.0-preview.172.g5d12df51f.hex` on a brand-new board.

robert-hh, 2025-01-05:
Which frequency has the crystal on your board? The 8M in der variant name stands for 8MHz.

davefes, 2025-01-05:
25MHz.

> Firmware (v3.1 board with 8MB SPI Flash)
`[v1.25.0-preview.172.g5d12df51f (2025-01-02) .dfu](https://micropython.org/resources/firmware/WEACT_F411_BLACKPILL-V31_FLASH_8M-20250102-v1.25.0-preview.172.g5d12df51f.dfu) / [[.hex]](https://micropython.org/resources/firmware/WEACT_F411_BLACKPILL-V31_FLASH_8M-20250102-v1.25.0-preview.172.g5d12df51f.hex)`
`

robert-hh, 2025-01-05:
I have a WEACT STM32F411 V3.1 Blackpill board with 25Mhz, (MB Flash here, and it works fine. I can download the actual v1.25-172  preview from the web site. Dowbload command:
`dfu-util --alt 0 -D WEACT_F411_BLACKPILL-V31_FLASH_8M-20250102-v1.25.0-preview.172.g5d12df51f.dfu `
So I conclude that the firmware is fine. Are you sure the board is 25Mhz? Did you look at fine print of the crystal?

davefes, 2025-01-05:
I says `25.000`.  Both the default firmware.hex, default firmware.dfu and nspsck's 8MB .hex firmware work.

Seems almost all of the problems I have here are related to 8MB flash.  My flash is `25Q64JVSIQ`, could you check that you have the same flash?

robert-hh, 2025-01-05:
The flash on my board is labeled `Winbond 25Q64VSIQ` as well. The flash is used for the file system only. So even if the flash is bad, the firmware should start.
BUT: I have problems using mpremote with that board as well. Similar error message than yours. It must have worked some time ago, because there are files on the board, and they are loaded on hard reset. Every time I try a soft reboot the board locks up. Since mpremote starts with a soft reset, that could be the problem.

davefes, 2025-01-05:
The 8MB .hex file that Andrew generated earlier in the development cycle worked.  I used that on several projects.  I looked at the first few lines and last few lines of the various .hex files to see if there was something different ... but I really don't know what I expect to see.

robert-hh, 2025-01-05:
I have an old set of board files that work as well. The difference is, that the old file uses SoftSPI for flash, and the new ones use hard SPI. The difference is in bdev.c of the board directory. You will not see that in the .hex files.

davefes, 2025-01-05:
Looking at old bdev.c files I think I see;
`STATIC const mp_soft_spi_obj_t soft_spi_bus = {'
for Andrew's older builds.

However for old nspsck builds I see:
`STATIC const spi_proto_cfg_t spi_bus = {`
He does not show his most recent build files.

Where is Andrew's source files, I have lost my way?

davefes, 2025-01-05:
> Edit: re-configuring the board files to SoftSPI make the flash work.
To reduce errors on my end would you post the modified files?  bdev.c? and mpconfigboard.h?

Thank you.

andrewleech, 2025-01-05:
Oh that's very interesting. I actually don't think there's many other spiflash boards that use hardware spi (most use qspi) so maybe a spi soft reset bug has snuck in.

robert-hh, 2025-01-06:
Confirmed. Building with PR #16452 fixes the problem.

robert-hh, 2025-01-06:
@davefes Please try to install and run the attached firmware.dfu file on your board. It runs on my Blackpill board with the signature:
`MicroPython v1.25.0-preview.182.g5c654d0e2 on 2025-01-06; WEACT_F411_BLACKPILL with STM32F411CE`
It works fine with mpremote and accepts soft reset. Successful soft reset shows, that the fix works.

[firmware.zip](https://github.com/user-attachments/files/18323809/firmware.zip)

davefes, 2025-01-06:
Well, that works.

So `firmware.zip` is different then `[v1.25.0-preview.180.g495ce91ca (2025-01-06) .dfu](https://micropython.org/resources/firmware/WEACT_F411_BLACKPILL-V31_FLASH_8M-20250106-v1.25.0-preview.180.g495ce91ca.dfu)` 

When that appeared last night I thought it had been brought up to date.

davefes, 2025-01-06:
When do the change(s) appear at:
`https://github.com/micropython/micropython/tree/master/ports/stm32/boards/WEACT_F411_BLACKPILL`

davefes, 2025-01-06:
Here is something even more interesting ... remember I thought I had damaged the SPIFLASH on three new boards.  Your firmware seems to work on them.

>>> import os
>>> os.statvfs('/flash')
(4096, 4096, 2048, 2034, 2034, 0, 0, 0, 0, 255)
2048 does say 8MB, correct?

davefes, 2025-01-06:
The system does behave differently using `rshell` than previously.  It looks like the files are missing until you do a `NRST` and fire-up `rshell` again.  I can live with that.

robert-hh, 2025-01-07:
> When do the change(s) appear at:

I just had built the firmware after using the latest state of the master branch with the commits from PR #16542. But the previous commits worked as well. I did not test the firmware from the download page, since these do not include the changes from PR #16542 and thus they will not work.

> 2048 does say 8MB, correct?

Yes. 2048*4096 == 8MB

davefes, 2025-01-07:
Had a look at the commits and I think I can see how things have to change.  However, for now, I will use your `firmware.zip`.

Thanks again for your help with this.

davefes, 2025-01-12:
Has `WEACT_F411_BLACKPILL-V31_FLASH_8M-20250111-v1.25.0-preview.183.g99ac8193e.dfu` got the changes mentioned above?

robert-hh, 2025-01-12:
No. PR #16452 is not yet merge to the master branch, and therefore the images from the download page do not work.

davefes, 2025-01-12:
Looks like I misunderstood `merged` tag mentioned `2 weeks ago`

davefes, 2025-01-17:
`WEACT_F411_BLACKPILL-V31_FLASH_8M-20250116-v1.25.0-preview.203.gf2cd1a3db.dfu` runs now.

Thank you.

closed 2025-01-17 by davefes