micropython/micropython #16490: time.mktime() returns incorrect Unix timestamp on ESP32

jurijp, 2024-12-28:
### Port, board and/or hardware

sysname='esp32', nodename='esp32', release='1.24.1', version='v1.24.1 on 2024-11-29', machine='Generic ESP32 module with ESP32.  Unique ID b'\x94\xe6\x86\x13o$' (not sure what this is

### MicroPython version

release='1.24.1', version='v1.24.1 on 2024-11-29',
When using this board and setting time using the ntp library thus ntptime.settime()    this synchronizes the RTC to UTC using an NTP server, and it seems to be working correctly since time.localtime() returns (2024, 12, 28, 18, 11, 1, 5, 363) (UTC time). As I understand it, this should reset the real time clock to the current time. If I then set time.mktime(time.localtime()) which should provide the Unix time, it does not, instead is spits out 788726670  which is 1994-12-29 18:44:30 UTC.  Can someone please explain this ? Is it a a bug?  Bottom line, how can the realtime clock be set to a specific time?

### Reproduction

from machine import RTC
import time

# Manually set the RTC to a known time (e.g., 2024-01-28 15:30:00 UTC)
rtc = RTC()
rtc.datetime((2024, 1, 28, 0, 15, 30, 0, 0))  # (year, month, day, weekday, hour, minute, second, subseconds)

# Get the Unix timestamp
unix_time = time.mktime(time.localtime())

print("Unix time:", unix_time)

Result: Unix time: 759771000  it should be closer to   1735413856.203176

### Expected behaviour

Expected the unix time to be determined correctly. That is it should look something like this 1735413022.697727

### Observed behaviour

The output is incorrect  - see Expected behaviour result above

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

robert-hh, 2024-12-28:
The various ports of MicroPython have different epoch dates. For ESP32 it's 1.1.2000. Unix uses 1.1.1970.

robert-hh, 2024-12-28:
So the large difference is related to the different epoch dates. But it should be exactly 30 years, or 946684800. But it is 975642856 seconds, equivalent to about 30.9 years. Also: ports like MIMXRT with a 1.1.1970 epoch return values different to UNIX for `time.mktime(time.localtime())`, when localtime is set to the actual time. The different is about 28959808 seconds or ~335 days. That looks wrong.

jurijp, 2024-12-29:
Of course you are correct regarding the slight difference -  I was just entering a date into the realtime clock. However I don't understand why unix_time = time.mktime(time.localtime()) even after setting the realtime clock continues to return a date in 1994 for the Unix time. After telling it a date and time in 2024, why does it go back to 1994  when I set the variable unix_time as above ?
Is there an explanation for this error ?  Thanks jp

robert-hh, 2024-12-29:
`time.mktime()` takes it argument, which is a full time tuple, and returns the seconds since a reference time to the time given in the argument. For Unix systems, this reference time is 1.1.1970, for MicroPython ports it is 1.1.2000 or 1.1.1970. This reference date is called `epoch`. The MicroPython documentation states about `time.mktime()`:
_This is inverse function of localtime. It’s argument is a full 8-tuple which expresses a time as per localtime. It returns an integer which is the number of seconds since Jan 1, 2000._
The difference in seconds between 1.1.1970 and 1.1.2000 is 946684800.

robert-hh, 2024-12-29:
The following ports use 1.1.1970 as `epoch` date:
- Unix
- Webassembly
- Windows
- RP2 (RP2040 and RP2350 based boards)
- MIMXRT (e.g. Teensy 4.x)

All other ports use 1.1.2000, like ESP32, ESP8266, STM32, CC3200, NRF, Renesas-RA

jurijp, 2024-12-29:
Ah, so just providing the tuple for localtime and driving it into the realtime clock on the board is not going to do anything, other than provide the number of seconds since the epoch date. Why on earth is this such a mess insofar novice programmers like myself are concerned. Additionally looking at the documentation https://docs.micropython.org/en/latest/library/time.html#time.mktime  it says "This is inverse function of localtime. It’s argument is a full 8-tuple which expresses a time as per localtime. It returns an integer which is the number of seconds since Jan 1, 2000.". There is no mention of 1970, which is in your previous response - so is this what you meant when you pinged maxi07 with "Note to @maxi07: another inconsistency in the time.mktime() doc."
Please dont think of this as a challenge, I'm just trying to figure this out.  Frankly for a first time user, the micropython documentation is a bit terse and imho lacks a number of examples.  But that's just on person's opinion.  Thank you again for taking the time to respond

robert-hh, 2024-12-29:
> Ah, so just providing the tuple for localtime and driving it into the realtime clock 

The (Micro)Python time module is not the realtime clock. The time module mostly provides time conversion methods. time.localtime() gets it's time value from a realtime clock (if present) or another time source, like a seconds counter.

The user maxi07 had just now raised an issue about a minor formal inconsistency of the `time.mktime()` documentation.

maxi07, 2024-12-29:
my original issue was addressed in the Micropython-stubs project. I noticed, that `time.mktime()` would normally take one positional argument of an 8-tuple, but it is documented otherwise in the official documentation. see original issue here: Josverl/micropython-stubs#771
this addresses the rp2 port.

Josverl, 2024-12-30:
Im (slowly) working on converting the textual documentation into stricter Python reference stubs, and merging the parameter and return types into the stubs project.

- time.mktime is [correctly typed in the reference-stubs](https://github.com/Josverl/micropython-stubs/blob/14b0abaa1feb48ee36057b35f45ba309f6786613/micropython-reference/time/__init__.pyi#L93)
- but currently the published stubs are indeed not correct, as these are based on the [incorrect documentation](https://docs.micropython.org/en/latest/library/time.html#time.mktime) only

Im working on fixing this and many other mismatches and emissions, and keeping the docs easy to maintain and readable by humans and typecheckers