micropython/micropython #16498: PYBV11: 'sys.implementation[2]' does not reflect the build variant

hmaerki, 2024-12-29:
### Port, board and/or hardware

PYBV11

### MicroPython version

v1.24.1 and before

### Reproduction

flash different build-variants on a PYBV11 and call `sys.implementation[2]`

### Expected behaviour

It would be beneficial, if `sys.implementation[2]` would differ for each variant. This is required by octoprobe to verify if the correct variant is flashed.

### Observed behaviour

## PYBV11

build: PYBV11
```
>>> sys.implementation[2]
'PYBv1.1 with STM32F405RG'
```

build: PYBV11-DP_TREAD
```
>>> sys.implementation[2]
'PYBv1.1 with STM32F405RG'  <<Wrong: I am missing  '-DP_TREAD'
```

## In contrast the RP2

build: RP2
```
>>> sys.implementation[2]
'RP2350'
```

build: RP2-RISCV
```
>>> sys.implementation[2]
'RP2350-RISCV'  <<Ok: '-RISCV' is there
```

### Code of Conduct

Yes, I agree

dpgeorge, 2025-02-27:
Note that on the rp2 in RISCV mode it's just a coincidence that it has the variant name in it.  What's actually happening there is that the MCU name is `RP2350-RISCV`, and that's the case regardless of the variant.

I don't think there's a strong case to store the board variant in the firmware.  The actual board name is not even stored in the firmware to begin with, just a user-friendly version of it.

For Octoprobe, can you instead store a file on the filesystem of the target device which contains the required metadata?  Then delete the file before updating the firmware, then create a new file with the new metadata once the firmware is updated.  That should be robust (atomic updates of metadata and firmware) and allow any relevant build data to be stored.

Josverl, 2025-02-27:
currently it **very hard** to precisely determine the exact firmware a board is running. 
I try to do that in mpflash - and i still frequently that i mismatch a board.

in order to build a firmware the minimum required informration is : 
- micropython version - `v1.24.1`( possibly including commit hash) 
- boardname/board_id : `PYBV11`
- variant : `DP, DP_THREAD, NETWORK, THREAD or '' `
the port is not required , as boardnames are unique across all ports
but the only thing that identifies a board in the firmware is the a description that doen not include any part of this information.
such as : 'PYBv1.1 with STM32F405RG'

To resolve this thing into board and variant across multiple MicroPython versions, it takes 
- building a list of all possible descriptions  from the header and cmake / make files , across multiple versions
- extracting the unique combinations of long / short descriptions , board_ids
- and store that in a database / json 
to resolve : 
- read the `sys.implementation[2]` (assuming that its a not an very old version of MicroPython) and the CPU 
- split off the 'with CPU' part - as that is not consistently added in all ports / boards 
- and then try to match that against either the long of short description, taking into account that some board_ids have been renamed  , prefixed, changed case , or have been changed from different boards to variant across versions

That problem is mostly worked around my [mpflash.board_id](https://github.com/Josverl/mpflash/tree/main/mpflash/mpboard_id) ( although it needs work in identifying variants) but I would much prefer to not need to have to do this.

Storing the board the board and variant , and retrieving it via `sys.implementation._board` and `sys.implementation._variant` would be very helpful to be able to positively identify the origin of the firmware.

afaik the longest Board is is 30 bytes : 'ADAFRUIT_ITSYBITSY_M4_EXPRESS' and variants are up to ~12 bytes , so the size impact should be quite low

hmaerki, 2025-02-27:
> Storing the board the board and variant , and retrieving it via sys.implementation._board and sys.implementation._variant would be very helpful to be able to positively identify the origin of the firmware.

That would be great!

dpgeorge, 2025-02-28:
OK, if there's more than one use case for this feature, it sounds like it's worth adding.

I do agree that it's nice to be able to robustly map a given firmware back to the source code it was compiled from.

Although it should be noted that any command line arguments passed to make that modify the configuration, eg `CFLAGS_EXTRA=-DMICROPY_PY_MARSHAL=1` will not be reflected in any variable in the firmware.  So the best you can assume is that the firmware was built without any changes from the command line.

> Storing the board the board and variant , and retrieving it via `sys.implementation._board` and `sys.implementation._variant` would be very helpful to be able to positively identify the origin of the firmware.

Instead of having two separate variable, how about a single `sys.implementation._board` which contains `<BOARD>-<VARIANT>`, with `-<VARIANT>` being omitted if it's the default variant.  This is easy to split up into board and variant if needed (because `-` is never part of a board or variant name).

hmaerki, 2025-02-28:
Your probosal - in other words:

`make BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT`

would result in

```python
> sys.implementation._board
'ESP32_GENERIC_S3-SPIRAM_OCT'
```

I love it 👍

Josverl, 2025-03-01:
> how about a single sys.implementation._board which contains <BOARD>-<VARIANT>, with -<VARIANT> being omitted if it's the default variant.

That would be **very useful**.
We would need to add clear rules about casing though, as the names of boards and variants are uppercase by convention, but afaik this is not formally documented (but could be wrong) . Docs are easy to add though.

And I think it's quite acceptable that not all build options are embedded. For firmwares built out of a repro there is still a commit hash and builds since last tag in the extended version info that can help pinpoint that, and more details for those use-cases

dpgeorge, 2025-03-03:
See #16843 for a new `sys.implementation._build` entry.