micropython/micropython #16499: ESPNow.config fails to get or set param "rate" (unknown config param)

martin-korinek-1985, 2024-12-29:
### Port, board and/or hardware

ESP32-C3 Super Mini

### MicroPython version

uname: (sysname='esp32', nodename='esp32', release='1.24.1', version='v1.24.1 on 2024-11-29', machine='ESP32C3 module with ESP32C3')

### Reproduction

```
import os

import espnow
import network

sta = network.WLAN(network.STA_IF)
sta.active(True)
sta.disconnect()

e = espnow.ESPNow()
e.active(True)

print('uname     :', os.uname())
print('rxbuf     :', e.config('rxbuf'))
print('timeout_ms:', e.config('timeout_ms'))
print('rate      :', e.config('rate'))
```

### Expected behaviour

Expectation is NOT to get error ValueError: unknown config param but get value of the "rate".

### Observed behaviour

Traceback (most recent call last):
  File "<stdin>", line 16, in <module>
ValueError: unknown config param

[image]

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

agatti, 2024-12-29:
As far as I know the ESP32 SDK doesn't provide any explicit function to get the current ESPNow wifi rate.  It only allows for it to be set (see `esp_wifi_config_espnow_rate` and `esp_now_set_peer_rate_config` in `components/esp_wifi/include/esp_now.h` in ESP-IDF 5.2 for example).

martin-korinek-1985, 2024-12-29:
I do appreciate all the work that is done on MicroPython and I very much like to use it. My intention is not to spend hours trying to do the same in low level language like C or ASM so I believe you there is no ESP-IDF API for what I was trying to do. Unfortunately I did not realize that based on the documentation:  
[image]  
and still have a few questions.
1) I think it would be great to update the documentation with note, that ESPNow.config(rate=XY) is only working on ESP32 && IDF>=4.3.0 && only the "set" part, not the "get" part. So the rate can be only set but the information is not possible to retrieve (at least not in this way).
2) Can you please tell me what is the default rate used (MP 1.24.1) when ESPNow is activated (as in my code example)? It would be also great to have this information in the documentation.
3) Am I correct to assume that mode for the longest distance is "WIFI_PHY_RATE_LORA_250K = 0x29" ?
4) Is there any other way to verify what "rate" is currently set?
Thanks in advance.

agatti, 2024-12-29:
>     1. I think it would be great to update the documentation with note, that ESPNow.config(rate=XY) is only working on ESP32 && IDF>=4.3.0 && only the "set" part, not the "get" part. So the rate can be only set but the information is not possible to retrieve (at least not in this way).

That's a valid point.  If nobody steps up with a PR for the documentation in a couple of days I'll probably do that myself.

>     2. Can you please tell me what is the default rate used (MP 1.24.1) when ESPNow is activated (as in my code example)? It would be also great to have this information in the documentation.

From a quick look at `ports/esp32/modespnow.c` there is no explicit call to `esp_wifi_config_espnow_rate` when the module is initialised, nor any rate setting from elsewhere (but don't quote me on the latter part, haven't put a breakpoint on the function call in question and checked).

So I'd say the rate being set upon initialisation is the same as if you'd leave that field alone when writing the same code in C.  If the ESP-IDF documentation doesn't say anything about that then they probably expect you to set it in any case (and it makes more sense that way, doesn't it?).

>     3. Am I correct to assume that mode for the longest distance is "WIFI_PHY_RATE_LORA_250K = 0x29" ?

Cannot really help you on this specific issue, maybe somebody more knowledgeable on the subject can chime in :)

>     4. Is there any other way to verify what "rate" is currently set?

If the rate is invalid or it couldn't be set you should get a Python exception, as the C API returned an error code in that case.

martin-korinek-1985, 2024-12-30:
Thanks for the reply. I did try to use only the SET RATE part of ESPNow.config, yet I am having issues:

```python
import os
import time
from collections import OrderedDict

import espnow
import network

rates = OrderedDict(WIFI_PHY_RATE_1M_L=0x00,
                    WIFI_PHY_RATE_2M_L=0x01,
                    WIFI_PHY_RATE_5M_L=0x02,
                    WIFI_PHY_RATE_11M_L=0x03,
                    WIFI_PHY_RATE_2M_S=0x05,
                    WIFI_PHY_RATE_5M_S=0x06,
                    WIFI_PHY_RATE_11M_S=0x07,
                    WIFI_PHY_RATE_48M=0x08,
                    WIFI_PHY_RATE_24M=0x09,
                    WIFI_PHY_RATE_12M=0x0A,
                    WIFI_PHY_RATE_6M=0x0B,
                    WIFI_PHY_RATE_54M=0x0C,
                    WIFI_PHY_RATE_36M=0x0D,
                    WIFI_PHY_RATE_18M=0x0E,
                    WIFI_PHY_RATE_9M=0x0F,
                    WIFI_PHY_RATE_MCS0_LGI=0x10,
                    WIFI_PHY_RATE_MCS1_LGI=0x11,
                    WIFI_PHY_RATE_MCS2_LGI=0x12,
                    WIFI_PHY_RATE_MCS3_LGI=0x13,
                    WIFI_PHY_RATE_MCS4_LGI=0x14,
                    WIFI_PHY_RATE_MCS5_LGI=0x15,
                    WIFI_PHY_RATE_MCS6_LGI=0x16,
                    WIFI_PHY_RATE_MCS7_LGI=0x17,
                    WIFI_PHY_RATE_MCS0_SGI=0x18,
                    WIFI_PHY_RATE_MCS1_SGI=0x19,
                    WIFI_PHY_RATE_MCS2_SGI=0x1A,
                    WIFI_PHY_RATE_MCS3_SGI=0x1B,
                    WIFI_PHY_RATE_MCS4_SGI=0x1C,
                    WIFI_PHY_RATE_MCS5_SGI=0x1D,
                    WIFI_PHY_RATE_MCS6_SGI=0x1E,
                    WIFI_PHY_RATE_MCS7_SGI=0x1F,
                    WIFI_PHY_RATE_LORA_250K=0x29,
                    WIFI_PHY_RATE_LORA_500K=0x2A)

sta = network.WLAN(network.STA_IF)
sta.active(True)
sta.disconnect()

esp_now = espnow.ESPNow()

print('uname     :', os.uname())
print('rxbuf     :', esp_now.config('rxbuf'))
print('timeout_ms:', esp_now.config('timeout_ms'), '\n')

longest_rate_name = max(len(name) for name in rates)
for name, value in rates.items():
    print(f'Setting rate {name:{longest_rate_name}}', end=' ')
    try:
        esp_now.active(False)
        esp_now.config(rate=value)
        esp_now.active(True)
        print('SUCCESSFUL')
    except Exception as exception:
        print('FAILED with exception:', exception)
    time.sleep(1)
```

```
uname     : (sysname='esp32', nodename='esp32', release='1.24.1', version='v1.24.1 on 2024-11-29', machine='ESP32C3 module with ESP32C3')
rxbuf     : 526
timeout_ms: 300000 

Setting rate WIFI_PHY_RATE_1M_L      SUCCESSFUL
Setting rate WIFI_PHY_RATE_2M_L      SUCCESSFUL
Setting rate WIFI_PHY_RATE_5M_L      SUCCESSFUL
Setting rate WIFI_PHY_RATE_11M_L     SUCCESSFUL
Setting rate WIFI_PHY_RATE_2M_S      SUCCESSFUL
Setting rate WIFI_PHY_RATE_5M_S      SUCCESSFUL
Setting rate WIFI_PHY_RATE_11M_S     SUCCESSFUL
Setting rate WIFI_PHY_RATE_48M       SUCCESSFUL
Setting rate WIFI_PHY_RATE_24M       SUCCESSFUL
Setting rate WIFI_PHY_RATE_12M       SUCCESSFUL
Setting rate WIFI_PHY_RATE_6M        SUCCESSFUL
Setting rate WIFI_PHY_RATE_54M       SUCCESSFUL
Setting rate WIFI_PHY_RATE_36M       SUCCESSFUL
Setting rate WIFI_PHY_RATE_18M       SUCCESSFUL
Setting rate WIFI_PHY_RATE_9M        SUCCESSFUL
Setting rate WIFI_PHY_RATE_MCS0_LGI  SUCCESSFUL
Setting rate WIFI_PHY_RATE_MCS1_LGI  SUCCESSFUL
Setting rate WIFI_PHY_RATE_MCS2_LGI  SUCCESSFUL
Setting rate WIFI_PHY_RATE_MCS3_LGI  SUCCESSFUL
Setting rate WIFI_PHY_RATE_MCS4_LGI  SUCCESSFUL
Setting rate WIFI_PHY_RATE_MCS5_LGI  SUCCESSFUL
Setting rate WIFI_PHY_RATE_MCS6_LGI  SUCCESSFUL
Setting rate WIFI_PHY_RATE_MCS7_LGI  SUCCESSFUL
Setting rate WIFI_PHY_RATE_MCS0_SGI  SUCCESSFUL
Setting rate WIFI_PHY_RATE_MCS1_SGI  SUCCESSFUL
Setting rate WIFI_PHY_RATE_MCS2_SGI  SUCCESSFUL
Setting rate WIFI_PHY_RATE_MCS3_SGI  SUCCESSFUL
Setting rate WIFI_PHY_RATE_MCS4_SGI  SUCCESSFUL
Setting rate WIFI_PHY_RATE_MCS5_SGI  SUCCESSFUL
Setting rate WIFI_PHY_RATE_MCS6_SGI  SUCCESSFUL
Setting rate WIFI_PHY_RATE_MCS7_SGI  SUCCESSFUL
Setting rate WIFI_PHY_RATE_LORA_250K FAILED with exception: [Errno 1] EPERM: ESP_FAIL
Setting rate WIFI_PHY_RATE_LORA_500K FAILED with exception: [Errno 1] EPERM: ESP_FAIL
```

As demonstrated here I am not able to set any of those two LORA modes....those that I want most. Any idea what am I doing wrong here? Thanks in advance!

agatti, 2024-12-30:
Ok, I've done some tests.

With this (temporary) patch

<detail>
```diff
diff --git a/ports/esp32/modespnow.c b/ports/esp32/modespnow.c
index 7873ff897..95152a7be 100644
--- a/ports/esp32/modespnow.c
+++ b/ports/esp32/modespnow.c
@@ -256,9 +256,9 @@ static mp_obj_t espnow_config(size_t n_args, const mp_obj_t *pos_args, mp_map_t
         self->recv_timeout_ms = args[ARG_timeout_ms].u_int;
     }
     if (args[ARG_rate].u_int >= 0) {
-        esp_initialise_wifi();  // Call the wifi init code in network_wlan.c
+        // esp_initialise_wifi();  // Call the wifi init code in network_wlan.c
         check_esp_err(esp_wifi_config_espnow_rate(ESP_IF_WIFI_STA, args[ARG_rate].u_int));
-        check_esp_err(esp_wifi_config_espnow_rate(ESP_IF_WIFI_AP, args[ARG_rate].u_int));
+        // check_esp_err(esp_wifi_config_espnow_rate(ESP_IF_WIFI_AP, args[ARG_rate].u_int));
     }
     if (args[ARG_get].u_obj == MP_OBJ_NULL) {
         return mp_const_none;

```
</detail>

this code works:

```python
import espnow
import network

sta = network.WLAN(network.STA_IF)
sta.active(True)
sta.config(protocol=8)
esp_now = espnow.ESPNow()
esp_now.config(rate=0x29)
print('LR mode on.')
```

Probably the patch to the espnow module isn't needed but I just wanted the LR rate setting to occur on a clean baseline.  Can you please try that with no C code patches too?

martin-korinek-1985, 2024-12-30:
Why dont you have to do ```esp_now.active(True)``` ?

agatti, 2024-12-30:
I probably need to activate the espnow instance, indeed - I only tested setting the LR mode.

Anyway, according to the ESP-IDF documentation long range mode needs to be activated first - and that's done using the `esp_wifi_set_protocol` C API call.  The Python code for that is in `ports/esp32/network_wlan.c` inside `network_wlan_config`, turns out it isn't documented.  I guess that most users won't need to mess with the radio interface at that level, and nobody ever asked until now?

The value 8 comes from `components/esp_wifi/include/esp_wifi_types.h` in ESP-IDF (given that it looks like a bitfield you can probably enable more than one protocol at once?):

```c
#define WIFI_PROTOCOL_11B         1
#define WIFI_PROTOCOL_11G         2
#define WIFI_PROTOCOL_11N         4
#define WIFI_PROTOCOL_LR          8
#define WIFI_PROTOCOL_11AX        16
```

And you know the rest :)  Still, documentation needs to be improved on that front, I totally agree with you on that.

martin-korinek-1985, 2024-12-30:
I am trying to create electronic game for kids, I want to use many ESP32-C3 Supermini nodes as they are cheap compared to proper LoRa SX chips so that is my motivation to get the longest possible range out of ESP32 Wi-Fi module.
Should there by any different if you first activate interface and use config after or vice versa first set the config and activate interface later?

agatti, 2024-12-30:
The order of operations for the STA interface actually matters.  So for example if you unplug the board, plug it back in (so the radio is completely reset), and then run this:

```python
import network
sta = network.WLAN(network.STA_IF)
sta.config(protocol=8)
sta.active(True)
```

The code fails with `RuntimeError: Wifi Unknown Error 0x0102`.  The espnow code doesn't seem to mind about the rate being set before activation, as in no exception is raised.  You still need to try it out on your end to see if data is transmitted/received, but you don't have many combinations to try if you find one that doesn't work :)

glenn20, 2025-02-10:
> ```
> Setting rate WIFI_PHY_RATE_MCS6_SGI  SUCCESSFUL
> Setting rate WIFI_PHY_RATE_MCS7_SGI  SUCCESSFUL
> Setting rate WIFI_PHY_RATE_LORA_250K FAILED with exception: [Errno 1] EPERM: ESP_FAIL
> Setting rate [...]

I suspect that these LORA modes are not supported on the C3 (I dont have experience with that chip or with LORA), or possibly not yet supported in the IDF version used by micropython.

martin-korinek-1985, 2025-02-10:
> I suspect that these LORA modes are not supported on the C3 (I dont have experience with that chip or with LORA), or possibly not yet supported in the IDF version used by micropython.

This way it works for me:
```Python
    try:
        print(f'INIT start')
        if_ap = WLAN(AP_IF)
        if_ap.active(False)

        if_sta = WLAN(STA_IF)
        if_sta.active(True)
        if_sta.config(protocol=8)  # WIFI_PROTOCOL_LR

        esp_now = ESPNow()
        esp_now.active(True)
        esp_now.config(rate=0x29)  # WIFI_PHY_RATE_LORA_250K
```
and I was able to use this communication for about 500m with only two 4cm whip antennas. Whether the LR mode of ESP-Now is actually activated or not I am not able to determine since it is not possible to read 'rate'.

martin-korinek-1985, 2025-02-10:
> That's a valid point. If nobody steps up with a PR for the documentation in a couple of days I'll probably do that myself.
>
>
> From a quick look at `ports/esp32/modespnow.c` there is no explicit call [...]

I was just checking https://docs.micropython.org/en/latest/library/espnow.html#espnow.ESPNow.config and there is still no mention about 'rate' not being possible to read. I would like to ask someone to update that. I do not need it any more, but it may help others, it is quite frustrating when it does not work and you don't know why even when you read the documentation several times.

agatti, 2025-02-10:
> I was just checking [docs.micropython.org/en/latest/library/espnow.html#espnow.ESPNow.config](https://docs.micropython.org/en/latest/library/espnow.html#espnow.ESPNow.config) and there is still no mention about 'rate' not being possible to read. I would like to ask someone to update that. I do not need it any more, but it may help others, it is quite frustrating when it does not work and you don't know why even when you read the documentation several times.

I was supposed to do it but then I got sidetracked with improvements to the native emitter, sorry.  I'll get around to it this afternoon.

agatti, 2025-02-10:
> I was able to use this communication for about 500m with only two 4cm whip antennas.

According to [the Internet™](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/wifi.html#wi-fi-protocol-mode) 802.11 LR theoretical range should be around 1 kilometre, no mention of which antenna though. @martin-korinek-1985 was 500 metres a limit you found due to lack of signal strength or were you limited by available space for testing?

@glenn20 the C3 does support LR modes, you may be thinking of the C2 not having LR modes available (see also the link I put in the previous paragraph).

martin-korinek-1985, 2025-02-10:
500m was not the limit, it was distance I used and it was working fine.

dpgeorge, 2025-02-11:
Documentation about `rate` not being settable added in 3b6252466139d5dfc7458eeea5d759b6c9850146

closed 2025-02-11 by dpgeorge

glenn20, 2025-02-17:
> According to [the Internet™](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/wifi.html#wi-fi-protocol-mode) 802.11 LR theoretical range should be around 1 kilometre, no [...]

LORA is different from the Espressif Long Range (LR) wifi operating modes.