micropython/micropython #16500: docs: time.mktime requires one positional argument

maxi07, 2024-12-29:
### Description

when running the function `time.mktime()`, an exception occurs as the function does require one positional argument. But this is not documented, although it states "t’s argument is a full 8-tuple" inside the description.

### Code of Conduct

Yes, I agree

peterhinch, 2024-12-30:
This seems clear:
> This is inverse function of localtime. It’s argument is a full 8-tuple which expresses a time as per localtime...

(Give or take an extraneous apostrophe.)

robert-hh, 2024-12-30:
I'm not sure what the concern is, but I assume that the heading should be something like `time.mktime(date_time)` and not just `time.mktime()`.

Josverl, 2024-12-30:
Its a Doc Bug 

The docs have a basic mismatch between : 
 - the function signature : time.mktime() - No parameters 
 - and the documentation below that , that mentions a required parameter  

Also See: https://github.com/Josverl/micropython-stubs/issues/771 and https://github.com/micropython/micropython/issues/16490