micropython/micropython #16510: Cannot convert to littlefs2

benj5378, 2025-01-01:
### Port, board and/or hardware

ESP32 S3 (Arduino Nano ESP32)

### MicroPython version

MicroPython v1.24.1 on 2024-11-29; Arduino Nano ESP32 with ESP32S3

### Reproduction

```py
>>> import vfs
>>> vfs.umount('/')
>>> vfs.VfsLfs2.mkfs(bdev)
>>> vfs.mount(bdev, '/')
```
as according to https://docs.micropython.org/en/latest/reference/filesystem.html#littlefs

### Expected behaviour

No errors, filesystem with littlefs2 made succesfully.

### Observed behaviour

```py
>>> import vfs
>>> vfs.umount('/')
>>> vfs.VfsLfs2.mkfs(bdev)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 22] EINVAL
```

:(

### Additional Information

Works for fat32:
```py
>>> import vfs
>>> vfs.umount('/')
>>> vfs.VfsFat.mkfs(bdev)
>>> vfs.mount(bdev, '/')
```

### Code of Conduct

Yes, I agree

robert-hh, 2025-01-02:
I tried the sequence below and it worked.
```
>>> from flashbdev import bdev
>>> bdev.info()
(1, 129, 2097152, 2097152, 'vfs', False)
>>> import vfs
>>> vfs.umount("/")
>>> vfs.VfsLfs2.mkfs(bdev)
>>> fs = vfs.VfsLfs2(bdev)
>>> vfs.mount(fs, "/")
```
What is returned by bdev.info() in your case?

benj5378, 2025-01-24:
@robert-hh 

```python
>>> from flashbdev import bdev
>>> bdev.info()
(1, 129, 6356992, 9830400, 'ffat', False)
>>> import vfs
>>> vfs.umount("/")
>>> vfs.VfsLfs2.mkfs(bdev)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 22] EINVAL
>>> fs = vfs.VfsLfs2(bdev)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 84
>>> vfs.mount(fs, "/")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'fs' isn't defined
```

benj5378, 2025-02-17:
@robert-hh any ideas?

robert-hh, 2025-02-17:
There was a discussion here https://github.com/orgs/micropython/discussions/13420  coming to the conclusion that it is linked to the partition label, which is `ffat` for your board, and `vfs` for generic boards. So unless you have the proper partition table layout, you cannot create the LFS file system.
There is https://github.com/glenn20/mp-image-tool-esp32, which allows modifying the partition table in the firmware file. But I did not see at first glance a way to change the partition label,

benj5378, 2025-02-17:
@robert-hh thank you very much! What is the reason for my board having `ffat` and not `vfs`? Is it Arduino manufacturing or the micropython image that I used?

robert-hh, 2025-02-18:
If you get the v24.1 .bin file from the download page, you can rename the file system partition with:
`mp-image-tool-esp32 ARDUINO_NANO_ESP32-20241129-v1.24.1.bin --delete ffat --add vfs:fat:0x610B:0x960B`
That will create a file called `ARDUINO_NANO_ESP32-20241129-v1.24.1-delete=ffat-add=vfs=fat=6356992=9830400.bin` with the following partition table:
```
                        Partition table (flash size: 16MB):                        
╭──────────┬──────┬─────────┬──────────┬──────────┬───────────┬───────┬───────────╮
│ Name     │ Type │ SubType │   Offset │     Size │       End │ Flags │           │
├──────────┼──────┼─────────┼──────────┼──────────┼───────────┼───────┼───────────┤
│ nvs      │ data │ nvs     │   0x9000 │   0x5000 │    0xe000 │   0x0 │ (20.0 kB) │
│ otadata  │ data │ ota     │   0xe000 │   0x2000 │   0x10000 │   0x0 │  (8.0 kB) │
│ app0     │ app  │ ota_0   │  0x10000 │ 0x300000 │  0x310000 │   0x0 │  (3.0 MB) │
│ app1     │ app  │ ota_1   │ 0x310000 │ 0x300000 │  0x610000 │   0x0 │  (3.0 MB) │
│ vfs      │ data │ fat     │ 0x610000 │ 0x960000 │  0xf70000 │   0x0 │  (9.4 MB) │
│ factory  │ app  │ factory │ 0xf70000 │  0x80000 │  0xff0000 │   0x0 │  (0.5 MB) │
│ coredump │ data │ 3       │ 0xff0000 │  0x10000 │ 0x1000000 │   0x0 │ (64.0 kB) │
╰──────────┴──────┴─────────┴──────────┴──────────┴───────────┴───────┴───────────╯
```
When loading that to the board and starting the board, you will most likely get an error message about a corrupted file system. Stop that message train with Ctrl-C and use vfs.VfsLfs2.mkfs() to format the partition.

robert-hh, 2025-02-18:
> What is the reason for my board having ffat and not vfs? Is it Arduino manufacturing or the micropython image that I used?

It is related to the Arduino board micropython image and a decision of Arduino to use FAT. Then, the files system can easily used with MSC to have it directly accessible from the PC, even if that is not recommended.

benj5378, 2025-03-06:
> It is related to the Arduino board micropython image and a decision of Arduino to use FAT

However, this also happens when I download the firmware from micropython.org - even if I do a full erase of flash first. Shouldn't the partition table in firmware downloaded from micropython.org be set to littlefs2? @robert-hh

benj5378, 2025-03-06:
@robert-hh further, I think there need to be changes to the docs. The docs just says, that (any) filesystem can be converted to littlefs2 using the command - which untrue, or at least highly misleading, if the firmware is shipped with a partition table, that *cannot* be converted to littlefs2. Not without external tools and highly specific knowledge.

This was also requested here https://github.com/orgs/micropython/discussions/13420#discussioncomment-11774530