micropython/micropython #16511: esp32s3/c3 hardware  JTAG control to external pins

emard, 2025-01-01:
### Description

Allows routing of ESP32C3 and ESP32S3 internal hardware usb-jtag
to external gpio pins making ESP32S3/C3 cheap and fast JTAG for
programming other devices.

Once enabled this ESP32 hardware services JTAG by itself
so no other software interrupts or any  server running
is needed.

### Code Size

12 lines of code to enable, 6 lines to disable here is arduino example

#if CONFIG_IDF_TARGET_ESP32S3
#define OMATRIX_TCK   85
#define OMATRIX_TMS   86
#define OMATRIX_TDI   87
#define OMATRIX_SRST 251
#define IMATRIX_TDO  251
#endif

#if CONFIG_IDF_TARGET_ESP32C3
#define OMATRIX_TCK   36
#define OMATRIX_TMS   37
#define OMATRIX_TDI   38
#define OMATRIX_SRST 127
#define IMATRIX_TDO   39
#endif

void route_usb_jtag_to_gpio()
{
  digitalWrite(LED_BUILTIN, LED_ON);
  pinMode(PIN_TCK, OUTPUT);
  pinMode(PIN_TMS, OUTPUT);
  pinMode(PIN_TDI, OUTPUT);
  pinMode(PIN_TDO, INPUT);
  pinMode(PIN_SRST, OUTPUT);
  WRITE_PERI_REG(USB_SERIAL_JTAG_CONF0_REG,
    READ_PERI_REG(USB_SERIAL_JTAG_CONF0_REG)
  | USB_SERIAL_JTAG_USB_JTAG_BRIDGE_EN);
  // esp_rom_gpio_connect_out_signal(GPIO, IOMATRIX, false, false);
  esp_rom_gpio_connect_out_signal(PIN_TCK,   OMATRIX_TCK,  false, false);
  esp_rom_gpio_connect_out_signal(PIN_TMS,   OMATRIX_TMS,  false, false);
  esp_rom_gpio_connect_out_signal(PIN_TDI,   OMATRIX_TDI,  false, false);
  esp_rom_gpio_connect_out_signal(PIN_SRST,  OMATRIX_SRST, false, false);
  esp_rom_gpio_connect_in_signal (PIN_TDO,   IMATRIX_TDO,  false);
}

void unroute_usb_jtag_to_gpio()
{
  WRITE_PERI_REG(USB_SERIAL_JTAG_CONF0_REG,
    READ_PERI_REG(USB_SERIAL_JTAG_CONF0_REG)
  & ~USB_SERIAL_JTAG_USB_JTAG_BRIDGE_EN);
  pinMode(PIN_TCK,  INPUT);
  pinMode(PIN_TMS,  INPUT);
  pinMode(PIN_TDI,  INPUT);
  pinMode(PIN_TDO,  INPUT);
  pinMode(PIN_SRST, INPUT);
  digitalWrite(LED_BUILTIN, LED_OFF);
}

### Implementation

I hope the MicroPython maintainers or community will implement this feature

### Code of Conduct

Yes, I agree

emard, 2025-01-02:
Does micropython have C equivalent of
esp_rom_gpio_connect_out_signal(GPIO, IOMATRIX, bbolean, boolean);
esp_rom_gpio_connect_in_signal(GPIO, IOMATRIX, bbolean, boolean);

And C equivalent like mem32 of
WRITE_PERI_REG()
READ_PERI_REG()

then I can make it myself

andrewleech, 2025-01-02:
That's neat, I didn't know the usb-jtag peripheral could be externally routed, I'll have to try that out! 

The code snippet you provided should be pretty straightforward to convert into a user_c_module or even a dynamic C module that could be loaded at runtime.

emard, 2025-01-02:
magic iomux numbers are undocumented for S3 but someone on the
forum posted their values, i tried and it works at 2 Mbit/s
needs slightly patched openocd

https://github.com/emard/esp32s3-jtag
https://github.com/emard/openocd

iomux jtag values for C3 are known too (even documented),
with experimentation probably it would be possible to find out
values for S2, C6, H2 ...

emard, 2025-01-02:
It is flexible,any gpio pin can be assigned
to become any signal of the external jtag.

emard, 2025-01-03:
problem arised:

installed micropython and idf, compiled for esp32s3
cexample which I added above commands and all
seem to be fine

USB from arduino was
VID:PID = 303a:1001
it had serial port and jtag

USB from micropython is
VID:PID = 303a:4001
it has serial port only, without jtag endpoints

How to setup hardware serial and jtag to appear
on micropython?

Dzarda7, 2025-01-04:
Just FYI, I recently started maintaining the project that does something like that and is implemented in C. You might find it useful, it is called [esp-usb-bridge](https://github.com/espressif/esp-usb-bridge).

emard, 2025-01-04:
Yes I have seen it is great! I will post one wishlist there too :)

It is great as firmware implementation of the usb jtag protocol that
is also in the hardware. Protocol documentation is not so readable
so it's wonderful there exists source too.

For micropython I'd like that hardware JTAG (HWCDC+JTAG) is used.
Compared to firmware implementation I expect it to be more stable
and save some esp32s3 resources (cpu, ram, development).

only I'm not familiar with micropython compilation so I don't know how
to configure micropython to use HWCDC+JTAG.

Dzarda7, 2025-01-04:
I agree, feel free to open the issue. I will try to take this suggestion into an account and think about using the hardware. I need to get into that a bit more, but there is always something else :smile:.

emard, 2025-03-30:
For example ESP32C3 micropython binary comes with this feature by default enabled
because C3 has no OTG option. I can recompile micropython but options for C3 and S3
are different, I tried to change something but with no success because I'm not shure
which S3 set of options is equivalent to C3's in order to
get factory default USB-SERIAL and USB-JTAG on S3 like C3 by default has.
(C3 has serial port and hardware jtag always enabled)

Or even better if USB_JTAG can be enable at runtime without recompiling please
help me how.