micropython/micropython #16519: RP2: machine.RTC().datetime() does not advance during lightsleep

madozu, 2025-01-03:
### Port, board and/or hardware

Raspberry Pi Pico W and Raspberry Pi Pico 2 W

### MicroPython version

All versions since v1.24.0 (including latest v1.25.0-preview.172.g5d12df51f)

### Reproduction

1. Copy code below to Pico and name it "main.py"
2. Reset the Pico and allow the code to complete (approx 5 seconds). No repl needed as the code writes a log file to the flash storage
3. Connect the Pico to the USB port and inspect the file "local.log" on the Pico

Code used to test:
```# Test RTC sleep behaviour
import machine, os, time

SLEEP_MS = const(2000)

def write_log(s: str) -> None:
    dt = machine.RTC().datetime()
    log_text = "[RTC={6:02d}:{7:02d}:{8:02d} ticks={0:d}]  {1:s}".format(time.ticks_ms(), s, *dt)
    with open("local.log", "a") as file:
        file.write(log_text + "\n")

def safe_sleep(ms: int) -> int:
    begin = time.ticks_ms()
    wake = -1
    sleep = ms
    while sleep > 0:
        machine.lightsleep(sleep)
        wake += 1
        sleep = ms - time.ticks_diff(time.ticks_ms(), begin)
    return wake

write_log(f"Machine = '{os.uname()[4]}'")
write_log(f"Version = '{os.uname()[3]}'")
time.sleep_ms(SLEEP_MS)
write_log(f"After time.sleep_ms({SLEEP_MS})")
wakeups = safe_sleep(SLEEP_MS)
write_log(f"After safe_sleep({SLEEP_MS}) with {wakeups} early wakeups")
```

### Expected behaviour

The real time clock should advance by 2 seconds after running 'machine.lightsleep(2000)' like it was on micropython versions up to and including 1.23.0. Example output from **1.23.0**:
```
[RTC=00:00:01 ticks=49]  Machine = 'Raspberry Pi Pico W with RP2040'
[RTC=00:00:01 ticks=54]  Version = 'v1.23.0 on 2024-06-02 (GNU 13.2.0 MinSizeRel)'
[RTC=00:00:03 ticks=2058]  After time.sleep_ms(2000)
[RTC=00:00:05 ticks=4065]  After safe_sleep(2000) with 0 early wakeups
```

### Observed behaviour

The real time clock does not advance during machine.lightsleep(2000) as shown in the output below running on **1.24.0**:
```
[RTC=00:00:01 ticks=36]  Machine = 'Raspberry Pi Pico W with RP2040'
[RTC=00:00:01 ticks=41]  Version = 'v1.24.0 on 2024-10-25 (GNU 13.2.0 MinSizeRel)'
[RTC=00:00:03 ticks=2044]  After time.sleep_ms(2000)
[RTC=00:00:03 ticks=4049]  After safe_sleep(2000) with 31 early wakeups
```

### Additional Information

While machine.RTC().datetime() does not advance during machine.lightsleep(), **time.ticks_ms advances as expected**.

The funtion "safe_sleep()" was used to encapsulate calls to machine.lightsleep() because machine.lightsleep() returns after ~64ms. See [Issue #15622](https://github.com/micropython/micropython/issues/15622) and [Issue #16181](https://github.com/micropython/micropython/issues/16181)

### Code of Conduct

Yes, I agree

ssotheremail, 2025-04-19:
Not just you. I tried one of the very late previews of 1.25.0 with and without 16454 and see the same issue with time.time. 
I have a datalogger app that I used a Rpi pico2w for because of the potential for low power on batteries.  
I have re-written it to record the start time with time.time before using lightsleep at all. 
Then take a reading every minute with some ds18b20s (remebering to use time.sleep_ms for the 750ms reading delay then a short sacrificial lighsleep as using time.sleep_ms makes the next lighsleep return early). 
Then I use "safe_sleep" to wait till the next minute. 
Only problem with safe_sleep is it is granular to 64ms but for the present I am just grateful I am recording results at reasonable battery lfe.

So I use 3 workarounds on Pimonis 0.1.0 (similar to 1..25.0) at present (I have an e-paper display on the datalogger):
for time.time
for early return
time.sleep_ms (and time.sleep) interferes with lightsleep

madozu, 2025-04-20:
Thanks for confirmation ... so I'm not caught in my own filter-bubble ;-)

My workaround is to simply not use the Pico2 W but stay with the **PicoW** and **version 1.23** as working around the "RTC does not advance during lightsleep" problem (e.g. sync with NTP) and working around the other lightsleep problems consumes way more power than the Pico2 W could ever save with it's RP2350.

Unfortunately I have to say that in battery-powerered scenarios where low energy consumption **and** accurate time is needed, the Pico2 W is pretty pointless. At least for now.

madozu, 2025-05-09:
Today, I retested the RTC behavior with the current master branch. On top, I applied pull requests #16454 and #17272. The very good news is that this **solves**
- The early return/wakeup from lightsleep
- The interference between time.sleep and machine.lightsleep()

Unfortunately, the RTC issue still remains unchanged: The real time clock is frozen during machine.lightsleep()

ssotheremail, 2025-05-10:
I had 16454 added, now I have added #17272. Tested on a pico2W. My test code disables the wireless at the beginning with:
wlan = network.WLAN(network.AP_IF)
wlan.disconnect()
wlan.active(False)
wlan.deinit()

The good news is that the "lightsleep early return" problem has gone on the 2W, and as a bonus (not sure why) the "lightsleep early return after sleep" problem is gone too.  

I have another version which turns off the wireless near the end, before a lightsleep(25000) so I can measure the current with the wireless off. I get 6.3mA. With the wireless at default settings the lightsleep tests give the same results. 

As you say the "time" issue persists, but the ticks are still ok.

madozu, 2025-05-28:
For me the downloaded firmware RPI_PICO-20250528-v1.26.0-preview.158.g5cfafb73d.uf2 works fine when it comes to "light_sleep after sleep" and "light_sleep early wakeups". The python [test script from the top of this issue](https://github.com/micropython/micropython/issues/16519#issue-2767727382) with 60 sec sleep/light_sleep creates this log file:
```
[RTC=00:00:00 ticks=33]  Machine = 'Raspberry Pi Pico 2 W with RP2350'
[RTC=00:00:00 ticks=74]  Version = 'v1.26.0-preview.158.g5cfafb73d on 2025-05-28 (GNU 14.2.0 MinSizeRel)'
[RTC=00:01:00 ticks=60114]  After time.sleep_ms(60000)
[RTC=00:01:00 ticks=120154]  After safe_sleep(60000) with 0 early wakeups
```
However, as expected, the **real time clock does still not advance** during light_sleep.

ssotheremail, 2025-05-29:
Thanks for doing that. I ran your test program on a Pico2_w and got the same results as you.  As a result I had a play with my test program. As a result I have produced a new one which produces clearer results:

`import machine, time, os

led = machine.Pin("LED", machine.Pin.OUT)
led.off()

start_time = time.time()
before_tick = time.ticks_ms()
after_tick = time.ticks_ms()
sleep_length0 = after_tick - before_tick

time.sleep(5)
machine.lightsleep(5000)    
after_tick = time.ticks_ms()
sleep_length1 = after_tick - before_tick
before_tick = time.ticks_ms()

led.on()
machine.lightsleep(5000)
machine.lightsleep(5000)
led.off()
after_tick = time.ticks_ms()
sleep_length2 = after_tick - before_tick
before_tick = time.ticks_ms()

machine.lightsleep(5000)
led.on()
machine.lightsleep(5000)
after_tick = time.ticks_ms()
sleep_length3 = after_tick - before_tick
    
print("Elapsed Measured by Pico:  "+"{:5d}{:10d}{:10d}{:10d}{:10d}".format(time.time()-start_time,sleep_length0,sleep_length1,sleep_length2,sleep_length3))
`
This produces:
Elapsed Measured by Pico:      5         0     10001      5002         2

If you comment out one or other of the led on or off bits of code the timing changes.
I was a bit surprised. I would be interested if you see the same. The original "lightsleep after sleep" problem does still seem to be fixed. The program run time (with a wristwatch) approx agrees with the ticks and yes the time.time still doesn't work.

madozu, 2025-05-29:
On the Pico-W variants, the **LED is wired to the WiFi chip**. Therefore:
- Turning on the LED activates the WiFi chip (and might cause interrupts waking early from light_sleep)
- All time measurements that include time.ticks() should be done using time.ticks_diff() instead of doing integer arithmetic

ssotheremail, 2025-05-30:
Thanks, for looking. I had thought the chances of a rollover of ticks was unlikely after a minute of power on, but I have redone my code with time.ticks_diff:
......
after_tick = time.ticks_ms()
sleep_length0 = time.ticks_diff(after_tick,before_tick)
before_tick = time.ticks_ms()
....  etc
and get the same results . 

Basically you can use led.off seemingly without effect but if you use led.on, even before going into the lightsleep sequences, then the effect of timing becomes unpredictable and complex. 

I am sure you are right about interrupts from the wifi chip. I was aware of the led on the Ws being via the Wifi but was surprised with the effect on lightsleep given the efforts to avoid the lwip timers issue when wireless networking was not in use. I would not have thought interrupts were needed, in principle for the LED, but maybe they are.

BTW Did you see the same effect as me?

ssotheremail, 2025-05-30:
Yes, mostly agree with your explanations, except I always get the same time on my watch as the total number of ticks printed. For example I just ran it (with the time.ticks_diff version) and and got:
Elapsed Measured by Pico:      5         0     10001      5002         2 
and on my wristwatch got 16sec (Far be it from me to challenge centuries of Swiss timekeeping but......)
Though its handy that safe_sleep seems to mask whatever it is.

In case it's of interest, I just did some current measurements.
If I just power up the Pico2w I get about 16ma.
If I put this at the beginning of the code and run the program:
import network
wlan = network.WLAN(network.AP_IF)
wlan.disconnect()
wlan.active(False)
wlan.deinit()
and comment out all the Led code
Then run the program and the current goes up to about 29ma and fall back to around 17ma. 
However if I run the same code again the current falls to 6ma and then during the later lightsleeps its 3.75ma and when the program finishes it goes back to 17ma.

It will continue with lower currents when rerun until a hard reboot, when it goes back to the higher values. 

If you dont do the wlan deinit stuff, and turn the led on at the beginning then you only ever see the higher values (min 18ma) and when the program finishes it rests at 32ma, even if you turned off the led before the program finishes..

I have not rerun these tests with a picoW.

As you say it appears batteries and pico2W (and possibly picoW) are not a good match.
I am getting much better battery life (3x better) with the Pico2 now with recent (Pimoroni micropython 1.25.0) code and the workarounds, so not all bad.

Clearly now I am aware of the implications I would not go near the led on the W's if I was using lightsleep. Do you think its worth an issue or adding it to one of the others?

madozu, 2025-07-01:
I just updated the bug report from January this year with tests on the latest 1.26 preview build available. The problem is still present in the latest 1.26 preview firmware.

I also removed the "safe_sleep" code from the small test script, as with the current build, lightsleep does not wake up early.

ssotheremail, 2026-08-30:
No, your email spurred me to retest this with vanilla micropython 1.29.0 out of curiosity.

In case its still of even passing interest, I think that things have improved with the wider lightsleep question. I haven't explored this in depth but after a quick set of tests, for me:

Time.time does not advance during light sleep on any product.

The Pico, PicoW and Pico2 give consistent correct timing both manual stopwatch and measured by ticks, with the led being used and leaving the wifi as is. 

On the Pico2W, with the wifi disabled you get a pretty good min current of 4mA during lightsleep but unless you stop using led() the ticks are all over the place.

madozu, 2026-08-31:
Thanks for the explanation, this is in sync what I am observing. I get similar low power consumption on PicoW and Pico2W when the WiFi chip is fully turned off before lightsleep. And yes, using the led on Picos with WiFi turns on the WiFi chip as the led is routed through it. Therefore I am not using the led on any of the projects where power consumption is relevant.

Besides the machine.RTC() (and therefore also time.time()) problems, lightsleep has really improved over the last few releases.

closed 2026-09-16 by projectgus

madozu, 2026-09-24:
Great news after my vacation! I can confirm that with the latest preview "v1.30.0-preview.71.g1ea77a5d65 on 2026-09-21 (GNU 16.1.0 MinSizeRel)" the **RTC continues running** while in lightsleep() and also the sleeps last as long as indicated by the given millisecond value (no early wakeups). **Great work** 🎉 🚀.

Thanks a lot  @projectgus for creating the fix and thanks to @dpgeorge for adding it to the next release 👍 💯

ssotheremail, 2026-09-25:
Echoing Madozu, Great work. For me, on the pico, picow and pico2 the changes give expected overall manual timing (test should take 30s), correct ticks and RTC timing, as does the pico 2W as long as you dont use the LED. Current flow was affected by leaving the wireless un-disabled but not the timing. I saw currents between 4 and 6ma during lightsleeps with no led on a 2W.
As an observation, if you test with the 2W and use the LED then whether the wifi is disabled or not, the test takes unpredictable times between 10 or 20 sec. It seems that the led affects some different invocations of lightsleep on the same run, unpredictably.