micropython/micropython #16526: SD card / SDMMC broken for ESP32-S3

specternecter, 2025-01-05:
### Port, board and/or hardware

esp32 port - ESP32-S3

### MicroPython version

MicroPython v1.25.0-preview.114.gbdda91fe7 on 2025-01-05

### Reproduction

import machine
import os
import esp
esp.osdebug(True, esp.LOG_VERBOSE)

# Initialize the SD card interface using SDMMC in 4-line mode
try:
    #sd = machine.SDCard(1)
    sd = machine.SDCard(slot=1, width=4)
    os.mount(sd, "/sd")
    print("SD card mounted successfully.")
    
    print("Contents of the SD card:")
    for filename in os.listdir("/sd"):
        print(filename)
except Exception as e:
    print("Error initializing SD card:", e)

### Expected behaviour

Print a list of files on the SD card.

### Observed behaviour

import sd_test
I (809804) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (809854) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (809854) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (809864) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (809874) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (809884) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
E (809924) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
Error initializing SD card: 16
>>>

### Additional Information

I have tried everything to get an SD card to work like it does in sdmmc example for esp-idf (https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card/sdmmc). If I run this example in esp-idf, everything works perfectly fine right out of the box. It literally took 3 seconds to set up. After 7 hours of work and research, I simply can't get the sd card to work in micropython. Not SPI, not MMC, nothing. Micropython is supposed to be built on top of esp-idf so I would assume it should be just as easy. I looked at the SDCard class (https://docs.micropython.org/en/latest/library/machine.SDCard.html) and found that I should be able to configure the pins in mpconfigboard.h, to which I tried `
#define USDHC_DUMMY_PIN NULL , 0
#define MICROPY_USDHC1 \
  { \
        .cmd   = { 35}, \
        .clk   = { 36 }, \
        .cd_b  = { USDHC_DUMMY_PIN },\
        .data0 = { 37 },\
        .data1 = { 38 },\
        .data2 = { 33 },\
        .data3 = { 34 },\
  }` 
like it says, but it still tries to use ESP32 pins when I call the machine.SDCard(), like it also says to do. If I try to add the pins inside machine.SDCard() it says I have too many arguments. At one point I found someone with the same complaint but it had literally no information aside from "modify machine_sdcard.c" with no information on how to modify it and honestly I find that an absolutely terrible idea. Why modify the entire port to make a single board work, that should work from the beginning by default? I have no problem modifying a board to make it work but to me, modifying the port makes no sense unless it gets modified for everybody. I have spent 7 hours trying to get an sd card to work in micropython when it took 3 seconds to get to work in esp-idf. The only information I find are complaints about the same thing but no answers. This is insane.

### Code of Conduct

Yes, I agree

emard, 2025-01-06:
this works for my ESP32-S3
`
from machine import Pin, SDCard
import os

#    XIAO ESP32S3 MINI
#    Mouser P/N: 713-113991114    7$
#
#           XIAO  ESP32S3                   SDCARD
#              ┌─────┐                 ┌───────────────┐
#          ┌───┘ USB └───┐             │8   DAT1       │
#      TCK │1          5V│             │7   DAT0  MISO │
#      TMS │2         GND│ GND         │6   VSS2  GND  │
#      TDI │3        3.3V│ 3.3V        │5   CLK   CLK  │
#      TDO │4           9│ MISO        │4   VDD   3.3V │
#     SRST │5           8│ MOSI        │3   VSS1  GND  │
#       CS │6           7│ CLK         │2   CMD   MOSI │
#          │43 TX   RX 44│             │1   DAT3  CS   │
#          └─────────────┘              ╲ 9 DAT2       │
#              top view                   ─────────────┘
#                                           bot view

# mpremote connect /dev/ttyACM0 cp sdtest.py :/
# mpremote connect /dev/ttyACM0
# >>> import sdtest

def log(message):
    print(message)

def test_custom_sd():
    try:
        # Initialize SD card
        log("Initializing SD card...")
        sd = SDCard(slot=3, width=1, 
          cs  =Pin(6, pull=Pin.PULL_UP),
          sck =Pin(7, pull=Pin.PULL_UP),
          mosi=Pin(8, pull=Pin.PULL_UP),
          miso=Pin(9, pull=Pin.PULL_UP),
          cd=None, wp=None)
        log("SD card initialized successfully.")
        sd.info()

        # Mount the SD card
        log("Mounting SD card...")
        os.mount(sd, "/sd")
        log("SD card mounted successfully.")

        # Test file operations
        log("Writing to SD card...")
        with open("/sd/test.txt", "w") as f:
            f.write("Hello, custom SD card driver!")
        log("File written successfully.")

        log("Reading from SD card...")
        with open("/sd/test.txt", "r") as f:
            content = f.read()
        log(f"File read successfully: {content}")

        # Unmount the SD card
        os.umount("/sd")
        log("SD card unmounted successfully.")
    except Exception as e:
        log(f"Error: {e}")

test_custom_sd()
`

projectgus, 2025-01-07:
@specternecter  The code @emard posted above should work in SPI mode (width=1) where you pass in the pins specifically. Does that code work for `width=1` on your board, with the correct pins?

In SD mode (width=4), the current driver doesn't allow overriding any pin selections (on ESP32 original that was the only option, but on ESP32-S3 the hardware and ESP-IDF supports changing the pin selections but MicroPython doesn't yet support this.)

MicroPython currently picks these pins, which are the driver defaults in ESP-IDF:
https://github.com/espressif/esp-idf/blob/v5.2.2/components/driver/sdmmc/include/driver/sdmmc_default_configs.h#L90

In the ESP-IDF example, Espressif has overridden the example defaults to use the pins of the ESP32-S3-USB-OTG board which looks like the board you have:
https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card/sdmmc#pin-assignments-for-esp32-s3

... I'm unsure why they changed the defaults in the ESP-IDF example but didn't change the defaults in the ESP-IDF driver.

>  I looked at the SDCard class (https://docs.micropython.org/en/latest/library/machine.SDCard.html) and found that I should be able to configure the pins in mpconfigboard.h

Unfortunately that section of the docs is for the mimxrt port. There is a section above it for the ESP32 port, but looks like that section only documents the original ESP32. :facepalm: 

This is something we certainly should fix. In the meantime you can either use `width=1` mode with pin settings, or you can probably patch `ports/esp32/machine_sdcard.c` to change the pin assignments for `width=4`. At line 294 there is a line:

```c
        sdmmc_slot_config_t slot_config = SDMMC_SLOT_CONFIG_DEFAULT();
```

Change this line to:

```c
sdmmc_slot_config_t slot_config = {
    .clk = GPIO_NUM_36,
    .cmd = GPIO_NUM_35,
    .d0 = GPIO_NUM_37,
    .d1 = GPIO_NUM_38,
    .d2 = GPIO_NUM_33,
    .d3 = GPIO_NUM_34,
    .cd = SDMMC_SLOT_NO_CD,
    .wp = SDMMC_SLOT_NO_WP,
    .width   = SDMMC_SLOT_WIDTH_DEFAULT,
    .flags = 0,
}
```

... I haven't tested this as I don't have an S3 board with an SD card slot, but it hopefully will work as a stop-gap.

emard, 2025-01-07:
On classic ESP32 I have board with correct hardware pins for
4-bit mode, speed gain is not much, 4-bit mode was about 2x
faster than 1-bit mode. Maybe ESP32S3 has faster SD driver in
4-bit mode or if it as slow then something else should be "fixed"
too I guess :)

specternecter, 2025-01-07:
@emard in theory it should be at least 60% faster based on the hardware alone, but really it's anyone's guess until it's tested.

projectgus, 2025-01-07:
> @projectgus it did finally pick the right pins but for some reason still didn't work.

Uh, bad news. I have an 4-bit wide SD card breakout on the way, so I will try and take a look when it shows up.

specternecter, 2025-01-07:
@projectgus I just ran the esp-idf test and it all the sudden didn't work. I had to do a hard reset and then it did. So I'm currently in the process of trying again

specternecter, 2025-01-07:
@projectgus it does work!

import sdmmc_test
I (21235) gpio: GPIO[36]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (21285) gpio: GPIO[35]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (21285) gpio: GPIO[37]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (21295) gpio: GPIO[38]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (21305) gpio: GPIO[33]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (21315) gpio: GPIO[34]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (21355) gpio: GPIO[34]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
SD card mounted successfully.
Contents of the SD card:
System Volume Information
boot.py
NIHAO.TXT
FOO.TXT
>>>

For some reason if you upload when SD is mounted in 1-bit mode, you have to do a hard reset to get 4-bit mode to work. Otherwise there is no issue whatsoever.

projectgus, 2025-01-07:
> For some reason if you upload when SD is mounted in 1-bit mode, you have to do a hard reset to get 4-bit mode to work. 

Oh, interesting. Maybe the card initialised in 1-bit mode and didn't respond until it had been cycled?

> Otherwise there is no issue whatsoever.

Great news. I think the correct fix here will be to support passing in custom pins to the SDCard constructor on ESP32-S3 in `width=4` mode. (Can test for this in a generic way with `#if SOC_SDMMC_USE_GPIO_MATRIX`.)

If anyone wants to submit a PR for this then it'd be welcome, otherwise I'll try to take a look when my card adapter arrives.

specternecter, 2025-01-07:
@projectgus  thank you so much!

PR #16702 (2025-03-13): esp32: Improve machine.SDCard support on newer chips
### Summary

Updating the esp32 port `machine.SDCard` support for other chips:

* Added support for configuring pins in SD/MMC mode on ESP32-S3. Closes #8514. Closes #16526.
* Added support for SDCard in SPI mode on ESP32-S2, C3 and C6. This is a little restrictive on C3 and C6 as they only have one available SPI bus, so can't use this and `machine.SPI` together. Have enabled on all boards by default (the code size impact is about 4KB).
* Explicitly disable slot 0 on original ESP32 (and document this). There are effectively zero ESP32 boards [... cut]
closed 2025-03-13 by dpgeorge